SpenceKonde/DxCore #2: Adapt Logic, Analog Comp library. Add actual DxCore library to supply some helper functions

SpenceKonde, 2020-08-08:
Logic will be trivial, analog comp. looks easy too.

DxCore library would be a good place to put the autotune related calls,

SpenceKonde, 2020-08-08:
Dear lord, Logic turned out to be an unholy battle!
It would have been a piece of cake if the library was working to begin with. But it wasn't, and recognizing and fixing that turned out to be a real job....

SpenceKonde, 2020-08-08:
And Analog Comparator library will be bigger deal to port than I thought, because I am not familiar with the C++ constructs .in that library, and it's built for the single comparator of the megaAVR 0-series, not more than one like we have here.

MCUdude, 2020-08-10:
Could you try to describe what issues you're experiencing with the Logic library? If the "silicon" implementation is similar to the mega0 series, it should be a fairly straight forward process.

And the Comparator library should be easy too. It's just that I don't have any DA hardware to test with

SpenceKonde, 2020-08-10:
Logic is in and working (also fixed some bugs, added some features, and sent you PR with those additions), it was just a much much bigger deal than I thought it would be, because of a bunch of problems that were present in mine but not yours, plus some really silly mistakes on my part.

SpenceKonde, 2020-08-10:
Well, I can fix the DA hardware problem easily enough, email me your mailing address. 

but the real problem for me is that the 1-series tinies and DX parts have 3 comparators. The mega 0-series has 1, and I realized - I know C. I basically don't know C++  - I can modify, and I can add options by copying all the things you did, but it is done without much understanding, I don't know how I would add two more comparators. I know how to check whether they're there, but I don't know how to make. You know I've never actually written a library right? I haven't. I've modified and extended the hell out of libraries.... but I've seen vastly different sorts of constructions in libraries, and have no idea how the author chose one or the other to use, or what the differences are. Some define a class.... some do that weird stuff with "namespaces" whatever that is, like jtag2updi does all over the place. And then there are libraries like EEPROM, which is... a struct?! I thought structs were like, objects made up of variables with predictable places in memory relative to eachother, so you can pass around pointers to them, no idea how the same thing can also be a bunch of callable functions.... but okay! See the place I'm coming from? I am really good at modifying, but I've never had any real experience with the ++ of C++ itself, and if I try to look it up, it's so different from embedded context that it doesn't look like any of the things I want to understand...

MCUdude, 2020-08-10:
> but the real problem for me is that the 1-series tinies and DX parts have 3 comparators.

This is the easy part. Just add

```c++
AnalogComparator Comparator1(1, AC1);
AnalogComparator Comparator2(2, AC2);
```

to Comparator.cpp and mark them as extern in Comparator.h. The user can now call `Comparator1` and `Comparator2`,
The only real fiddly part will be to make sure that right IO pins get configured correctly in init(). At the moment it's hardcoded to match mega0's, but since the analog comparator number is known at compile-time, I believe I can make the compiler optimize out code that "never will be true".

To be honest, It took me quite a while before I actually wrote my first library. I've also usually just modified existing ones, and sometimes made a PR. And I do agree that the Logic and Comparator libraries are a little more Cplusplus'y than you're probably used to. But this makes the library more userfriendly. Or at least, I think so. They are quite simple to use compared to setting the peripherals up manually, right?

The EEPROM library is special indeed. It's called "Meta programming", and it's basically the compiler that creates the code at compile time. Very powerful, very efficient, and almost impossible to read and understand. [I've created template-based code](](https://github.com/MCUdude/SigmaDSP/blob/d027cf2998da4370c00a900660ecee864e5f2745/src/SigmaDSP.h#L162-L174)) once, and that was for my SigmaDSP library. And in that exact application, it made sense.

MCUdude, 2020-08-10:
Oh, it will also make sense to create a very similar library for the built-in zero-cross detection (ZCD)! I'll take care of that if you'll accept a PR. I promise to document it properly!

ZCD is actually pretty useful in certain applications. I made a ZCD circuit using a series resistor and some clamping diodes with an ATtiny13. I had an issue where the power amplifier in my living room would output a loud clicking noise the exact moment I flipped its power switch. With a precisely timed ZCD circuit, I've totally eliminated the noise.

SpenceKonde, 2020-09-07:
Huzzah - thanks to the hard work of @MCUdude we have Comparator library now! And ZCD soon, and I'm about to pull in his fixes to logic as well! Hoping to get a bunch of other stuff sorted out for the DB parts (they do need their own board definitions, sadly - need menu option to enable/disable MVIO)

SpenceKonde, 2020-10-11:
This has already been sorted out quite some time ago

closed 2020-10-11 by SpenceKonde