SpenceKonde/DxCore #5: Support for AVR128DA48 Curiosity Nano

sstaub, 2020-08-09:
Can you support the AVR128DA48 Curiosity Nano? The upload should work like the ATMEGA4809 Curiosity Nano, it is the same chip for UPDI Debug. Atmel nEDBG, see MegaCoreX.

sstaub, 2020-08-09:
Doesn't work:
avrdude: Can't find programmer id "curiosity_updi"

SpenceKonde, 2020-08-09:
Should be sorted now.

sstaub, 2020-08-09:
Get following error:
Arduino: 1.8.13 (Mac OS X), Board: "AVR DA 48, AVR128DA48, 24 MHz, 1.9V, Disabled, Disabled, EEPROM retained, Input (no output, ever), TCB2 (recommended)"

Sketch uses 3138 bytes (2%) of program storage space. Maximum is 131072 bytes.
Global variables use 363 bytes (2%) of dynamic memory, leaving 16021 bytes for local variables. Maximum is 16384 bytes.
/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/sstaub/Library/Arduino15/packages/DxCore/hardware/megaavr/1.0.0/avrdude.conf -v -pavr128da48 -ccuriosity_updi -Pusb -Uflash:w:/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino_build_71553/blinky4809.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/sstaub/Library/Arduino15/packages/DxCore/hardware/megaavr/1.0.0/avrdude.conf"
         User configuration file is "/Users/sstaub/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AVR128DA48
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          4    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           flash          0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        512   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.17 (rel. 514)
         Serial number   : MCHP3280031800000588
         Vtarget         : 3.30 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 100 kHz

avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sstaub, 2020-08-14:
Any idea what the reason?

SpenceKonde, 2020-08-14:
Haven't the faintest.

MCUdude, 2020-08-16:
That error looks familiar to me. I got a similar one when I used my 4809 Curiosity Nano board with MegaCoreX.

The issue that caused this problem on my board was a bug in the nEDBG firmware. When the device suddenly got unplugged from my computer without unmounting, the onboard voltage regulator sometimes was set to output 1.8V into the ATmega4809 instead of 5V. I used Atmel Studio to fix this. It can easily be checked by measuring the voltage on VCC pin

sstaub, 2020-08-16:
I can program the board with Atmel Studio 7 without any problem and additional
 configuration.

MCUdude, 2020-08-16:
But is the measured voltage at VCC 5.0V, or is it 1.8V?

sstaub, 2020-08-17:
> But is the measured voltage at VCC 5.0V, or is it 1.8V?

VTG is 3.3V as expected

MCUdude, 2020-08-17:
OK. Then I don't know what's wrong. I'll see if I can get myself an AVR128DA48 curiosity board. There are someone with inside information I can ask. But I'm not sure when I can get a proper answer.

If I were you I'd ask over at the Avrfreaks forum. There are some true wizards over there that may be able to help. There are also Atmel/Microchip employees there that really know stuff.

sstaub, 2020-08-17:
I have an ATmega 4809 Curiosity board which works without a problem with the MegaCoreX.
It is the same debugger part like the AVR128DA48 curiosity board, so I would expect that it works seamless with the DxCore.

MCUdude, 2020-08-17:
> I have an ATmega 4809 Curiosity board which works without a problem with the MegaCoreX.
It is the same debugger part like the AVR128DA48 curiosity board, so I would expect that it works seamless with the DxCore.

That's my experience too. Never had an issue with the 4809 Curiosity Nano except for the bug that caused the output voltage to go down to 1.8V (or even lower, I don're remember exactly). Again, Avrfreaks!

sstaub, 2020-08-26:
I got the board working with the Arduino Pro IDE v0.1.1. The temporary trick is to move the compiled .hex file manually to the file system of the Curiosity nEDGB debugger. 
The Serial port to the debugger is USART1, Serial1.xxx() make the connection.
It would nice to have the printf functionality like with the MegaCoreX framework.

MCUdude, 2020-09-04:
https://www.avrfreaks.net/forum/avr128da48-curiosity-nano-and-avrdude

I'm not willing to go down the rabbit hole and try to solve this...

sstaub, 2020-09-04:
> https://www.avrfreaks.net/forum/avr128da48-curiosity-nano-and-avrdude
> 
> I'm not willing to go down the rabbit hole and try to solve this...

Ok, thank you for your efforts.

SpenceKonde, 2020-10-11:
Looks like this one is out of my league, heh.

closed 2020-10-11 by SpenceKonde

alexmchp, 2021-01-19:
Looks like the issue was caused by an avrdude incompatibility with the C-nano. It is fixed in the latest avrdude patch, more details here: https://www.avrfreaks.net/forum/avr128da48-curiosity-nano-and-avrdude

SpenceKonde, 2021-01-20:
Yes, 1.3.0 is going to have support for it, and I'm working on finishing up the packages to support that for release before the clock strikes midnight

commit 3361c17d (2020-08-09): Fix #5? 

Maybe? I don't have the hardware to test with.
commit 41700faa (2020-08-09): Curiosity support

Fix #5