SpenceKonde/DxCore #6: Arduino IDE v1.0.1 is not seen in the board menu

sstaub, 2020-08-16:
After updating the DX core form v1.0.0 to v1.0.1, the boards are nor more present in the Tools/Board menu. The core is listed in the board manager.

SpenceKonde, 2020-08-16:
I am fully aware of this. Didn't have a chance to finish fixing the board manager files last night. And Per has tipped me off to an issue with my repacked toolchains which is going to be... awkward... to fix

SpenceKonde, 2020-08-16:
Currently in the process of repacking them correctly. I must say, it's bloody annoying how windows shits on the permissions of unix files.

SpenceKonde, 2020-08-16:
Okay, there are other problems with 1.0.1 - it's been pulled from boards manager. I apparently did something wrong with my repacked toolchains packages, but I'm not exactly sure what.

SpenceKonde, 2020-08-16:
It looks like at present, it may not be possible to release third party avr-gcc packages - they get set to not-executable... https://github.com/SpenceKonde/ReleaseScripts/pull/32

closed 2020-08-16 by SpenceKonde