SpenceKonde/DxCore #7: Version 1.0.1 "device-specs/specs-avr128da64: No such file or directory"

DustinWatts, 2020-08-16:
Hi!

Version 1.0.1 gives me an "avr-g++: error: device-specs/specs-avr128da64: No such file or directory" error. 

Version works fine and a simple blink sketch complies without error.

SpenceKonde, 2020-08-16:
Yet you don't get a permission denied error... ?

Is that the board manager installation?

Also, I did not mean for board manager version 1.0.1 to be deployed, because I need to repackage the toolchain files, apparently my autoupdate script was never deactivated :-(

per1234, 2020-08-16:
I don't know if it's the cause this time, but several times I have seen reports of this `avr-g++: error: device-specs/specs-...: No such file or directory` error that turned out to have been caused by an `GCC_EXEC_PREFIX` environment variable. Apparently GCC has some special internal handling of this environment variable.

Reference:
https://per1234.github.io/ino-troubleshooting/compiling.html#avr-g-error-device-specsspecs--no-such-file-or-directory

DustinWatts, 2020-08-16:
No, no permission denied error. And yes trough the board manager installation. But since version 1.0.1 is not there anymore, let's close this to confuse things in the future.

closed 2020-08-16 by DustinWatts

SpenceKonde, 2020-08-16:
Those files are definitely in the archive....

SpenceKonde, 2020-08-16:
I think I may be in deeper water than I was expecting here... I am not going to try to sort this out at this moment, as I don't really know what to do, and these problems imply that it is significantly more complicated than I was giving it credit for.

per1234, 2020-08-19:
Here's a current reference for the `GCC_EXEC_PREFIX` environment variable as a cause of the `avr-g++: error: device-specs/specs-...: No such file or directory` error:
https://forum.arduino.cc/index.php?topic=701026

It's fixed it every time the error is reported on the forum. I think it's always been Windows users before now (which is why my instructions are Windows-specific), but that might simply be because the majority of Arduino users use Windows.

An easy check would be to run this from the terminal:
```
echo "$GCC_EXEC_PREFIX"
```