SpenceKonde/DxCore #8: avrdude: jtagmkII_paged_write(): fatal timeout/error communicating with programmer (status -1)

DustinWatts, 2020-08-16:
Hi!

Using jtag2updi on an ATMega328p with a FTDI231. After successfully burning the bootloader to an AVR128DA64 I get the following error:

avrdude: jtagmkII_paged_write(): timeout/error communicating with programmer (status -1)
avrdude: jtagmkII_paged_write(): fatal timeout/error communicating with programmer (status -1)

I tested the chip using MPLABX and a PICkit4 and it programs fine. Using jtag2updi on a ATMega328p.

Might worth mentioning on fuse5 and fuse6 i get an error when burning the bootloader:

```
avrdude: reading input file "0b11001001"
avrdude: writing fuse5 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse5 written
avrdude: verifying fuse5 memory against 0b11001001:
avrdude: load data fuse5 data from input file 0b11001001:
avrdude: input file 0b11001001 contains 1 bytes
avrdude: reading on-chip fuse5 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "fuse5", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xf2 instead of 0xc9 (double check with your datasheet first).
avrdude: 1 bytes of fuse5 verified
avrdude: reading input file "0x03"
avrdude: writing fuse6 (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of fuse6 written
avrdude: verifying fuse6 memory against 0x03:
avrdude: load data fuse6 data from input file 0x03:
avrdude: input file 0x03 contains 1 bytes
avrdude: reading on-chip fuse6 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "fuse6", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xf8 instead of 0x03 (double check with your datasheet first).
avrdude: 1 bytes of fuse6 verified
```

SpenceKonde, 2020-08-17:
The datasheet states very clearly that unused fusebits should be written zero on these new parts, I hope it's not going and writing them 1's anyway... (btw, some of the supposedly unused bytes do remember their values when written... which is....kind of strange?)

DustinWatts, 2020-08-17:
I quickly checked. I don't think it does anything. When the message comes back saying "You probably want to use 0xf8 instead of 0x03". It suggest the value it is at. Because when I set it to 0x00 using MPLABx it now suggests: "You probably want to use 0x00 instead of 0x03"

BTW I don't think the startup delay (fuse6) is related to my uploading a sketch issue. But it does seem strange.

DustinWatts, 2020-08-17:
Alright! "Fixed" the issue. The problem was that the jtag2updi sketch on my programmer was apparently too old. Although I could program the new ATtiny's just fine (and thus thinking the programmer was fine), I needed to reprogram it with an newer download of jtag2updi. 

The thing that confused me was that it came back saying the bootloader was burned successfully. It didn't burn the bootloader at all. 

I'll close this one for now :)

closed 2020-08-17 by DustinWatts