SpenceKonde/DxCore #10: SPI1 library doesn't work

SpenceKonde, 2020-08-21:
I attempted to copy the SPI library, replacing every instance of SPI with SPI1. On Wire, despite having to do it to more files, and far more functions, this worked fine, but my SPI1 library gets WACKY errors. Anyone want to look at it and see what the hell I did wrong?

MCUdude, 2020-08-23:
Could you post the errors here?

SpenceKonde, 2020-08-28:
```
C:\arduino-1.8.13-portable-skbk\arduino-builder -dump-prefs -logger=machine -hardware C:\arduino-1.8.13-portable-skbk\hardware -hardware C:\arduino-1.8.13-portable-skbk\portable\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13-portable-skbk\tools-builder -tools C:\arduino-1.8.13-portable-skbk\hardware\tools\avr -tools C:\arduino-1.8.13-portable-skbk\portable\packages -built-in-libraries C:\arduino-1.8.13-portable-skbk\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrdx28opti:chip=128A,clock=24,adcworkaround=simple,timeout=1sec,bodvoltage=1v9,bodmodeactive=disabled,bodmodesleep=disabled,millis=tcb2,bootloaderusart=ser0 -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_516377 -warnings=all -build-cache C:\Users\Spence\AppData\Local\Temp\arduino_cache_319645 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr\libraries\SPI1\examples\DigitalPotControl\DigitalPotControl.ino
C:\arduino-1.8.13-portable-skbk\arduino-builder -compile -logger=machine -hardware C:\arduino-1.8.13-portable-skbk\hardware -hardware C:\arduino-1.8.13-portable-skbk\portable\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13-portable-skbk\tools-builder -tools C:\arduino-1.8.13-portable-skbk\hardware\tools\avr -tools C:\arduino-1.8.13-portable-skbk\portable\packages -built-in-libraries C:\arduino-1.8.13-portable-skbk\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrdx28opti:chip=128A,clock=24,adcworkaround=simple,timeout=1sec,bodvoltage=1v9,bodmodeactive=disabled,bodmodesleep=disabled,millis=tcb2,bootloaderusart=ser0 -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_516377 -warnings=all -build-cache C:\Users\Spence\AppData\Local\Temp\arduino_cache_319645 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr\libraries\SPI1\examples\DigitalPotControl\DigitalPotControl.ino
Using board 'avrdx28opti' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr
Using core 'dxcore' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr
Detecting libraries used...
"C:\\arduino-1.8.13-portable-skbk\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdx28 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB2 -DSIMPLE_ADC_WORKAROUND "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_516377\\sketch\\DigitalPotControl.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for SPI1.h: [SPI1@1.0.2]
ResolveLibrary(SPI1.h)
  -> candidates: [SPI1@1.0.2]
"C:\\arduino-1.8.13-portable-skbk\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdx28 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB2 -DSIMPLE_ADC_WORKAROUND "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\SPI1\\src" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_516377\\sketch\\DigitalPotControl.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\arduino-1.8.13-portable-skbk\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdx28 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB2 -DSIMPLE_ADC_WORKAROUND "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\SPI1\\src" "C:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\SPI1\\src\\SPI1.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\arduino-1.8.13-portable-skbk\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdx28 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB2 -DSIMPLE_ADC_WORKAROUND "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\DxCore\\megaavr [... cut ...]
   SPI1.transfer(value);
        ^~~~~~~~
Using library SPI1 at version 1.0.2 in folder: C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr\libraries\SPI1 
exit status 1
'SPI_t {aka struct SPI_struct}' has no member named 'begin'

```

What have I missed??? Why is it talking about SPI_t ?! I changed every reference to SPI1! Where is it pulling the SPI_t from?! and what on earth does `error: expected ')' before '*' token` mean?! Neither of those tokens make any sense to me for the line in question!

MCUdude, 2020-08-28:
Well, SPI1 is already defined in ioavr128da48.h:

```c
#define TWI0                  (*(TWI_t *) 0x0900) /* Two-Wire Interface */
#define TWI1                  (*(TWI_t *) 0x0920) /* Two-Wire Interface */
#define SPI0                  (*(SPI_t *) 0x0940) /* Serial Peripheral Interface */
#define SPI1                  (*(SPI_t *) 0x0960) /* Serial Peripheral Interface */
```

MCUdude, 2020-09-07:
OK, I got the SPI1 library to compile without errors or warnings. Here's what you'll have to do:

* Change the object name to something else than `SPI1`. I used `spi1`. Do this here:
https://github.com/SpenceKonde/DxCore/blob/3e98c53ea8671d0704ac75ec1b0b6d9bdf82777e/megaavr/libraries/SPI1/src/SPI1.h#L226
https://github.com/SpenceKonde/DxCore/blob/3e98c53ea8671d0704ac75ec1b0b6d9bdf82777e/megaavr/libraries/SPI1/src/SPI1.cpp#L344
  
  
* Replace `SPISETTINGS` class name with `SPI1SETTINGS. Do this here:
https://github.com/SpenceKonde/DxCore/blob/3e98c53ea8671d0704ac75ec1b0b6d9bdf82777e/megaavr/libraries/SPI1/src/SPI1.cpp#L147
https://github.com/SpenceKonde/DxCore/blob/3e98c53ea8671d0704ac75ec1b0b6d9bdf82777e/megaavr/libraries/SPI1/src/SPI1.cpp#L254

SpenceKonde, 2020-09-07:
THANK YOUUUUUU!!!!

SpenceKonde, 2020-09-15:
Fixed!

closed 2020-09-15 by SpenceKonde