SpenceKonde/DxCore #13: add printf() support

SpenceKonde, 2020-08-28:
MegaCoreX has this. DxCore and megaTinyCore should, but don't.

MCUdude, 2020-09-04:
Are you 100% sure that DxCore and megaTinyCore don't have printf?

https://github.com/SpenceKonde/DxCore/blob/e3789005948d00a8e1018b22046134c35c65128a/megaavr/cores/dxcore/api/Print.h#L81-L82

https://github.com/SpenceKonde/DxCore/blob/e3789005948d00a8e1018b22046134c35c65128a/megaavr/cores/dxcore/api/Print.cpp#L207-L227

SpenceKonde, 2020-09-04:
Huh, someone complained about them not being in, didnt have time to check...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2020-09-07:
You can safely close this. I used DxCore `Serial1.printf` yesterday, and it works perfectly.

Personally, I _love_ printf, and it's stupid, strange, and sad that the Arduino developers refuse to add it to the print class. It's very useful for other things than Serial too. Since it's available for use with all libraries, it works out of the box on a bunch of LCD/OLED/TFT libraries. I've been using printf with the U8g2 graphical library
(`u8g2.printf(F("Hello World %d\n"), myVal);`), and it's just excellent.

sstaub, 2020-09-07:
Can also confirm that printf works. The report from me in #5 was wrong, Sorry.

closed 2020-09-07 by SpenceKonde