SpenceKonde/DxCore #16: Pinout diagrams

MCUdude, 2020-09-07:
We need pinout diagrams for the AVR DB series as well. I'll see what I can do about this.

After #15 is merged, all pinout diagrams need to be updated to indicate that the ZCD hardware is actually there and fully supported by DxCore.

SpenceKonde, 2020-09-09:
Yes, thank you! I was planning to either create an issue or message you directly about this.... We also need to clearly mark:
PORTC as supporting MVIO, and what was the power pin in the middle of PORTC as being VDDIO2 (the MVIO supply) on DB48, DB64 parts. PD0 is replaced with VDDIO2 on DB28, DB32 parts - that pin number is to be skipped (PC3 is 11, then there's VDDIO2, then PD1 is 13) - I think the payoff in consistency is well worth the gap.

Also, the opamp pins want to be marked... PORTD and E continue to be the analog ports (though VDDA, between the analog-heavy PORTD, and the analog-heavy PORTE, is may be connected to VDD internally - it says like 5 times in the datasheet that they're connected, right next to diagrams and paragraphs and tables saying that they are separate power domains - I have a support inquiry open with Microchip about this)
PD1: OP0 INP, PD2: OP0 OUT, PD3: OP0 INN, PD4: OP1 INP, PD5: OP1 OUT, PD7: OP1 INN, (PD6 is the DAC output, and it wouldn't do to have two key analog-related features fight eachother for that pin.)

I also have an inquiry with Microchip asking exactly which parts have the problem with the TCD0 PORTMUX (a followup to when I reported it)... neither the AVR32DA32 nor the AVR128DB28 have the bug... I want to make the core use the more desirable pins if the part isn't effected by the bug... particularly since I also think I might want to pinswap USART0 by default. PA0 and PA1 are the pins used for the external crystal... and which pins are the TCD0 PWM outputs currently on? the ones that USART0 pinswaps to!

MCUdude, 2020-09-13:
Creating pinout diagrams for the entire DB series takes some time, but here are updated AVR DA pinout diagrams where ZCD has been added!
[AVR-DA.zip](https://github.com/SpenceKonde/DxCore/files/5213927/AVR-DA.zip)

SpenceKonde, 2021-01-09:
DB-series pinouts are now checked in.

closed 2021-01-09 by SpenceKonde