SpenceKonde/DxCore #17: Add Opamp library in the spirit of ZCD, Logic, and Comparator

SpenceKonde, 2020-09-08:
The DB series has a sick set of on-chip opamps. Really cool stuff. I don't do much with analog, as it happens, but it sounds like these are to analog what CCL is to digital logic. They very much lend themselves to a similar library.

MCUdude, 2020-09-12:
OPAMPS are great. I've used them for all sorts of Analog audio stuff, but I don't think the built-in OPAMPS the DB series is meant for that.

They are useful if you want to invert, gain, buffer or do math (in the analog domain) on an analog signal. DxCore definitely needs an OPAMP library! But now that you have the ZCD and Comparator libraries; two very simple and straight forward libraries (compared to Logic), you should easily be able to pull this off.

MCUdude, 2021-01-29:
I have some spare time, so I'm probably going to look into creating a library for the opamp peripheral if that's ok? However, I don't own any AVR-DB hardware, so all I can do is to make sure the code compiles.

MCUdude, 2021-01-29:
The idea is that even though OPAMP is a single peripheral (there are no OPAMP0.[something] and OPAMP1.[something]), I want each Opamp to have its own object. This can be achieved by passing all seven opamp specific register to the constructor.

MCUdude, 2021-01-30:
I haven't tested the library yet, but the fundamentals is there. I'm currently working on creating examples to demonstrate what the hardware is capable of. Here's an example sketch for setting up a non-inverting amplifier with programmable gain. Thoughts?

```c++
/***********************************************************************|
| AVR-DB Opamp library                                                  |
|                                                                       |
| Non_inverting_amplifier.ino                                           |
|                                                                       |
| A library for interfacing with the built-in AVR-DB Opamps             |
| Developed in 2021 by MCUdude                                          |
| https://github.com/MCUdude/                                           |
|                                                                       |
| In this example we use opamp 0 as a non-inverting amplifier with a    |
| gain multiplier set by the built-in resistor ladder.                  |
|                                                                       |
|                 | Gain |  Wiper value  | R1  | R2  |                  |
|                 |------|---------------|-----|-----|                  |
|                 | 1.06 | wiper::wiper0 | 60k | 4k  |                  |
|                 | 1.14 | wiper::wiper1 | 56k | 8k  |                  |
|                 | 1.33 | wiper::wiper2 | 48k | 16k |                  |
|                 | 2.00 | wiper::wiper3 | 32k | 32k |                  |
|                 | 2.66 | wiper::wiper4 | 24k | 40k |                  |
|                 | 4.00 | wiper::wiper5 | 16k | 48k |                  |
|                 | 8.00 | wiper::wiper6 | 8k  | 56k |                  |
|                 | 15.0 | wiper::wiper7 | 4k  | 60k |                  |
|                                                                       |
|                          |\                                           |
|                          |  \                                         |
|                 PD1 -----| +  \                                       |
|                          |     |-----+----- PD2                       |
|                     +----| -  /      |                                |
|                     |    |  /        |                                |
|                     |    |/          |        Gain = 1 + (R2 / R1)    |
|          ______     |     ______     |                                |
|    +----|__R1__|----+----|__R2__|----+                                |
|    |                                                                  |
|   ___                                                                 |
|    -                                                                  |
|                                                                       |
| See Microchip's application note TB3286 for more information.         |
|***********************************************************************/

#include <Opamp.h>

void setup()
{
  // Configure opamp input pins
  Opamp0.input_p = in_p::pin;   // Connect positive input to external input pin (PD1)
  Opamp0.input_n = in_n::wiper; // Connect negative input to the "middle" position of the resistor ladder
  
  // Configure resistor ladder
  Opamp0.ladder_top = top::output;     // Connect the resistor ladder top to the opamp output
  Opamp0.ladder_bottom = bottom::gnd;  // Connect the resistor ladder bottom to ground
  Opamp0.ladder_wiper = wiper::wiper3; // Set a gain of two
  
  // Enable opamp output
  Opamp0.output = out::enable;
  
  // Initialize Opamp0
  Opamp0.init();

  // Start the Opamp hardware
  Opamp::start();
}

void loop()
{

}

```

MCUdude, 2021-01-30:
Here is another example where Opamp0 drives the inverting Opamp1 amplifier:

```c++
/***********************************************************************|
| AVR-DB Opamp library                                                  |
|                                                                       |
| Inverting_amplifier_with_follower.ino                                 |
|                                                                       |
| A library for interfacing with the built-in AVR-DB Opamps             |
| Developed in 2021 by MCUdude                                          |
| https://github.com/MCUdude/                                           |
|                                                                       |
| In this example we use opamp 0 as voltage follower to drive the       |
| inverting opamp 1. Since the output can't go below 0V, we center the  |
| output voltage around Vdd/2, which is 5V if your supply voltage is    |
| 5V. This means that if 1.5V is applied to PD1, PD5 will output 3.5V.  |
|                                                                       |
|                 | Gain   |  Wiper value  | R1  | R2  |                |
|                 |--------|---------------|-----|-----|                |
|                 | -0.067 | wiper::wiper0 | 60k | 4k  |                |
|                 | -0.143 | wiper::wiper1 | 56k | 8k  |                |
|                 | -0.334 | wiper::wiper2 | 48k | 16k |                |
|                 | -1.00  | wiper::wiper3 | 32k | 32k |                |
|                 | -1.667 | wiper::wiper4 | 24k | 40k |                |
|                 | -3.00  | wiper::wiper5 | 16k | 48k |                |
|                 | -7.00  | wiper::wiper6 | 8k  | 56k |                |
|                 | -15.0  | wiper::wiper7 | 4k  | 60k |                |
|          |\                                                           |
|          |  \                                                         |
|  PD1 ----| +  \           ______           ______                     |
|          | OP0 |----+----|__R1__|----+----|__R2__|----+               |
|     +----| -  /     |                |                |               |
|     |    |  /       |                |    |\          |               |
|     |    |/         |                |    |  \        |               |
|     |               |                +----| -  \      |               |
|     +---------------+                     | OP1 |-----+----- PD5      |
|                                 Vdd/2 ----| +  /                      |
|                                           |  /                        |
|                                           |/      Gain = - (R2 / R1)  |
|                                                                       |
|                                                                       |
| See Microchip's application note TB3286 for more information.         |
|***********************************************************************/

#include <Opamp.h>

void setup()
{
  // Configure Opamp0 as a voltage follower tied to the Opamp1 resistor ladder
  Opamp0.input_p = in_p::pin;    // Connect positive input to external input pin (PD1)
  Opamp0.input_n = in_n::output; // Connect negative input to the opamp output
  Opamp0.output  = out::enable;  // Enable output

  // Configure Opamp1 pins
  Opamp1.input_n = in_n::wiper;    // Connect negative input to the "middle" position of the resistor ladder
  Opamp1.input_p = in_p::vdd_div2; // Connect positive input to Vdd/2 internally
  Opamp1.output  = out::enable;    // Enable output

  // Configure Opamp1 resistor ladder
  Opamp1.ladder_top    = top::output;   // Connect the resistor ladder top to the opamp output
  Opamp1.ladder_bottom = bottom::link;  // Connect the resistor ladder bottom to the output of the previous opamp
  Opamp1.ladder_wiper  = wiper::wiper3; // Set a gain of -1

  // Initialize Opamps
  Opamp0.init();
  Opamp1.init();

  // Start the Opamp hardware
  Opamp::start();
}

void loop()
{

}
```

SpenceKonde, 2021-02-08:
Opamp is in, seems to work! Had to make some minor changes - there was a problem with start checking the status, since it was static method referring to non-static property, and I added an option to not wait for settled, and to control enabled (anything user has initialized will now be enabled, unless they explicitly disabled it). Added output to readme. Inrange was backwards, and made it not change anything unless user has set it for the opamp they're initializing. Seems to work. Good job, thank you!!

SpenceKonde, 2021-02-08:
I'm all for examples - they look good to me, and most Arduino people probably don't know how to use opamps, One comment - "so why would someone want to amplify an analog signal? Can we measure the output of the opamps with the ADC or something?" (I am assuming that you read the opamp output pin?

closed 2021-02-08 by SpenceKonde

MCUdude, 2021-02-08:
> so why would someone want to amplify an analog signal?

There are many reasons why one would like to amplify a signal. But there's so much more to it than just this. You can use them as a balanced line driver for weak signals, like the DAC for instance. Or balanced line receivers. Usually, opamps can be used as comparators with a push-pull output, but they can't have protection diodes between the input, something I don't think the DB's have anyways.

You can for instance use the opamp to drive the MVIO voltage pin, where you can set the opamp voltage using the internal resistor ladder. It doesn't look like you can read the opamp output internally using an ADC, so you'll have to use a physical pin to read the opamp output.

SpenceKonde, 2021-02-08:
do we think the opamp can output enough currrent for the MVIO pin?

and wait, so analogRead() on the output pin doesnt work?

MCUdude, 2021-02-08:
> do we think the opamp can output enough current for the MVIO pin?

well, Microchip suggests this in an application note, but they don't know the current demand for every application. 

> and wait, so analogRead() on the output pin doesn't work?

This will probably work, but you'll have to use the physical pin, it can't be done internally like on some other peripherals like CCL.

SpenceKonde, 2021-02-08:
Okay, and yeah, just checked, that does work. 

i mean if they suggest it, that gives it some serious gravitas, heh...