SpenceKonde/DxCore #21: Make optiboot fit in 512b!

SpenceKonde, 2020-09-10:
It already fits on the 32k parts, and I'm 7 instruction words away from 512b on the 64k ones, and 15 instruction words away on the 128k ones....

MCUdude, 2020-09-10:
Any plans on pushing your AVR DA/DB Optiboot port back to the official Optiboot repo?

SpenceKonde, 2020-09-10:
Probably once it fits in 51w and other things are sorted, first things first

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-10-11:
ff2c8bfab8484c712c6c3b081e8ff624e2661c5b

B A D A B I N G! 

**I tell ya, sometimes I'm so good I scare myself!**

In the worst case (128k flash, serial on alternate pins) we've got a whole 24 bytes of flash left - PLENTY of space, I tell ya!

closed 2020-10-11 by SpenceKonde