SpenceKonde/DxCore #25: broken Images/Links

ProxyPlayerHD, 2020-09-18:
all imagines on this page are broken:
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/MakeUPDIProgrammer.md

and the link to the page "Making a cheap UPDI programmer" is broken on here:
https://github.com/SpenceKonde/DxCore/blob/master/Wiring.md

SpenceKonde, 2020-10-21:
Thanks, updated and fixed.

closed 2020-10-21 by SpenceKonde

ProxyPlayerHD, 2020-10-21:
the first one i mentioned is fixed, but the URL to the "Making a cheap UPDI programmer" page is still broken in the README.md file.

SpenceKonde, 2020-10-23:
Looks fixed now. - not sure if I fixed after you mentioned that, or had fixed in local copy (as I just pushed update that included the readme)

ProxyPlayerHD, 2020-10-23:
hmm strange, it's till broken for me.

I'm on the master branch on this page specifically: https://github.com/SpenceKonde/DxCore