SpenceKonde/DxCore #26: EEPROM library not working

obdevel, 2020-10-09:
Arduino 1.8.13, macOS 10.13.6

AVR128DA28 DIP package w/Optiboot

EEPROM writes not sticking, e.g.
```
EEPROM[34] = 10;
Serial.println(EEPROM[34]); // prints 255
```

I haven't got jtag2updi working so I set the fuses and programmed the bootloader using pyupdi. I based the fuse settings on the avrdude command line displayed by Tools->Burn Bootloader

Everything else seems to be working so far.

SpenceKonde, 2020-10-09:
Known issue need to port fix from megatinycore

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

obdevel, 2020-10-10:
Any pointers to how I might solve this myself ? I don't see any relevant appnotes and the one AS Start example isn't helpful either. I've read the NVMCTRL section of the datasheet but I don't know how to convert this to code.

SpenceKonde, 2020-10-10:
Hold your horses, it's a 5-minute fix from my end...

SpenceKonde, 2020-10-10:
Wait wait wait - the avr-libc routines don't work? Ugh... Okay, that will be a bit longer :-(

SpenceKonde, 2020-10-10:
Need to implement those macros, in that case...

SpenceKonde, 2020-10-10:
What's in there now is definitely totally hosed though

SpenceKonde, 2020-10-10:
I just pushed a version of EEPROM that I would expect to work (in github repo - obv. not in a board manager release). 

Does that not work?

SpenceKonde, 2020-10-10:
If it doesn't work, please describe how it doesn't work...

obdevel, 2020-10-10:
Thanks. I copied that straight over the file in /Users/xxx/Documents/Arduino/hardware/DxCore/megaavr/libraries/EEPROM/src.

Compilation output: [eeprom_test_compile_v1.txt]. I restarted the IDE just to make sure their wasn't anything cached. (https://github.com/SpenceKonde/DxCore/files/5359825/eeprom_test_compile_v1.txt)

Same result - whatever value or address I write reads back as 255, e.g.
```
EEPROM[0] = 22;
Serial.println(EEPROM[0]);
EEPROM.write(1, 22);
Serial.println(EEPROM.read(1));
```
But, if it's calling eeprom_read_byte() surely that's resolved from avr-libc ?

SpenceKonde, 2020-10-10:
Check NVMCTRL.STATUS after attempting a write, see if there are any error flags. 

I suspect read is working correctly, and write is not.

obdevel, 2020-10-10:
After a write, NVMCTRL.STATUS is 16. Per the datasheet (sec 10.5.3), that's the LSB of the error code, which translates to 0x1 INVALIDCMD (The selected command is not supported).

SpenceKonde, 2020-10-10:
Yup, I had a feeling that was what you'd see....

Yeah, eeprom_write_byte needs to be reimplemented....

obdevel, 2020-10-11:
I installed MPLABX and used the ccm to generate some code, but I don't know how to implement this correctly. Maybe this will be a helpful pointer ?

```
/**
 * \brief Read a byte from eeprom
 *
 * \param[in] eeprom_adr The byte-address in eeprom to read from
 *
 * \return The read byte
 */
uint8_t DA_ReadEepromByte(eeprom_adr_t eeprom_adr)
{
		// Read operation will be stalled by hardware if any write is in progress		
		return *(uint8_t *)(EEPROM_START + eeprom_adr);
	
}

/**
 * \brief Write a byte to eeprom
 *
 * \param[in] eeprom_adr The byte-address in eeprom to write to
 * \param[in] data The byte to write
 *
 * \return Status of write operation
 */
nvmctrl_status_t DA_WriteEepromByte(eeprom_adr_t eeprom_adr, uint8_t data)
{
		/* Wait for completion of previous write */
		while (NVMCTRL.STATUS & (NVMCTRL_EEBUSY_bm|NVMCTRL_FBUSY_bm));

		/* Program the EEPROM with desired value(s) */
		ccp_write_spm((void *)&NVMCTRL.CTRLA, NVMCTRL_CMD_EEERWR_gc);

		/* Write byte to EEPROM */
		*(uint8_t *)(EEPROM_START + eeprom_adr) = data;
		
		/* Clear the current command */
		ccp_write_spm((void *)&NVMCTRL.CTRLA, NVMCTRL_CMD_NONE_gc); 

		return NVM_OK;		
}

static inline void ccp_write_spm(void *addr, uint8_t value)
{
	protected_write_io(addr, CCP_SPM_gc, value);
}
```

There is no implementation provided for ccp_write_spm() so I assume it's buried somewhere in some library ?

obdevel, 2020-10-11:
Ok, this seems to work. Again, cobbled together from disparate MCP sources.

```
...
#include <avr/xmega.h>
...

/** Datatype for return status of NVMCTRL operations */
typedef enum {
    NVM_OK    = 0, ///< NVMCTRL free, no write ongoing.
    NVM_ERROR = 1, ///< NVMCTRL operation retsulted in error
    NVM_BUSY  = 2, ///< NVMCTRL busy, write ongoing.
} nvmctrl_status_t;

uint8_t da_eeprom_read_byte(uint8_t * index) {
  return *(uint8_t *)(EEPROM_START + index);
}

nvmctrl_status_t da_eeprom_write_byte(uint8_t *index, uint8_t in) {

  /* Wait for completion of previous write */
  while (NVMCTRL.STATUS & (NVMCTRL_EEBUSY_bm|NVMCTRL_FBUSY_bm));

  /* Program the EEPROM with desired value(s) */
  _PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_EEERWR_gc);
  
  /* Write byte to EEPROM */
  *(uint8_t *)(EEPROM_START + index) = in;
    
  /* Clear the current command */
  _PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_NONE_gc);

  return NVM_OK;    
}
```

SpenceKonde, 2020-10-11:
540351215f239f2aaf62b2a2d891bd13b23a7e5f .... but yeah your approach is better if that works, somehow I thought (looks like incorrectly) that you weren't supposed to change the command while it was busy....

closed 2020-10-11 by SpenceKonde

obdevel, 2020-10-11:
I was hoping you'd know better than me ! As I said, it was hacked together using MCC, a couple of app notes and some googling.

There is one problem I've found ... I can't use the EEPROM object in multiple source files without triggering a linker error. Probably my misunderstanding of the EEPROM.h file or how cores are structured. e.g.

```
/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/arduino_build_784877/libraries/CBUSconfig/CBUSconfig.cpp.o (symbol from plugin): In function `initVariant':
(.text+0x0): multiple definition of `da_eeprom_read_byte(unsigned char*)'
/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/arduino_build_784877/sketch/CBUS_Turntable_Driver_v1.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/arduino_build_784877/libraries/CBUSconfig/CBUSconfig.cpp.o (symbol from plugin): In function `initVariant':
(.text+0x0): multiple definition of `da_eeprom_write_byte(unsigned char*, unsigned char)'
/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/arduino_build_784877/sketch/CBUS_Turntable_Driver_v1.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```
I thought that's what include guards were supposed to prevent ? Probably just needs the file rearranging ?

SpenceKonde, 2020-10-12:
I had a similar problem when I tried to do it in the past defining a macro like that. I was as baffled as you were. That's why I didn't do it like that this time - I'd tripped over it before

obdevel, 2020-10-13:
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/libraries/EEPROM/src/EEPROM.h#L38

Oops :) index needs to be wider than 8 bits to address 512 bytes. Wondered why I was getting strange overwrites.

commit 29af6850 (2020-10-11): This way is better

Fixes #26