SpenceKonde/DxCore #28: [Question] Why have separate manual and boards manager release?

MCUdude, 2020-10-29:
I see you're doing this for this core, megaTinyCore, and ATTinyCore. Why not only do one release, and instead make the release script remove the avr/megaavr folder?

For reference, here's the release script I'm using. All I do is to publish a new release and run the script. It pulls the latest release, removes the avr/megaavr folder, compresses it into an archive again, and appends the json file.

I just had a look at your ReleaseScripts repo, but I couldn't really find the script you're using. Are you doing the boards manager stuff manually?

SpenceKonde, 2020-10-29:
How could it be done without a separate "release", while still having a downloadable "manual" release? There would need to be two separate archives, and they'd need different arrangement of files... I don't know of any way to do that without creating a branch for the board manager release, and doing two separate releases, one from master and one from the board manager branch...

Where is the release script? You referred to one, but didn't include the link.

I do not have a script to generate the released version; for 2.1.4 of megaTinyCore, I finally started using a batch file to do the moving of stuff out of megaavr and deleting the pinout images from the docs, since nobody is going to dig through an installed package to find the docs, and it massively shrinks the size of the download.

MCUdude, 2020-10-29:
> How could it be done without a separate "release", while still having a downloadable "manual" release? There would need to be two separate archives, and they'd need different arrangement of files... [...]

I don't have any other branches than `master` and `gh-pages`. 
I only provide a manual install release on Github, since nobody would use `git clone` to install the latest boards manager package.
Here is what my script does (excluding renaming files):

* ./Boards_manager_release.sh
* Downloads the latest release archive (that includes the megaavr folder)
* Extracts the archive
* Moves the content in the megaavr folder to one level up
* Deletes megaavr
* Compresses it
* Calculates the size and checksum
* Uses [jq](https://stedolan.github.io/jq/) to create a JSON snippet
* Appends it to the existing boards manager JSON file
* Cleanup

All there is left is to push the newly added archive and the updated JSON file, and you're done.
In your case, it would be trivial for the script to remove the `extras` folder to reduce the file size of the compressed folder.

The reason why I created the script in the first place is that before per1234 got busy, he was the one doing all the boards manager release stuff for me. When I had to do it myself, I did it manually and pretty much always screwed up something every time. It got to a point where I hesitated to do a new release because the whole boards manager stuff was such a headache.

> Where is the release script? You referred to one, but didn't include the link.

I was looking in the https://github.com/SpenceKonde/ReleaseScripts repo. I looked through the two scripts you have there, but they only do some file copying. That's why I was assuming you manually handled the size calculation, checksum, and JSON formatting.

SpenceKonde, 2020-10-31:
Woah, crazy - that's wizardry to me... I don't know how to write a script to do any of that except moving the files and compressing it - nor do I understand how to push a binary to github without doing a release?

MCUdude, 2020-10-31:
No, it's not wizardry at all. I've never created anything like this before, I just had an idea. Awk was the only tool I'm a little familiar with that's used in the script. I hadn't even heard of jq before!
An idea, Google and basic programming skills is all you need really.

Everything you'll need is in the script. Google is your friend. Read the comments and tear it apart to see how it works and what it does.

It doesn't include any git magic either. It just pulls the latest release, does some things to it and creates a new archive. The new information is appended to the boards manager Json file. I'll have to add, commit and push the new archive and the json file using git.

FYI I keep all the boards manager stuff in a separate branch. That's where the script is too.

MCUdude, 2020-11-01:
Here's an example that works for megaTinyCore.

First, it expects an existing boards manager file called `package_SpenceKonde_megaTinyCore_index.json`. It can't be empty, but has to look like this in order for `jq` to be able to append to it:

```json
{
  "packages": [
    {
      "name": "megaTinyCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/megaTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [
      ],
      "tools": []
    }
  ]
}

```

And here is the script. It removes the megaavr folder if present (no need for a separate boards manager release anymore), and it removes the extras folder for you.

```bash
#!/bin/bash

##########################################################
##                                                      ##
## Shell script for generating a boards manager release ##
## Created by MCUdude                                   ##
## Requires wget, jq and a bash environment             ##
##                                                      ##
##########################################################

# Change these to match your repo
AUTHOR=SpenceKonde      # Github username
REPOSITORY=megaTinyCore # Github repo name

# Get the download URL for the latest release from Github
DOWNLOAD_URL=$(curl -s https://api.github.com/repos/$AUTHOR/$REPOSITORY/releases/latest | grep "tarball_url" | awk -F\" '{print $4}')

# Download file
wget --no-verbose $DOWNLOAD_URL

# Get filename
DOWNLOADED_FILE=$(echo $DOWNLOAD_URL | awk -F/ '{print $8}')

# Add .tar.bz2 extension to downloaded file
mv $DOWNLOADED_FILE ${DOWNLOADED_FILE}.tar.bz2

# Extract downloaded file and place it in a folder (the #"v" part removes the v in the version number if it is present)
printf "\nExtracting folder ${DOWNLOADED_FILE}.tar.bz2 to $REPOSITORY-${DOWNLOADED_FILE#"v"}\n"
mkdir -p "$REPOSITORY-${DOWNLOADED_FILE#"v"}" && tar -xzf ${DOWNLOADED_FILE}.tar.bz2 -C "$REPOSITORY-${DOWNLOADED_FILE#"v"}" --strip-components=1
printf "Done!\n"

# Move files out of the megaavr folder
mv $REPOSITORY-${DOWNLOADED_FILE#"v"}/megaavr/* $REPOSITORY-${DOWNLOADED_FILE#"v"}

# Delete the extras folder
rm -rf $REPOSITORY-${DOWNLOADED_FILE#"v"}/extras

# Delete downloaded file and empty megaavr folder
rm -rf ${DOWNLOADED_FILE}.tar.bz2 $REPOSITORY-${DOWNLOADED_FILE#"v"}/megaavr

# Compress folder to tar.bz2
printf "\nCompressing folder $REPOSITORY-${DOWNLOADED_FILE#"v"} to $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2\n"
tar -cjSf $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2 $REPOSITORY-${DOWNLOADED_FILE#"v"}
printf "Done!\n"

# Get file size on bytes
FILE_SIZE=$(wc -c "$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2" | awk '{print $1}')

# Get SHA256 hash
SHA256="SHA-256:$(shasum -a 256 "$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2" | awk '{print $1}')"

# Create Github download URL
URL="https://${AUTHOR}.github.io/${REPOSITORY}/$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2"

cp "package_${AUTHOR}_${REPOSITORY}_index.json" "package_${AUTHOR}_${REPOSITORY}_index.json.tmp"

# Add new boards release entry
jq -r                                   \
--arg repository $REPOSITORY            \
--arg version    ${DOWNLOADED_FILE#"v"} \
--arg url        $URL                   \
--arg checksum   $SHA256                \
--arg file_size  $FILE_SIZE             \
--arg file_name  $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2  \
'.packages[].platforms[.packages[].platforms | length] |= . +
{
  "name": $repository,
  "architecture": "megaavr",
  "version": $version,
  "category": "Contributed",
  "url": $url,
  "archiveFileName": $file_name,
  "checksum": $checksum,
  "size": $file_size,
  "boards": [
   {"name": "ATtiny3216/1616/1606/816/806/416/406"},
   {"name": "ATtiny1614/1604/814/804/414/404/214/204"},
   {"name": "ATtiny412/402/212/202"},
   {"name": "ATtiny3217/1617/1607/817/807"}
  ],
  "toolsDependencies": [
    {
      "packager": "arduino",
      "name": "avr-gcc",
      "version": "7.3.0-atmel3.6.1-arduino7"
    },
    {
      "packager": "arduino",
      "name": "avrdude",
      "version": "6.3.0-arduino17"
    },
    {
      "packager": "arduino",
      "name": "arduinoOTA",
      "version": "1.3.0"
    }
  ]
}' "package_${AUTHOR}_${REPOSITORY}_index.json.tmp" > "package_${AUTHOR}_${REPOSITORY}_index.json"

# Remove files that's no longer needed
rm -rf "$REPOSITORY-${DOWNLOADED_FILE#"v"}" "package_${AUTHOR}_${REPOSITORY}_index.json.tmp"

```

Here! Download the zip file, run the Boards_manager_release.sh script and look at the contents in the json file and the newly created archive. It isn't magic, it's just very convenient! You might have to install `jq`: https://stedolan.github.io/jq/
[megaTinyCore_boardsmanager.zip](https://github.com/SpenceKonde/DxCore/files/5470465/megaTinyCore_boardsmanager.zip)

MCUdude, 2020-11-01:
OK, I figured out how it could work with your existing boards manager file.

Take this JSON file for instance:
```json
{
  "packages": [
    {
      "name": "ATTinyCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/ATTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [],
      "tools": []
    },
    {
      "name": "megaTinyCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/megaTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [],
      "tools": []
    },
    {
      "name": "DxCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/megaTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [],
      "tools": []
    }
  ]
}
```

Now the script can figure out what package it's working on, so the output of the script would be this:

```json
{
  "packages": [
    {
      "name": "ATTinyCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/ATTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [],
      "tools": []
    },
    {
      "name": "megaTinyCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/megaTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [
        {
          "name": "megaTinyCore",
          "architecture": "megaavr",
          "version": "2.1.5",
          "category": "Contributed",
          "url": "https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.1.5.tar.bz2",
          "archiveFileName": "megaTinyCore-2.1.5.tar.bz2",
          "checksum": "SHA-256:981496497a81f19fabeec21a0568d11f3de95d4365bc81bd904045a069d33ccc",
          "size": "267785",
          "boards": [
            {
              "name": "ATtiny3216/1616/1606/816/806/416/406"
            },
            {
              "name": "ATtiny1614/1604/814/804/414/404/214/204"
            },
            {
              "name": "ATtiny412/402/212/202"
            },
            {
              "name": "ATtiny3217/1617/1607/817/807"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino7"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
        }
      ],
      "tools": []
    },
    {
      "name": "DxCore",
      "maintainer": "SpenceKonde",
      "websiteURL": "https://github.com/SpenceKonde/megaTinyCore",
      "email": "",
      "help": {
        "online": ""
      },
      "platforms": [],
      "tools": []
    }
  ]
}

```

All you have to do is to modify the scrip I posted eaerlier and replace
```
'.packages[].platforms[.packages[].platforms | length] |= . +
```

with this
```
'(.packages[] | select(.name==$repository)).platforms[.packages[].platforms | length] |= . +
```

MCUdude, 2020-11-15:
I'm a little surprised you haven't responded to this. You don't need all these branches anymore, you can still have one JSON file for all your cores and it isn't magic _at all_.

The only thing you'll have to do it set up github gh-pages. Create a branch called `gh-pages`, and select this to be your "pages" branch under the repo settings. The idea is that the generated archive is stored the gh-pages branch, so the URL becomes this for instance: `https://SpenceKonde.github.io/ATTinyCore/ATTinyCore-1.4.1.tar.bz2`.

MegaTinyCore script (no modifications needed) <details>
#!/bin/bash

##########################################################
##                                                      ##
## Shell script for generating a boards manager release ##
## Created by MCUdude                                   ##
## Requires wget, jq and a bash environment             ##
##                                                      ##
##########################################################

# Change these to match your repo
AUTHOR=SpenceKonde      # Github username
REPOSITORY=megaTinyCore # Github repo name

# Get the download URL for the latest release from Github
DOWNLOAD_URL=$(curl -s https://api.github.com/repos/$AUTHOR/$REPOSITORY/releases/latest | grep "tarball_url" | awk -F\" '{print $4}')

# Download file
wget --no-verbose $DOWNLOAD_URL

# Get filename
DOWNLOADED_FILE=$(echo $DOWNLOAD_URL | awk -F/ '{print $8}')

# Add .tar.bz2 extension to downloaded file
mv $DOWNLOADED_FILE ${DOWNLOADED_FILE}.tar.bz2

# Extract downloaded file and place it in a folder (the #"v"} part removes the v in the version number if it is present)
printf "\nExtracting folder ${DOWNLOADED_FILE}.tar.bz2 to $REPOSITORY-${DOWNLOADED_FILE#"v"}\n"
mkdir -p "$REPOSITORY-${DOWNLOADED_FILE#"v"}" && tar -xzf ${DOWNLOADED_FILE}.tar.bz2 -C "$REPOSITORY-${DOWNLOADED_FILE#"v"}" --strip-components=1
printf "Done!\n"

# Move files out of the megaavr folder
mv $REPOSITORY-${DOWNLOADED_FILE#"v"}/megaavr/* $REPOSITORY-${DOWNLOADED_FILE#"v"}

# Delete the extras folder
rm -rf $REPOSITORY-${DOWNLOADED_FILE#"v"}/extras

# Delete downloaded file and empty megaavr folder
rm -rf ${DOWNLOADED_FILE}.tar.bz2 $REPOSITORY-${DOWNLOADED_FILE#"v"}/megaavr

# Compress folder to tar.bz2
printf "\nCompressing folder $REPOSITORY-${DOWNLOADED_FILE#"v"} to $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2\n"
tar -cjSf $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2 $REPOSITORY-${DOWNLOADED_FILE#"v"}
printf "Done!\n"

# Get file size on bytes
FILE_SIZE=$(wc -c "$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2" | awk '{print $1}')

# Get SHA256 hash
SHA256="SHA-256:$(shasum -a 256 "$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2" | awk '{print $1}')"

# Create Github download URL
URL="https://${AUTHOR}.github.io/${REPOSITORY}/$REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2"

cp "package_${AUTHOR}_${REPOSITORY}_index.json" "package_${AUTHOR}_${REPOSITORY}_index.json.tmp"

# Add new boards release entry
jq -r                                   \
--arg repository $REPOSITORY            \
--arg version    ${DOWNLOADED_FILE#"v"} \
--arg url        $URL                   \
--arg checksum   $SHA256                \
--arg file_size  $FILE_SIZE             \
--arg file_name  $REPOSITORY-${DOWNLOADED_FILE#"v"}.tar.bz2  \
'(.packages[] | select(.name==$repository)).platforms[(.packages[] | select(.name==$repository)).platforms | length] |= . +
[... 136 lines cut ...]
  ],
  "toolsDependencies": [
    {
      "packager": "arduino",
      "name": "avr-gcc",
      "version": "7.3.0-atmel3.6.1-arduino7"
    },
    {
      "packager": "arduino",
      "name": "avrdude",
      "version": "6.3.0-arduino17"
    },
    {
      "packager": "ATTinyCore",
      "name": "micronucleus",
      "version": "2.0a4"
    }
  ]
}' "package_${AUTHOR}_${REPOSITORY}_index.json.tmp" > "package_${AUTHOR}_${REPOSITORY}_index.json"

# Remove files that's no longer needed
rm -rf "$REPOSITORY-${DOWNLOADED_FILE#"v"}" "package_${AUTHOR}_${REPOSITORY}_index.json.tmp"
</details>

The script for DxCore should be very simple to create. Just duplicate the megaTinyCore script and replace the content inside the `jq` command

SpenceKonde, 2020-11-15:
> github.io/${REPOSITORY}/$REPOSITORY-${DOWNLOADED_FILE#
> ATTinyCore #!/bin/bash

Hadn't seen it, gotta finish the fix for servo before I take a look at
this,... I'm almost there, and like - my servo implementation did *NOTHING*
that in *ANY WAY* resembles what it should...

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

SpenceKonde, 2021-01-10:
Woah, this *definitely* didn't register the first time around! I totally forgot that you'd actually given me code here, and I attempted to hack together something with a batch file for 2.2.x of megaTinyCore - but of course, that didn't work out; the line endings got mangled resulting in megaTinyCore 2.2.x not working on Linux platforms.

I'm looking into this now - I've sort of forced the issue on myself with my bungled attempt with 2.2.x of megaTinyCore. The awkward bit is that I don't have a Linux machine at home to run this on, so I'm going to be running it on a system in the cloud...

MCUdude, 2021-01-10:
For everything to work as smooth as possible, you should enable Github pages for the repo you want to use, create a branch called `gh-pages` and just mimic what I've done.

The script isn't bullet proof, but I've been using it for a while now on all my cores, and it has never failed. I've never had to modify a JSON file ever since. But download a local copy of your JSON script and experiment with the script so you can get familiar with it.

> I'm looking into this now - I've sort of forced the issue on myself with my bungled attempt with 2.2.x of megaTinyCore. The awkward bit is that I don't have a Linux machine at home to run this on, so I'm going to be running it on a system in the cloud...

You don't need a Linux computer, you can just use Ubuntu for Windows! I'm using it on my work computer all the time. It's simply a matter of enabling it. It's a full blown UNIX terminal, just what you need!

SpenceKonde, 2021-01-11:
Wow. Yeah, this seems to work quite well. Much thanks for this - sorry it was left sitting for so long; I'm many weeks/months behind on these cores...

closed 2021-01-11 by SpenceKonde