SpenceKonde/DxCore #30: Use DxCore with AVR128DA48 Curiosity Nano Evaluation Kit?

technoblogy, 2020-11-05:
Is it possible to use the DxCore with the AVR128DA48 Curiosity Nano Evaluation Kit?

https://www2.mouser.com/ProductDetail/Microchip-Technology/DM164151

I tried uploading Blink, after selecting **Atmel nEDBG (ATSAMD21E18/Curiosity)** from the **Programmer** menu, but get:

````
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
````

SpenceKonde, 2020-11-05:
No, avrdude does not support the required addressing mode. Am investigating other methods and will make more information available when it becomes available to me.

closed 2020-11-05 by SpenceKonde

technoblogy, 2020-11-05:
Thanks! Perhaps add a note to your documentation to stop other people asking the same question.

SpenceKonde, 2020-11-05:
Clearly I need to do something like that as I just closed another issue asking the exact same thing a few days ago.

technoblogy, 2020-11-05:
Sorry, I should have searched for closed issues.

technoblogy, 2020-11-05:
Perhaps move your new section under:

https://github.com/SpenceKonde/DxCore#supported-parts-click-link-for-pinout-diagram-and-details

It's a bit over prominent where it is!

MCUdude, 2020-11-05:
More info:
https://www.avrfreaks.net/forum/avr128da48-curiosity-nano-and-avrdude

SpenceKonde, 2020-11-05:
That is the thread that convinced me that this won't be able to work with avrdude.

technoblogy, 2020-11-05:
I've managed to program my AVR128DA48 Curiosity Nano Evaluation Kit using your DxCore by connecting an Arduino Nano-based UPDI programmer to the VTG (5V), GND (GND), and DBG0 (UPDI) pins on the connectors next to the USB connector on the Curiosity Nano (making sure to leave the USB unconnected).

Having done that, I could use the Curiosity Nano USB connector to communicate with Serial1 via the USB port.

I'll write up the full procedure over the next day or so.

SpenceKonde, 2020-11-05:
Would you guys please show a tiny bit of patience? I have a Zoom call tomorrow morning with some people to sort out plans for the new tooling which will allow programming through the USB port.

sstaub, 2020-11-05:
@technoblogy it is possible to program the curiosity board simply by moving the .hex file to the curiosity file system by drag and drop.

MCUdude, 2020-11-05:
> Would you guys please show a tiny bit of patience?

Aren't we allowed to discuss the topic even though you have plans to get USB support working?

> @technoblogy it is possible to program the curiosity board simply by moving the .hex file to the curiosity file system by drag and drop.

That's true, but sadly it's not as reliable

SpenceKonde, 2020-11-05:
Sure - but do you really want to spend time documenting a workaround that may be obsoleted as soon as this weekend?

technoblogy, 2020-11-06:
@SpenceKonde I didn't realise that there was any hope of fixing this - I'll be happy not to document my workaround!

SpenceKonde, 2020-11-07:
Okay, we actually only have one option, I had a brain fart when I said we had two ways - we have two ways to get programming using just a serial adapter and a resistor (pyupdi and pymcuprog), but only one way to get programming with the on-board programmer to work, and that's pymcuprog. But we're going to use pymcuprog for both. I got binaries generated for me, but the wrong version of the source was used (code we thought was newer, but was actually older and with a critical bug - though it wouldnt effect this, I don't have a curiosiry nano to test with....). I'm not sure whether he'll take another crack at it before monday,

technoblogy, 2020-11-07:
Thanks! No hurry.