SpenceKonde/DxCore #31: Some documentation suggestions

technoblogy, 2020-11-05:
Thanks for providing this core! I've just successfully uploaded Blink to an AVR128DB28 on a breadboard.

Couple of comments:

1. Using a Nano-based UPDI programmer I previously used to program 0-series and 1-series chips, I got the error:

       avrdude: jtagmkII_paged_write(): timeout/error communicating with programmer (status -1)

Thanks to the issue by @DustinWatts I realised I need to update jtag2updi, and it then worked. Perhaps mention this?

2. The link **Making a cheap UPDI programmer** in megaTinyCore (now) gives a 404.

technoblogy, 2020-11-19:
3. Another one; in the section:

https://github.com/SpenceKonde/DxCore#dac-support

shouldn't the sentence "This voltage must be lower than Vcc to get the correct voltages." be removed?

SpenceKonde, 2020-11-22:
Thanks, I've fixed these issues, please let me know if you find any more! (the DAC bit was supposed to indicate that the reference voltage must be lower than supply voltage. Somewhere I even have the number on how much lower, but didn't feel like digging it out.

closed 2020-11-22 by SpenceKonde

technoblogy, 2020-11-22:
> the DAC bit was supposed to indicate that the reference voltage must be lower than supply voltage

But I thought you could give VDD as the reference voltage?

technoblogy, 2020-11-26:
In:

https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/DirectPortManipulation.md

the lines:

    PORTA.OUT |= PIN4_bm; // write PB4 high
    PORTA.OUT &=~PIN4_bm; // write PB4 low

should be:

    PORTA.OUT |= PIN4_bm; // write PA4 high
    PORTA.OUT &=~PIN4_bm; // write PA4 low