SpenceKonde/DxCore #34: EEPROM addressing goes wrong above 256 bytes

technoblogy, 2020-11-22:
The following demo program illustrates this. It should write into the 512 bytes of EEPROM on an AVR128DA48:

    0 0 1 1 2 2 3 3 4 4 ..... 255 255

Program:
````
#include <EEPROM.h>

void setup() {
  Serial1.begin(9600);
}

void loop() {
  for (int i=0; i<512; i++) EEPROM.write(i, i/2);
  Serial1.println("Read");
  delay(1000);
  for (int i=0; i<512; i++) { Serial1.print(EEPROM.read(i)); Serial1.print(' '); }
  Serial1.println();
  delay(1000);
}
````
Output:

Read
128 128 129 129 130 130 131 131 132 132 133 133 134 134 135 135 136 136 137 137 138 138 139 139 140 140 141 141 142 142 143 143 144 144 145 145 146 146 147 147 148 148 149 149 150 150 151 151 152 152 153 153 154 154 155 155 156 156 157 157 158 158 159 159 160 160 161 161 162 162 163 163 164 164 165 165 166 166 167 167 168 168 169 169 170 170 171 171 172 172 173 173 174 174 175 175 176 176 177 177 178 178 179 179 180 180 181 181 182 182 183 183 184 184 185 185 186 186 187 187 188 188 189 189 190 190 191 191 192 192 193 193 194 194 195 195 196 196 197 197 198 198 199 199 200 200 201 201 202 202 203 203 204 204 205 205 206 206 207 207 208 208 209 209 210 210 211 211 212 212 213 213 214 214 215 215 216 216 217 217 218 218 219 219 220 220 221 221 222 222 223 223 224 224 225 225 226 226 227 227 228 228 229 229 230 230 231 231 232 232 233 233 234 234 235 235 236 236 237 237 238 238 239 239 240 240 241 241 242 242 243 243 244 244 245 245 246 246 247 247 248 248 249 249 250 250 251 251 252 252 253 253 254 254 255 255 128 128 129 129 130 130 131 131 132 132 133 133 134 134 135 135 136 136 137 137 138 138 139 139 140 140 141 141 142 142 143 143 144 144 145 145 146 146 147 147 148 148 149 149 150 150 151 151 152 152 153 153 154 154 155 155 156 156 157 157 158 158 159 159 160 160 161 161 162 162 163 163 164 164 165 165 166 166 167 167 168 168 169 169 170 170 171 171 172 172 173 173 174 174 175 175 176 176 177 177 178 178 179 179 180 180 181 181 182 182 183 183 184 184 185 185 186 186 187 187 188 188 189 189 190 190 191 191 192 192 193 193 194 194 195 195 196 196 197 197 198 198 199 199 200 200 201 201 202 202 203 203 204 204 205 205 206 206 207 207 208 208 209 209 210 210 211 211 212 212 213 213 214 214 215 215 216 216 217 217 218 218 219 219 220 220 221 221 222 222 223 223 224 224 225 225 226 226 227 227 228 228 229 229 230 230 231 231 232 232 233 233 234 234 235 235 236 236 237 237 238 238 239 239 240 240 241 241 242 242 243 243 244 244 245 245 246 246 247 247 248 248 249 249 250 250 251 251 252 252 253 253 254 254 255 255 
Read

SpenceKonde, 2020-11-22:
yup, Looks to me like it wraps around at 255

technoblogy, 2020-11-22:
Wasn't a problem with the new ATtinys, none of which had more than 256 bytes EEPROM.

Floessie, 2020-11-23:
`idx` is `uint8_t` throughout the [code](https://github.com/SpenceKonde/DxCore/blob/master/megaavr/libraries/EEPROM/src/EEPROM.h#L179).

Floessie, 2020-11-23:
@SpenceKonde I just wanted to let you know, that I got the DB running with your `chmod`ded 1.2.0-a and ElTangas `jtag2updi`. As with the t1614 I didn't manage to upload anything with your `jtag2updi` fork on an Arduino Nano. It just blinks the builtin LED but won't sync. Nevertheless, I am completely content with the current situation. :+1:

Floessie, 2020-11-25:
@SpenceKonde I've already patched it. I'll give you a PR in a few minutes.

closed 2020-11-30 by SpenceKonde