SpenceKonde/DxCore #38: Add support for serial adapter updi programming pyupdi-style and Curiosity Nano.

SpenceKonde, 2020-11-22:
pymcuprog support is going into 2.2.0 of megaTinyCore; pulling that into here will have enormous benefits as it will allow USB programming of curiosity nano parts that require 24-bit addressing.

MCUdude, 2020-12-15:
I'll just leave this here:
https://savannah.nongnu.org/patch/?10000

Pretty much my dreams come true!

MCUdude, 2020-12-17:
It has already been merged into Arduino's Avrdude distribution!
https://github.com/facchinm/avrdude/pull/11

I haven't tried with AVR DA's yet, but I can confirm that the MPLAB SNAP programmer works. I'll try with my Curiosity 128DA48 later

MCUdude, 2020-12-17:
I just couldn't wait. The Curiosity 128DA48 works flawlessly with the new Avrdude version. No need to bundle pymcuprog anymore!

SpenceKonde, 2020-12-17:
No, we do still need pymcuprog for programming with just serial adapter and 4.7k resistor - but as it happens, getting the other programmers working looked like it would be much harder.

MCUdude, 2020-12-17:
> No, we do still need pymcuprog for programming with just serial adapter and 4.7k resistor

I thought it was pyupdi that did this, not pymcuprog?

From the pymcuprog Pypi website:
> The following Atmel/Microchip debuggers are supported:
JTAGICE3 (only firmware version 3.x)
Atmel-ICE
Power Debugger
EDBG
mEDBG
nEDBG / PKoB nano
MPLAB PICkit 4 ICD (only when in 'AVR mode')
MPLAB Snap ICD (only when in 'AVR mode') Although not all functionality is provided on all boards. See device support section below.

But yes, I can imagine that bundling pyupdi so it works on all OSes is more complicated than just referring to an Avrdude version in the boards manager file.

SpenceKonde, 2020-12-17:
It's also got the pyupdi functionality in there.

We're pretty much there at this point - it was support for everything other than the pyupdi-style programming that was going to be much harder.

MCUdude, 2020-12-17:
But why bother with pymcuprog when Avrdude will cover everything except the pyupdi part? Why not just bundle pyupdi instead? Unlike pymcuprog, pyupdi open-source, which is a big deal.

SpenceKonde, 2020-12-28:
Because the developer of pyupdi and two Microchip employees at the meeting about this in November were vehement that pymcuprog should be used instead of pyupdi :-P

MCUdude, 2020-12-28:
> Because the developer of pyupdi and two Microchip employees at the meeting about this in November were vehement that pymcuprog should be used instead of pyupdi :-P

But closed-source is closed-source, regardless of what the Microchip developers feel. Personally I'm not going to use pymcuprog until it goes open. Hopefully, it will happen soon.

> We plan to put it out early next year (some QA first).
https://github.com/mraardvark/pymcuprog/issues/1

When its source code is available, I'll be mention pymcuprog as a viable alternative to pyupdi and Avrdude in the [MegaCoreX PlatformIO readme](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md). I don't think I'll bother trying to get pyupdi working with Arduino IDE. PlatformIO is so brilliant these days anyways.

closed 2021-01-12 by SpenceKonde