SpenceKonde/DxCore #44: No pinout for AVR DB series

Kozma04, 2020-11-29:
As the question states, there is no pinout image for any MCU from the DB-series.

I am planning to migrate from AVR128DA64 to AVR128DB64 in my project so as a second question, how does the pinout and peripheral pins differ from the DA-series and DB-series MCUs I mentioned?

Thank you.

MCUdude, 2020-11-29:
I'm the one that has created all the pinout pics for the AVR DA series, so I guess it's my job to make the AVR DB ones too. But you got all information you need to answer your own question. You have the AVR DB datasheet where all the peripherals are mapped to the GPIOs in a nice table, and if you need to refer to Arduino pins, you can use the PIN macros: `digitalWrite(PIN_PA7, HIGH);`

MCUdude, 2020-11-29:
I haven't had this confirmed by @SpenceKonde himself, but I'm pretty sure this one is correct. Very, very similar to the DA series:

[image]

SpenceKonde, 2020-11-30:
Oh, shit, I did forget to get back to you on that didn't I? 

Well, they have since admitted that the one errata that would have tempted me to do things differently were it fixed (TCD0 hosed except on PORTA) is present on the DB-series too, so...

The MVIO supply pin, they call it "VDDIO2" 
And ugh, Oh yeah, this was why I initially deferred responding, because I was unhappy about the box I was being forced into on PORTA. 

I guess we move the two TCD0 PWM pins to PA6 and PA7, because otherwise for anyone using a crystal, that's going to be their serial monitor... Uuuuugh... on my next rev of breakout boards, I guess I will be defaulting to the serial header being on those pins with a pair of solder bridge jumpers you can change to connect PA0, PA1 instead, I guess).

Regarding the TCD PWM pins, part of me _really_ wants to make it so it gives you whichever two pins out of the 4 that you write to first >.> Would even provide a graceful way out of the inevitable "okay, great they fixed it, now how to I expose that when there's still a ton of busted hardware out there" situation if it were extended to let you do it for the port too... (well, I guess they could just leave it broken like they did with all the stuff on the 0/1-series)...

SpenceKonde, 2020-11-30:
(but to be clear, I won't have time to do what I dream of with TCD0 any time soon - wouldn't that be neat though?)

Kozma04, 2020-11-30:
Thanks for the pinout, but I recently checked the datasheet and shouldn't TX/RX functionalities from pins PE0 and PE1 be reversed? Same thing for the DA series pinout.

SpenceKonde, 2020-11-30:
Yup, never noticed that, heh.... Yeah, TX always on pins 0, alt 4, RX on pins 1, alt 5, for DA and DB

For the ones coming down the pike after that though - it looks like the dart-throwing monkeys from the Atmel days managed get their jobs back >.>

MCUdude, 2020-11-30:
> Thanks for the pinout, but I recently checked the datasheet, and shouldn't TX/RX functionalities from pins PE0 and PE1 be reversed? Same thing for the DA series pinout.

Thanks! That has just slipped through...

@SpenceKonde can you tell me exactly which pins to label as TCD0 outputs? And is there anything else that's wrong/missing apart from the VDDIO2 label?

SpenceKonde, 2020-12-10:
FlexTCD will not be implemented in the near term.... soooo TCD0 on PA6. PA7 for the forseable future. I can MS-paint move around the TCD0 labels though, if worse comes to work.

SpenceKonde, 2020-12-22:
Yeah, TCD0 on PA6, PA7, amd the correction on the PE TX/RX serial pins, and the MVIO supply pin named VDDIO2 and this should be good to go, thanks :-)

MCUdude, 2020-12-29:
[AVR-DB pinouts.zip](https://github.com/SpenceKonde/DxCore/files/5752290/AVR-DB.pinouts.zip)

Please look through all of them to make sure there are no errors in there.

SpenceKonde, 2021-01-09:
Thanks - my initial impression is that these look good; will do another pass as I get to come back to the cores after finishing work for a client today, but they are now checked in.

closed 2021-01-09 by SpenceKonde

killerkalamari, 2021-02-27:
I notice that some of the pins have dots and some have arrows. What is the significance of an arrow vs a dot?

SpenceKonde, 2021-02-28:
@killerkalamari The Arrows, you'll note, are going into pins 2 and 6 within each port; those are the "fully asynchronous" pins, which are able to wake the chip from sleep modes during which the main oscillator has been stopped (ie, anything that actually saves enough power to bothet with). Also on other pins in order to be confident that you will detect a very brief pulkse on the pin, you need to meet the nyquist criteria, you need the pulse to last longer than a clock cycle. But when the fully async pins are configured as interrupt sources they can recognize detect significantly shorter pulses. This does also mean that, when used as interrupt sources (which is where tthis behavior is relevant), they are also more sensitive to noise, on "Rising" and "falling" interrupts (instead of just "change" or "low level"  in sl

SpenceKonde, 2021-02-28:
See section 18.3.3 of the AVR128DB datasheet or section 17.3.3 of the AVR128DA datasheet for more information. All modern AVRs are like that, by the way. I thin kthe interrupts can also retrigger more frequently

killerkalamari, 2021-02-28:
Thanks for the great explanation! Possible errata (or Tindie spoiler?) the AVR128DB28 pinout has an "LED" marking on pin 1.

SpenceKonde, 2021-02-28:
Physical pin 1? That is, arduino/logical pin 7, PIN_PA7? That is the pin
set as LED_BUILTIN, I also put the led there on my boards. Lets you use
sketches that assume it is defined (ex, blink) without modification, and I
use it for compile testing. I mean, you dont want ti waste an mvio pin on
portc with an led, nor an analog pin on portd, not with all them opamps on
them. And after that you are down to just the two pins on portf whoch are
the only ones that can take a watch crystal. So its gotta be porta right?
Cant be 0 or 1 - that is the crystal if using one, probably Serial if not.
Or 4 or 5, which are Serial alt pins, and that's where you put serial if
using an HF crystal. 4/5/6 is SPI data lines anyway, not great thing to
have an led dangling off of, and 2/3 is i2c which most certainly cant have
an led! So, pa7 it was! And kept it on the same pin between different
pincount versions for consistency (and so the same bootloader binaries,
which assume pa7 for the led, work on all. I do the same thing for
megatinycore (pa7 too, as luck would have it, except on 8 on parts), and
recently realized that the bootloaders I had been distributing for 14, 20,
and 24 pin parts were binary identical - only difference was the name! (Now
I only distribute one copy of it :-P)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com