SpenceKonde/DxCore #45: Default state of "MVIO on PORTC" menu option

technoblogy, 2020-12-01:
The initial state of the **MVIO on PORTC** menu option seems to be **Enabled**.

Wouldn't it be more logical to swap the order of the options, and make the default state **Disabled**, as this is the setting without an explicit fuse change?

SpenceKonde, 2020-12-10:
8.8.2.5 System Configuration 1
Name: SYSCFG1
Offset: 0x06
Default: 0x08

(snip)
Bits 4:3 MVSYSCFG: MVIO System Configuration
0x0 Reserved
0x1 DUAL
0x2 SINGLE
0x3 Reserved

Ergo, default configuration is dual supply. If your virgin chips are not set like that, that is a defect in the factory programming (ie, yet another errata that isn't listed in the official docs). If you think about this for a moment, that's actually the only way it could be, because the parts are designed to be able to be programmed in system, ie, boards assembled with virgin chips and then programmed as an assembled board. So the stock configuration has to be both "safe" and functional in all configurations (the same reason classic AVRs start up at 1 MHz, and all AVRs come stock with BOD disabled - they have to work at the absolute lowest supported voltage). Since with MVIO disabled, Vdd<sub>io</sub> is internally connected to Vdd, it **must** be set to dual supply on a virgin chip, since a supported configuration could apply different voltage to Vdd<sub>io</sub> and Vdd.

closed 2020-12-10 by SpenceKonde

technoblogy, 2020-12-11:
Thanks - you're correct, and the FUSE section says that dual-supply is the default.

I was misled by the following text on page 199: