SpenceKonde/DxCore #46: Undefined reference to `ccp_write_io()`

Floessie, 2020-12-10:
Hi DrAzzy,

I'm using your DxCore 1.2.0-a under Linux with an AVR128DB28. When trying to use the `ccp_write_io()` function from `<avr/cpufunc.h>` I get a linker error.

I'm looking for a workaround right now. Just wanted to get you informed...

Best,
Flössie

Floessie, 2020-12-10:
Workaround: using `CCPWrite` from [here](https://github.com/synthetos/Motate/blob/master/MotateProject/motate/platform/atmel_xmega/xmega.cpp).

SpenceKonde, 2020-12-10:
Why not use the compiler provided builtin, _PROTECTED_WRITE()? That is what DxCore uses for setting protected registers...Look

SpenceKonde, 2020-12-10:
Except that it has an unnecessary disabling of interrupts (when CCP is written, interrupts are automatically disabled - see datasheet; that implementation disables them around the call to CCP, which just wastes uh... 6 bytes of flash and 3 CPU clocks?

SpenceKonde, 2020-12-10:
(it = CCPWrite() - _PROTECTED_WRITE() does it correctly.... )

Though, I'm surprised that cpufunc has changed? I don't suppose you could post the exact linker error?

Floessie, 2020-12-10:
@SpenceKonde Didn't know that macro. Thanks for the pointer!

> Though, I'm surprised that cpufunc has changed? I don't suppose you could post the exact linker error?

I'll prepare a minimal example...

Floessie, 2020-12-10:
```c++
#include <avr/cpufunc.h>

void setup() {
   uint8_t temp;
    
    temp = CLKCTRL.OSC32KCTRLA;
    temp |= CLKCTRL_RUNSTDBY_bm;
    ccp_write_io((uint8_t*)&CLKCTRL.OSC32KCTRLA, temp);
}

void loop() {
}
```

```
Arduino: 1.8.13 (Linux), Board: "AVR DB-series (no bootloader), AVR128DB28, 24 MHz internal, No, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2, Enabled, 8ms"

/opt/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/user/.arduino15/packages -hardware /home/user/src/Arduino/hardware -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/user/src/Arduino/libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8 -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_302614 -warnings=default -build-cache /tmp/arduino_cache_613580 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/user/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/user/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azzy1.path=/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1 -prefs=runtime.tools.arduinoOTA.path=/home/user/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/user/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /tmp/arduino_modified_sketch_9454/sketch_dec10a.ino
/opt/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/user/.arduino15/packages -hardware /home/user/src/Arduino/hardware -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/user/src/Arduino/libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8 -vid-pid=1A86_7523 -ide-version=10813 -build-path /tmp/arduino_build_302614 -warnings=default -build-cache /tmp/arduino_cache_613580 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/user/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/user/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azzy1.path=/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1 -prefs=runtime.tools.arduinoOTA.path=/home/user/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/user/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /tmp/arduino_modified_sketch_9454/sketch_dec10a.ino
Using board 'avrdb' from platform in folder: /home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a
Using core 'dxcore' from platform in folder: /home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a
Detecting libraries used...
/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR -DDXCORE=1.1.1 -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=2UL -DDXCORE_PATCH=0UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/cores/dxcore/api/deprecated -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/cores/dxcore -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/variants/28pin-standard /tmp/arduino_build_302614/sketch/sketch_dec10a.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/home/user/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azzy1/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR -DDXCORE=1.1.1 -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=2UL -DDXCORE_PATCH=0UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/cores/dxcore/api/deprecated -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/cores/dxcore -I/home/user/.arduino15/packages/DxCore/hardware/megaavr/1.2.0-a/variants/28pin-standard /tmp/arduino_build_302614/sketch/sketch_dec10a.ino.cpp -o /tmp/arduino_build_302614/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
[... cut ...] TION2=0x10000,--section-start=.FLMAP_SECTION3=0x18000 -mmcu=avr128db28 -o /tmp/arduino_build_302614/sketch_dec10a.ino.elf /tmp/arduino_build_302614/sketch/sketch_dec10a.ino.cpp.o /tmp/arduino_build_302614/../arduino_cache_613580/core/core_e338eab05b22211458d2037f06274d6b.a -L/tmp/arduino_build_302614 -lm
/tmp/ccagVgyB.ltrans0.ltrans.o: In function `setup':
/tmp/arduino_modified_sketch_9454/sketch_dec10a.ino:8: undefined reference to `ccp_write_io(unsigned char*, unsigned char)'
exit status 1
```

Floessie, 2020-12-10:
`_PROTECTED_WRITE` did the trick. Thanks, again!

As I'm playing with the PIT at the moment, do you know a way to calibrate the internal 32k osc? It's a tad too slow. From the datasheet I see that there's `RTC.CALIB`, but it's only available for the external 32k crystal...

SpenceKonde, 2020-12-10:
Don't think you can; that is *not* what it is meant for! It's meant to be a "save as much power as you can by sacrificing accuracy" option... If you want accuracy, they would tell you to use a watch crystal or 32kHz active oscillator.

It's also fairly strongly dependent temperature (used to be dependent on voltage too... but they run the internal oscillators on the Dx-series from Vdd<sub>core</sub> which is regulated down from Vdd, and varies little - if anything - over operating voltage range.... That's why they are spec'ed for the full 4-24 MHz over the full voltage range; Semiconductor performance = f(p, V, T) - process (manufacturing process, including design, etc), voltage (positive correlation) and temperature (negative correlation) 

Actually, now that would be an interesting experiment... monitor the frequency of PWM, on a part with BOD disabled while you sloooooowly lower the voltage. At some Vdd<sub>low</sub> suddenly the frequency will start to change with voltage. This Vdd<sub>low</sub> = Vdd<sub>core</sub>+V<sub>dropout</sub> of their core voltage regulator, wouldn't it?

SpenceKonde, 2020-12-10:
By the way, if you do anything spiffy with the PIT, I'd love to hear about it. I just enabled discussions here, so there's a mini-forum attached to DxCore repo now :-) 
(that scream you heard in the distance was gitter, the github-repo-focused discussion forum site :-P )

closed 2020-12-10 by SpenceKonde

Floessie, 2020-12-11:
Sure. I just thought there was something similar to the `SIGROW.TEMPSENSE[01]` that I missed...

Nothing spiffy: I'll simply use it as the time source for FreeRTOS on megaavr like we do with the WDT on avr.