SpenceKonde/DxCore #51: Error compiling for board AVR DB-series

jamesy0ung, 2020-12-22:
I get this error with optiboot and without optiboot. Using standard libraries and latest IDE

jamesy0ung, 2020-12-22:
It also says explicitly in the readme that it is supported, but the hyperlink leads to a 404 (https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/DB28.md)

SpenceKonde, 2020-12-22:
You haven't included the error, but my guess is that you haven't installed the -a version via board manager (windows/macos only) which included the updated compiler toolchain for the DB-series parts. I think I have a fix for the problem that kept linux toolchains from working, but haven't had a chance to implement it.

SpenceKonde, 2020-12-22:
Missing documentation is known issue, gated by #44

jamesy0ung, 2020-12-24:
Nope I downloaded the latest version from this repository - 1.2.0
Here's the error 
`
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/james/Library/Arduino15/packages -hardware /Users/james/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/james/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/james/Documents/Arduino/libraries -fqbn=DxCore-1.2.0_manual:megaavr:avrdbopti:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8,timeout=1sec,bootloaderusart=ser0 -ide-version=10813 -build-path /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493 -warnings=none -build-cache /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_cache_411941 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/james/Library/Arduino15/packages -hardware /Users/james/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/james/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/james/Documents/Arduino/libraries -fqbn=DxCore-1.2.0_manual:megaavr:avrdbopti:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8,timeout=1sec,bootloaderusart=ser0 -ide-version=10813 -build-path /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493 -warnings=none -build-cache /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_cache_411941 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'avrdbopti' from platform in folder: /Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr
Using core 'dxcore' from platform in folder: /Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR -DDXCORE=1.1.1 -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=2UL -DDXCORE_PATCH=0UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/cores/dxcore/api/deprecated -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/cores/dxcore -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/variants/28pin-standard /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Error while detecting libraries included by /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493/sketch/Blink.ino.cpp
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR -DDXCORE=1.1.1 -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=2UL -DDXCORE_PATCH=0UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/cores/dxcore/api/deprecated -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/cores/dxcore -I/Users/james/Documents/Arduino/hardware/DxCore-1.2.0_manual/megaavr/variants/28pin-standard /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493/sketch/Blink.ino.cpp -o /var/folders/w3/m4nllj792bxg538hlvt4g1fm0000gn/T/arduino_build_446493/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
avr-g++: error: device-specs/specs-avr128db28: No such file or directory
exit status 1
Error compiling for board AVR DB-series (Optiboot).`

SpenceKonde, 2020-12-24:
Yes, this is expected, it is not supported with manual installations because the IDE does not include a compiler toolchain that supports this part. You must install 1.2.0-a (not supported for Linux) using board manager. Have been trying to get the Arduino team to roll a new official toolchain that will actually work for months now.

closed 2020-12-24 by SpenceKonde