SpenceKonde/DxCore #53: Error in 1.30 manual install

obdevel, 2021-01-03:
Just upgrading my install from some (working) previous version. Deleted existing under <sketchbook>/hardware and copied over files from downloaded 1.3.0 DxCore-master.zip.

On opening boards manager to check:
```
Could not find boards.txt in /Users/xx/Documents/Arduino/hardware/megaavr/libraries. Is it pre-1.5?
Could not find boards.txt in /Users/xx/Documents/Arduino/hardware/megaavr/bootloaders. Is it pre-1.5?
Could not find boards.txt in /Users/xx/Documents/Arduino/hardware/megaavr/cores. Is it pre-1.5?
Could not find boards.txt in /Users/xx/Documents/Arduino/hardware/megaavr/extras. Is it pre-1.5?
```
Randomly copying .txt files around results in multiple entries in Tools menu and compilation errors.

Which .txt files should be in which subdirectories ?

Thanks.

Arduino IDE 1.8.13
macOS 10.13.6 High Sierra

SpenceKonde, 2021-01-03:
> existing under /hardware and copied over files from downloaded 1.3.0

You are missing a folder level. Everything should be in a DxCore folder
within hardware, those errors indicate that instead the contents of that
folder were placed in hardware, meaning the ide thinks each subfolder is a
core

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

obdevel, 2021-01-03:
That's it, thanks.

Re-reading Installation.md, it's not immediately clear that's what I should have done. Should've backed up/renamed the previous tree rather than trashing it !

closed 2021-01-04 by SpenceKonde