SpenceKonde/DxCore #54: Wire1 library and compatibility with existing libraries

MCUdude, 2021-01-17:
As you already know, the Wire1 library really doesn't have anything to do with the Wire library. They are separate libraries, and they have different class names, TwoWire and TwoWire1. This is fine if you're using the library as it is, but not so good if it is used within a library or passed through a constructor like I know lots of the [Adafruit libraries do](https://github.com/adafruit/Adafruit_MCP4725/blob/master/Adafruit_MCP4725.h). In this example, there is no easy way to use the TWI1 hardware with the MCP4725 library.

The Wire and Wire1 libraries should ideally come from the same class, TwoWire. This may sound like a difficult task, but it has already been done for [MiniCore](https://github.com/MCUdude/MiniCore/tree/master/avr/libraries/Wire1) (ATmega328PB) and [MightyCore](https://github.com/MCUdude/MightyCore/tree/master/avr/libraries/Wire1) (ATmega324PB). @asukiaaa was so kind and did all the hard work for me a while ago. It should at least be considered 🙂

SpenceKonde, 2021-01-17:
Oh, huh, yeah I guess it would have that problem. Hadn't thought about it. These fucking "classes" again!

Well, I'm not about to jump onto this >.> 

Did you see why I did with SPI?  (see #32 and #35) - you can now use the one SPI library with either SPI0 or SPI1 (since those libraries are incompatible with having more than 1 SPI interface decleared to other libraries if the thing called SPI1 isnt a member of spi_class (like for examplke, the SPI_t struct from the manufacturers io headers....) I decided that one really doesn't care about having multiple SPI ports - especially since the arduino library only supports master mode) - and it's more about the flexibility in which pins you're using.  Anyway, I don't feel like doing serious work on another basic peripheral interface library so soon!

MCUdude, 2021-01-17:
> These fucking "classes" again!

Well, classes can be your best friend if you know how to use them for what they are. I know it can be a pain in the butt, but if you succeed, you'll have sleek and elegant code that is IMO a joy to use. I'm pretty much self-taught when it comes to classes and C++ in general. I've written a few libraries I host here on Github, and that's about it.

Just wait until you stumble across templates, or even worse, recursive templates! That's super hairy and super powerful.

> Did you see why I did with SPI?

Couldn't you just

```c
#undef SPI1
#define SPI_1 (*(SPI_t *) 0x0960)
```

somewhere in the core files? I doubt this would cause much of an issue for your average DxCore user.

SpenceKonde, 2021-01-17:
I thought about that approach... but I really don't like the idea of renaming defines from the IO headers...

SpenceKonde, 2021-01-23:
I looked at the changes they made - I ain't got a prayer of making that happen for this library, lol

SpenceKonde, 2021-03-26:
The nonworking Wire1 library has been removed from this core. - somehow it wound up not even compiling, period, and there are a bunch of changes that  I would need to a,have taken in order to get the desired output.

SpenceKonde, 2021-04-01:
Did not intend to close though nothing has changed here.

MX682X, 2021-06-05:
Due to an unfortunate event I had to start using the second TWI. After seeing how messy the maintaining of a second TwoWire Class would have been, I decided to try to rewrite the Wire library.
I've uploaded the files here: https://github.com/MX682X/Wire_Reworked_DxCore
Features: 
- When using only one TWI at the same time, the TWI can be swapped, which should give the possibility to use TWI1 directly with the Wire library
- Uses a bit less RAM when used for TX and RX

Disadvantages:
- Needs a bit more Flash space
- Needs an extra dropdown in the Board feature selection where the user can decide to use only one TWI or both, because I was not able to make the compiler decide on compile time if the Wire1 class should be initialized or not. 
- Needs some reviewing and beta testing
- When only TX or only RX is used, it needs more RAM

I'm also thinking about to create a pulled solution for the Master functionality, since Arduino requires a block while the transimission is in progress. Might save some bytes in the Flash.

About the dropdown:
Maybe it would make sense to have 3 options: Legacy, Reworked (Single TWI) (beta), Reworked (Dual TWI) (beta)

MX682X, 2021-06-13:
@SpenceKonde After I had to start using the second TWI, I basically started rewriting the code for the Wire Class. I think I'm mostly done now. I just need a bit of help with the swap and similar pin functions, because I still don't really get how it works...
Anyway, I've uploaded the new Wire library here: https://github.com/MX682X/Rewritten_Wire_for_DxCore/tree/main/Wire/src

Since I was only able to test the code on a Tiny1614 and an AVR128DA48, I recomend to keep the old library for now. Since my library requires some global defines (for test purposes to be found in twi.h) I suggest to include another dropdown menu in the board selection with following options:
 - Legacy
 - 1x Wire, Master or Slave
 - 1x Wire, Master and Slave
 - 2x Wire, Master or Slave
 - 2x Wire, Master and Slave
 
 I've added following functions to support the simultanious M and S:
 - `void   enableDualMode(bool fmp_enable);`			
 - `size_t writeSlave(uint8_t);`
 - `size_t writeSlave(const uint8_t *, size_t);`
 - `int    availableSlave(void);`
 - `int    readSlave(void);`
 - `int    peekSlave(void);`
 - `void   flushSlave(void);`
 
 In order to support the "virtually unlimited slave addresses" I've also added `uint8_t getIncomingAddress(void);` which returns the address that triggered the slave interrupt
 
 Furthermore, in order to give the option to use the TWI1 pins with the Libraries that only use Wire, I've added `bool swapModule(TWI_t *twi_module);` that gives the user the option to reassign the Module that Wire works with. Doesn't work when both Wire are enabled.
 
 By rewriting the library, I was able to significantly reduce the Flash usage on small chips. While the old library was using >1.5kB on a tiny202, this one compiles with <1kB in the Arduino IDE with the test sketch.
 
 Will write a function documentation later.

SpenceKonde, 2021-06-14:
Wow.... this is exciting. I didn't seriously think anyone was going to step up to this one. 

And we get dual mode too? and it'll help shut up the whiners on the 2k parts? 

Daaamn. This is intense. 

It looks to me like you create a copy of several methods to support the slave operation in dual mode, which would break code compatibility, and so someone couldn't get a working combination by just combining their master and slave sketch. We would very much like to avoid this, as code compatibility is the most valuable asset of any library. especially something like Wire which gets used by other libraries all over hell which expect it to work in one way. While it is not going to be as common to have a library use the slave mode of Wire.h, it is entirely legal for it to do so, and I would be shocked if libraries like that didn't exist. Telling people they need top modify a library, that doesn't make them happy.

But I think in the case of dual mode, we can determine from TWI._STATUS registers whether we're in the slave or master context, because the slave commands are ONLY used in the context of slave interrupts, right? So we could either deduce from the status registers of the TWI._STATUS which we would probably need to read from anyway - does that work? If we had to we could set a global (or even a GPIOR bit, if the overhead was prohibitive), before calling the user code for onReceive, clear it after we leave and that tells uis that we should use the slave version of the method if it's set when the method is called....  Am I missing something? I don't claim to have a great understanding of that peripheral - (I think it'a one of the most complicated of them, too) very possible that I am misunderstanding something.

MX682X, 2021-06-14:
Setting a flag should be actually possible to determine the context the read comes from.
I'll fix that later this day

MX682X, 2021-06-18:
@SpenceKonde I think I'm finished now with the Wire library itslef, you'll have to check the swap(), pins(), usePullups() and ClearPins() functions though that I put in the twi_pins.h file since I still have trouble figuring out the decision tree. I tried adding them for TWI1, but I'm not usre if I done it right. Also, since dual mode is now supported, maybe check if the alternative position 1 works?
The stream functions are using a bit in one of the variables now to check for the calling context. I decided to go with a variable, because it is more reliable in my opinion.

MX682X, 2021-06-18:
And of course I have now an idea how to reduce the RAM usage/increase the buffer size. I came to the conclusion that the TWI does not need a seperate tx and rx buffer since the two things happen one after another. Technically it should be possible to use only one buffer for rx and TX. 
I'm not sure if this will be still compatible though. Debugging is also easier with two.

SpenceKonde, 2021-06-21:
I thought about that as well. Is it possible  use the slave versions of the methods from outside of an ISR+? if so, do do know of any libraries that do that..... It would be a huge boon for the smaller parts. 

A question about that "remember to uncommment the defines" - are some of those new features mutually excliusive which would require some means of telling it which one to use?

MX682X, 2021-06-21:
> Is it possible use the slave versions of the methods from outside of an ISR

Well, not if the master and slave is enabled - without that, there is the standard Arduino interface accessing the "shared" variables, meaning that if a library is indeed writing to the slave outside the ISR, the user just won't be able to use the extra functionality of a simultanious master. Furthermore, making extra slave functions would lead to the same result - rewriting the library. 

> It would be a huge boon for the smaller parts

I added a define in the twi.h file as an option to merge the buffers. It does not save a lot of flash, but a bit of RAM. It is not really relevant on the bigger ones. The problem is though, that it is really hard to guess if it will work everywhere, so I had labeled it "experimental". 

> are some of those new features mutually excliusive which would require some means of telling it which one to use
No, they are not mutually exlusive, it's more of a checkbox of which features are needed. I wrote it that way so the people can decide on what they really need for their project. When every define is commented out, the feature set is the same as in Arduino. Some of them don't work on every part though: `USING_TWI1` works only on parts with two TWIs, `TWI_DUALCTRL` works only on devices with the DUALCTRL register, `TWI_MANDS` should work everywhere except on the x12 and x02 parts (should as in the IDE must not give an option to enable it, the parts simply don't support it)

SpenceKonde, 2021-06-28:
Wow, had missed your last reply. 

My understanding was that SPI.write/read/etc, if they were to act on the slave functionality, must always be called from within the ISR.... so any time they were called outside the ISR, we could safely assume they were meant for the master, and if they were called during interrupt, we can tell which interrupt it is, and hence whether we expect it to be working with master or slave. 

I see a lot of potential problems with Wire using a shared slave/master buffer, I think that's asking for trouble, nobody has complained about RAM usage of the Wire libraries - only the flash usage, and the fact that it can't do master and slave at the same time.

MX682X, 2021-06-28:
@SpenceKonde  I think there is a misunderstanding. The Slave and Master Buffers are always seperate when the user is using the simultanious operation (TWI_MANDS). 
But when only one operational mode is needed, there is no need to use two buffers, so there are aliases to use the master buffers in slave mode, like the Arduinno implementation does.

The TWI_MERGE_BUFFERS define is an experimental define based on a thought of mine and the comments you left in the old version: Since the code is blocking, there should never be a case, where a simultanious transmission and receive of data takes place. I've written the code using ring-buffers, like the Serial core. When I realised, that there are no simultanious rx/tx (especially since there are START and STOP conditions), I thought, there is no need for seperate write and read buffers (don't confuse with master/slave) and I added some aliases that will take care of removing one of the buffers virtually. I used aliases to keep the code maintanable.
I've written 'should' on purpose since I can not gurantee that this will always be the case. 

By my knowledge of people I can see how they will use the buffer merge option at the beginning of a project, write a code that barely fits on a chip, only to find ot it doesn't work. And then they start complaining why there was such option to begin with. So I hid it, as something like the last resort.

simengangstad, 2021-07-09:
I've tested with a DB64 and DA48 against a range of devices with I2C interface now, works mostly fine on both Wire and Wire1😄I've fixed some things in the pin, swap and pull-up functions.  One needs to rethink those functions especially if one wants to support both TWI0 and TWI1 and do checks at compile time though. As by now if you call Wire1.pins with a valid configuration it fails since it checks this pin configuration towards TWI0 as well during compile time. I've pushed a PR to your repo @MX682X. The temporary fix in this PR is just to omit that badArg call and return false instead.

If one wants to do this during compile time I suggest possibly making swap and pin functions virtual and make two small subclasses for Wire and Wire1 which just implement these methods and thus separates TWI0_Pins/TWI1_Pins and TWI0_Swap/TWI1_Swap.

MX682X, 2021-07-09:
Thanks for the help!

I totally forgot to add the dual-mode to the pin manipulations. Will tackle that on the weekend.

I see what I can do about the compile-time check.

MX682X, 2021-07-17:
@simengangstad I tried to make the pin functions virtual, but failed (this was also my first attempt with virtual functions, didn't wrote much in C++ so far). I did not manage to make it to compile and work. I managed to come up with a workaround though that at least checks if the supplied pins belong to one of the TWI modules. However it will not be able to check at compiletime if the pins belong to the correct TWI module. 

You can find my attempt at the subclasses in the same repository, I made an extra branch for that. I'd appreciate the help if you have any idea how to fix it.

SpenceKonde, 2021-07-17:
Both of you are heros :-)  I am sorry I haven't gotten to give this the attention it deserves myself. 

It strikes me that there are two distinct areas being discussed here. 
The first is the Dual Mode and Multi-master (ie, device acting as master and slave using the same pair of pins) 

And the second is the title issue of a Wire1. I want to make sure that in persuit of Wire1, I want to make sure we do not trade off code size until it is no longer usable on tinyAVR devices, because bring able to act as both master and slave on the same pins, which the documentation suggests is supported is I think the second most loudly requested feature there, second only to the sleep and power save library.

I am going to say that I - for one - am perfectly happy with a Wire library that takes the copout that I did with SPI (where SPI.swap() just includes the SPI1 and SPI0 pin sets, and uses whichever peripheral corresponds to the pins you selected;-) ) and only lets you use one at a time. A use case that requires both of them to be operating at the same time.... I am unsure as to what real world situation that would correspond to., while the use case for acting as both slave and master is pretty clear, and people seem to really want it.

MX682X, 2021-07-19:
At this point the codesize is small enough to fit a master and slave functionality on a 212 - even though, just barely. I made an example sketch that makes the chip a remote ADC that communicates with I2C: [mands-212](https://github.com/MX682X/Rewritten_Wire_for_DxCore/tree/main/Wire/tests/sizetest_mands_212) . Could only test it on a 1614 though, don't have a 212 at home.

A use for a parallel I2C would be as Level-Shifter or having different peripherals for different voltages. Of course this case is limited to the DB series, but why cripple something that already works fine? I took good care to write the library in a space efficent manner by using a lot of defines to only use as much space as neccessary. The only trade of is the semi-working pin check on compile time. In fact, the user can decide to use only one Wire object, but change the TWI module with Wire.swapModule(&TWI1) .

But, if you really want to reduce the functionality, you are free to do so. I intentionally used the MIT license so the code can be altered and the license chaged the GNU Lesser license.

P.S. the Dual Mode and the Multi-Master functionality can be enabled seperately with defines as well - check out the code

simengangstad, 2021-08-06:
Sorry for the late reply, but I've been quite busy with other work. Tested more with a DB64 and a DA48. Multi master seem to be working fine between them as well.

MX682X, 2021-08-07:
@simengangstad If it is infact an collision aka "Arbitration Error" you can check it by changing the line 346 in the file twi.c from `goto beginning; ` to `return 0xFF;` and add an if case to the sketch files that checks the return value of `Wire1.endTransmission()` for this Value. If it's true, let it print out on the Serial Monitor. You might want to add an endless loop at the end so the line won't get lost. The loop that they are stuck in is probably at line 338.

I'd like to check this first, otherwise, it means that the Bus is kept in a busy state for some reason that would need some more debugging.

simengangstad, 2021-08-16:
Checked now, didn't get stuck due to arbitration. It is during the TWI_MasterWrite though. I've tried to reproduce it with debugging in Microchip Studio with vMicro, but I can't reproduce it then, which begs the question of whether there are some timing issues which only happen when certain optimisation levels are used and not within debug.

MX682X, 2021-08-16:
My best guess is that the Master waits for the bus to get idle, but the slave is keeping the SCL line low. To make sure it is not the case, I'd like to ask you to do another edit. On line 684 in twi.c I left a commented out endless loop. I used it to find all the cases for the switch-case. Maybe I missed one? 
Since you can use the debugger, let it run untill it locks up, then use the "attach to target" option to connect to the running Slave and look up the TWI SSTATUS Register and tell me its value. 
Should Microchip studio show another line then 684 of twi.c when using the "attach to target", the problem will be infact in the MasterWrite function. In that case I already have an idea how to rewrite the function that might help with the timings.

simengangstad, 2021-08-17:
Sorry, I meant that the debugger didn't wasn't able to catch it. I tried using vMicro, but it doesn't seem to work well with my setup, so debugging is a bit tricky. I did however use an logic analyzer on the line and when the hangup happens both lines are high. do you have any tips for debugging an arduino project? I've mostly been on bare metal, where debugging is quite plug and play with the right equipment and Microchip studio.

SpenceKonde, 2021-09-06:
With 2.4.2 of megaTinyCore released, I checked in this code for 2.4.3

SpenceKonde, 2021-09-06:
Fails code linting: 
https://github.com/SpenceKonde/megaTinyCore/actions/runs/1205395272
https://github.com/SpenceKonde/megaTinyCore/actions/runs/1205395274
Fails spelling check
https://github.com/SpenceKonde/megaTinyCore/actions/runs/1205395285
And all of the compile tests fail on all tinyAVR devices. 
https://github.com/SpenceKonde/megaTinyCore/runs/3523017140

SpenceKonde, 2021-09-06:
Oh,. except on the 2k partts with optiboot where we don't even try because there;s no space. I thought the examples I carried over were *designed* to work there....

MX682X, 2021-09-06:
The problem is most likely 
#define USING_WIRE1   
#define TWI_DUALCTRL   
in twi.h. They need to be comented out. By defining USING_WIRE1, the code uses TWI1 which does not exist on tinies. Gonna fix the linting and spelling when I get home.

MX682X, 2021-09-06:
Alright, I've fixed (hopefully) all lintning and spelling. I also uploaded a boards.txt that will allow people to enable the independent Master and Slave functionality

SpenceKonde, 2021-09-07:
Aaah, where can I find this new version?

MX682X, 2021-09-07:
In my [repository ](https://github.com/MX682X/Rewritten_Wire_for_DxCore). Thought about adding a link but forgot. I will try to update the boards.txt for the Dx Series too.

SpenceKonde, 2021-09-12:
Got things buiilding on megaTinyCore. and put some time into studying and making them pass linter (I turned off the indent checking on most files - it insists on not indenting preprocessor conditionals and the code is a wall of text without that. Nice flash savings looking at the example sketches. Next people need to try it out in real-world circumstances and see if it works....

MX682X, 2021-09-12:
> Spence: Isn't that situation only relevant on parts with dual mode? I didn't think you could receive between beginTransmission and endTransmission... except on a part with dual mode
(from [twi.h](https://github.com/SpenceKonde/megaTinyCore/blob/38f8c660bf97932cddbbd4a189e0d62247706695/megaavr/libraries/Wire/src/twi.h#L45))

It is possible. This is due to the restrictions created by the original implementation from Arduino. Wire.beginTransmission() and Wire.write() are only preparing data. The actual write only happens when the code calls Wire.endTransmission(). When there is only one buffer for rx and tx, the rx data might overwrite the data prepared by Wire.write().

I've also finished the boards.txt file for the Dx-Core, but the fact that the 28-pin version doesn't have 2 TWIs it will require one more edit to twi.h in a future release to throw an error when a wrong combination is selected. I've uploaded a twi.h to my repository too, as a suggestion how to check for that.

I've also like to excuse the missing linting in my code - never had to work with a linter before, so I did not know what to pay attention for. Since I was thinking about writing some good examples on how to use the library, I'd like to ask, which linter are you using?

SpenceKonde, 2021-09-13:
Yeah I realized that after I wrote it but couldn't find where I'd said that to remove it and was meaning to track down and remove the comment. 

I use astyle because that's what @per1234 set up and I think it;'s what the IDE uses which is what led to its choice. I absolutely despise it, because it is inconsistent over
```
#if condition
  stuff
#endif     
```
vs
``` 
#if condition
stuff
#endif    
```
but when it decides on one indentation method in a place, it demands it otherwise it fails the file. 

As far as I am concerned, 
``` 
#if condition
stuff
#endif    
```
is always wrong, and every conditional chunk of code  should be indented. I don't love this result because preprocessor and runtime conditionals are fundamentally different, but it's not like there's some other whitespace character that is visually distinct from a space (indeed, none is possible because then it by definition would not be whitespace) that could be used to indent preprocessor conditionals, and we must have something. If you look at what the linter wants things changed to, it;s a fucking wall, nobody could read that shit. Which is exactly the result that linting is supposed to prevent!! But I can't seem to configure this piece of shit to see the light. 

But turning off the linting for a file including just the indent linting causes it to miss all sorts of other issues, so I can't do that either

And I need a linter, and I can't switch to another one because that would require a much deeper knowledge of how to work with github actions than I have (I can barely do anything to change the github actions in the CI - I don't understand anything, and all the resources I find are impenetrable, inapplicable, or don't behave as claimed when I try to use them, and people who swear up and down that they work with github workflows all day and are experts, when I ask them questions they look like I've shown them a scroll covered in heiroglyphics

So yeah,. I can provide no advise on linters. 

Some of the valuable things the linting does is catch deviant styles such as 

```
function x()
{
```
instead of 
```
function x() {
```
and 
```
}
else
```
instead of 
```
} else {
```
The linters are generally very intolerant of things that I think are pardonable like single line if/then statements, though I generally acqueiss to it on that front. 

I will also note that within the core proper, I don't run the linter because it simply would never pass any code that was readable for most files within the core. Mostly because of the stupid indentation behavior.

SpenceKonde, 2021-09-20:
> I've also finished the boards.txt file for the Dx-Core, but the fact that the 28-pin version doesn't have 2 TWIs it will require one more edit to twi.h in a future release to throw an error when a [...]

Wouldn't just checking if USING_WIRE1 is defined and TWI isn't, and then #error if that is the case work?

MX682X, 2021-09-20:
This would indeed work. Fixed it now. (Sometimes you just loose track of all the defines, especially when working on different projects)

SpenceKonde, 2021-09-22:
I see that there are a few additions to Wire,h 

```
    uint8_t getIncomingAddress(void);
    void   enableDualMode(bool fmp_enable);      //Moves the Slave to dedicated pins

    void    TWI_onReceiveService(int numBytes);
    uint8_t TWI_onRequestService(void);

    static void onSlaveIRQ(TWI_t *module);    //is called by the TWI interrupt routines
```
I'm guessing that the last three aren't something users are expected to interact with directly? 

getIncomingAddress seems like an obvious addition in order to make use of the address mask or secondary address, which should have been added a long time ago had someone realized that it really ought to exist. 

It looks like we don't provide a way to disable dual mode (which seems to be a very strange thing to do)? 

What restrictions are there on the order of operations here?

(am updating docs)

MX682X, 2021-09-22:
you are correct, the last three are used internally only, they are called by twi.c

About disabling dual mode: I considered that many will have their setup hardwired and have the slave always on the same pins. I just don't expect people to switch the slave between two buses. If necessary, it would be better to call endSlave() and begin(addr) instead, I think.

getIncomingAddress returns a left shifted address resulting in a 8-bit Address. To check for the address, this has to be considered, see this [example ](https://github.com/MX682X/Rewritten_Wire_for_DxCore/blob/main/Wire/examples/slave_dual_address/slave_dual_address.ino)

enableDualMode() has to be called before begin() because there is a check in twi_pins.c that checks if dualMode is enabled and takes action to make sure the slave pins are in the correct state. I'll try to write an example for that today.

SpenceKonde, 2021-09-22:
Cool - that is somewhat less urgent since DxCore needs a release to fix the criticals before this gets checked in for the following version (out of an abundance of caution)

SpenceKonde, 2021-09-22:
Huh! Well, the fact that it is left-shifted a bit definitely needs mention in the docs - Now I may be missing something I shouldn't be. but why use the 8-bit address? Doesn't the rest of Arduino prefer 7-bit ones. 
Also, what happens if this is called at an inappropriate time?

My "instinct" argues, albeit without supporting evidence (as instincts are wont to do) it should return a 7-bit address, or -1 (0xFF) when the damned fool uses it where we know it has no meaning. But I don't even know how difficult detecting that is, or if the LSB is also encoding meaningful information here.

SpenceKonde, 2021-09-24:
Is there a reason for the #include "avr/io.h" in twi_pins.c? It already includes Arduino.h,  which #includes io.h (this is always true within AVR-based Arduino cores; the Arduino.h always includes io.h) - so that can come out right?

Also, - can you explain what you mean here? 
```
/**
 *@brief              TWI_MasterCalcBaud calculates the baud for the desired frequency
 *
 *                    Moved temporarily to twi_pins.h due to licensing reasons
 *                    Can be moved back into twi.c on merge to the DxCore and license adjustment
 */
```

What should I do on this front? I'm not an expert in software licenses, so I'm feeling a little bit confused here about how to handle the fact that two files are provided with a new type of license. (As you are probably aware, my cores are LGPL 2.1 *unless stated otherwise* - the tinyNeoPixel and tinyNeoPixel_static libraries for example. are LGPL 3, since they were derived from Adafruit's library.

per1234, 2021-09-24:
This is a helpful reference for determining compatibility between GNU licenses:
https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility

MX682X, 2021-09-24:
`Is there a reason for the #include "avr/io.h" in twi_pins.c` 
I probably added this sometime at the begining, no need for it anymore

`Also, - can you explain what you mean here?`
I don't know much about licenses aswell, that's why I choose the MIT License. It's a do what you want license, and it includes the possibilty to make a LGPL 2.1 out of it. But to make sure to not violate the license when writing the library, I moved the Baud calculation, that you have measured and written, to the file that had other parts of your code. This function would be thematically better fitiing in the twi.x files

SpenceKonde, 2021-09-25:
Got it, so since it is licensed under MIT I can cast to... i mean relicense as LGPL 2.1 safely to match the rest of the core, rather than having another exception. (me? I didn't write that baud function! The one I wrote (after someone pointed out that the second attempt to fix it still generated wrong results) was significantly crappier, someone else saw it, probably did a facepalm, and then implemented it more rigorously. I know I called them out by name somewhere.

SpenceKonde, 2021-10-02:
On devices that support it, should we be switching the input levels to the to the SMBus 3.0 option? 
Default thresholds are based on Vdd - and at 5, are nerve-wrackingly close to 3.3v for a high (the specs say 3.5, typical characteristics say 3,1 at room temp), which would make one of the big advantages of I2C (that you can use it to talk to a lower voltage bus without fear) non-viable; even the "easy" case of 5v device to 3.3v bus is right on the edge., With the SMBus levels, no matter the operating voltage, the low threshold is always at like 1.09V (guaranteed to be above 0.8) and the high threshold around 1.24V (guaranteed to be below 1.35). That would let us talk even with 1.8v sensors over I2C.

Am I missing some downside to this? It seems like it would be a generally Good Thing.

MX682X, 2021-10-02:
technically, that should work, practically, it should be tested. (A beta program where some volunteers test the code on their setups would be cool)

btw, I'm working on a modification on the library that should be able to detect a problem with the pullups, or at least a missing high-idle voltage on the bus. But it might take a while, because A: the code is pretty complex and debugging is a pain when Microchip studio isn't showing local variables (I haven't tried adding the debug option in the compiler yet) and B: I have a couple of other tasks to handle right now. It's gonna take at least one more week, I think, since I'm programming in the evenings most of the time. If it will ever work. And will be small enogh. Because I'm trying to write a software that catches the problem on the bus itself, so I wonder if this (probably) 100 bytes are even worth it. One idea I had when writing the endless while loops was that if there is a problem, you wouldn't want it to be missed, but investigate why it didn't work. This week showed me that it doesn't work that way.

here is a preview, I've modified the MasterRead. It still has some bugs in terms of correctly identifiying the problem and it reads  more then specified when reconnecting the wires. 
[twi.zip](https://github.com/SpenceKonde/DxCore/files/7271784/twi.zip)
 
 P.S.:

> you know you're not supposed to increment when you can decrement, right?

The compiler actually makes a subi 0x01 out of a `++timeout` so there was no difference in FLASH usage wether or not decrementing or incrementing

SpenceKonde, 2021-10-05:
Thanks for the update

Yeah - I'm not sure if the work of implementing something more advanced than a crude timeout is worth the space and effort either, it sounds both like non-trivial amount of work and like it would use a lot of flash for something most won't use (as long as what to do was documented ("check pin levels if problems is transient when the situation clears do this to put wire back in a good state (does that even need to be done? Is there a way to, once the bus is fixed, make sure the TWI gets back into a good state? (ex, calling begin() again? end then begin?  or anything like that))" ).  

I try to make a point of not actually doing debugging for people (well, at least within the code, I inevitably end up doing it in a different sense...) (I've been tempted to implement some sort of lastError() type scheme, or core error flags, but I decided that was going too far. I generally instead try to make sure it left them in a position to handle problems when they occur, as the needs of their application dictate...  

I think what's making it such a problem for people is the fact that their code just hangs for an eternity, they can't program it to react to the I2C failure so it requires manual intervention to revive. With even a simple timeout, if it times out, I2C may be out of service, but they can detect and react to it, and maybe the microcontroller has an output wired to all the reset pins of the slaves or something, or they know the problem eventually fixes itself, or they can continue their other duties, - it's the devices getting hung and requiring manual intervention all the time that I think is the real problem, because they're already deploying things at an early phase of design. 

Huh! Surprised, because if you subi or sbiw, and testing against 0, it becomes a breq or brneq, if testing if it has counted up to a target is a separate cpi - unless you;'re upcounting until it rolls over, or the compiler is smart enough to make it do that.

SpenceKonde, 2021-10-10:
FYI: The TIMEOUT bits in the TWI configuration are known to be busted on some parts. Not that it's particularly useful most of the time - but aopparently on some parts the bits are not writable.

Thank you for the reminder Microchip, every time. Every damned time, I see the PCN notification, so exciting.... but alas, it is always to add new errata not another die rev to fix them (they apparently changed the wording of that erratum, nothing serious. yeah, timeout bits are treated like reserved bits on a lot of the older parts,. 

They found another one on the type D timers with one of the events. I'd say I was amazed it took this long to find, but I;m not, because I've worked with that timer and that contraption is unruly enough to keep all but the most dedicated developers away. It's complexity to errata option was WAY out of whack at first. Only the Dx-series has gotten people really playing with them, since now it really has something to offer in the way of that PLL clock... 

This one for the 16k 1-series broke the pattern though. They removed one that they said wasnt relevant. It's the start of frame detection one. Said it wasn't relevant. That's the kind of thing that makes you raise an eyebrow.... The bug described would, if the opart was used in the most natural way, trash serial reception when in sleep mode, and the whole point of waking on serial is so you can receive it.... And that erratum is on EVERY modern AVR errata sheet. I suspect the answer will be made clear in the coming weeks; my guess is that it's being fixed with a pencil (or maybe just an eraser), but maybe I'm just too pessimistic.
Also we got some datasheet clarifications. Someone who can convert between hexadecimal and binary read the fuses section?  And the sleep mode/oscillator tables are multiplying, sounds like they're getting too many inquiries through support.

SpenceKonde, 2021-10-15:
I wouldn't bother going through and renaming anything - I'm keeping the more widely used master and slave. Adding more pairs of names for same old thing will huju[spread confusion, particularly among those for whom english is not a first language, and only know a few key words in broken english and rely on the consistency of the industry jargon. Particularly with the confusing names they picked. They only captured the one(or few)-to-many aspect, but I thought the more salient part of I2C master vs slave (like SPI) si that the master is the one who is responsibel for initiating all communication, providing all clocking, and so on. And in that paradigm, these terms are backwards, since the client initiates the actions  on a network, while the host just answers them? And they managed to find a pair of words that don't cleanlty map to the roles, are more ambiguous, and start with different letters - AND will disagree with datasheets going back decades. The new readme will get a note but while publushed datasheets with new names remain rare, I'm sticking with the old ones (they certainly left no room to question which one was which)... 

I sure *hope* we get a bunch of fixes backported after the DD arrives. It would appear rthat almost all of thesubstantive bugs are long since fixed, and we're just.... waiiting and waiiiting and waiiiting for them to pout out silicon with fixes (frankly the ones I care sbout asre a pretty small subset too....
CCL 2.4.1 The CCL Must be Disabled to Change the Configuration of a Single LUT (Everything - this change is HUGE for me)
RTC 2.6.1 Any Write to the RTC.CTRLA Register Resets the RTC and PIT Prescaler
RTC2.6.2 Disabling the RTC Stops the PIT
NVMCTRL 2.5.1 Flash Mapping Into Data Space not Working Properly  128DA only (actually doesn't make a difference now, my code all works around it by uding word writes, which are better anyway)
TCA 2.9.1 TCA1 Pinout Alternative 2 and 3 not Functional (128DA only)
TCB 2.10.1 CCMP and CNT Registers Operate as 16-Bit Registers in 8-B (Everything, but I never use that mode anyway because it kinda sucks) 
TCD 2.11.2 CMPAEN Controls All WOx for Alternative Pin Functions All parts (All Dx ,This one I really hate)

So at this point, since I'm expecing to release DxCore 1.3.7 shortly, I'm going to be wanting to check in the new version of wire for testing as soon as possible after I do that.

SpenceKonde, 2021-10-31:
Okay ~1.3.7~ 1.3.8 is in, will be pulling in your changes today for open beta testing.

SpenceKonde, 2021-11-05:
Can't pull in until 1.3.9 is released, and 1.3.9 release is needed urgently to resolve breakage to SPI.h and several other critical defects in 1.3.8.

MX682X, 2021-11-06:
This comes quite in handy, actually, since there is one minor improvement I wanted to add that would reduce the code size a little bit more. Didn't know how to call static class functions when writing the new implemetation so I was saving a pointer to it to call it from. Still have to check if the changes work.

MX682X, 2021-11-06:
Seems to work, so I've uplloaded it to the repository

SpenceKonde, 2021-11-21:
This is essentially in now and will expect only a few more tweaks! Huzzah!

closed 2021-11-21 by SpenceKonde