SpenceKonde/DxCore #56: AVR128DB32: DTR autoreset only works once after burning bootloader

Miraculix200, 2021-01-19:
DxCore version: 1.2.0-a
MCU: AVR128DB32
Board: AVR DB Series (optiboot)
Programmer: FTDI Basic 3.3V
Bootloader: optiboot_dx128_ser0.hex
Fuses: Everything default except
BOOTSIZE fuse: 0x01 (512 byte)
SYSCFG0 (RSTPINCFG) fuse: 0xc8 (hardware reset enabled)

Schematic (scroll down): https://oshwlab.com/Miraculix200/minimegazero24_copy_copy_copy_copy
(includes 100n cap between DTR and RST)

jtag2updi doesn't work for me, so I had to use pymcuprog to burn the bootloader and set fuses. 

After burning the bootloader and setting the fuses using pymcuprog, it is possible to upload a sketch using DxCore on Arduino IDE using DTR autoreset.

Once that sketch is running autoreset doesn't work anymore, and the reset button has to be pushed manually to upload sketches. To make it work again I have to burn the bootloader again.

I've tried setting SYSCFG1 to 0x06, because I saw it in the DxCore source code. Setting it to 0x0E didn't help either.

Any idea what the problem might be?

SpenceKonde, 2021-01-20:
Okay, first off, DTR autoreset doesn't work once for you. It never works The only reason you can upload immediately once is that when there is no sketch on the board, the bootloader runs continuously. Well, sort of--- IIRC there was some problem with it and it doesn't actually behave like it's supposed to or something.

Secondly, yes, I know exactly why  your autoreset doesn't work. You have a 1uF cap between Reset and Ground. The100nF cap you have between DTR and Reset can at most bring the voltage on reset down to only 10/11ths of Vcc. However per table 39-7 in the datasheet, the reset pin is guaranteed to read high (and not reset the part, presumably) above 0.8*Vcc, and is not guaranteed to read low until it is brought down to 0.2 * Vcc or below.  You're also supposed to have a physical pullup too 

Please refer to the schematic of working designs which use an autoreset circuit: 
10k ohm resistor between Vcc and RST. 
0.1uF capacitor between reset and DTR. 
And no capacitor between reset and ground (I suspect you can get away with one, but only if it's smaller than 1/10th or 1/20th of the value of the DTR cap? I've never had problems with interference resetting my board....  *shrug*

Definitely snag the 1.3.0 bootloader code though when I check it in later tonight - it fixes at least one entry condition problem, makes it fully honor the entry conditions, , and adds a way to call the bootloader from application to write to unused areas of the flash for mass storage!

SpenceKonde, 2021-01-20:
Closing because this does not represent a defect in the core

closed 2021-01-20 by SpenceKonde

Miraculix200, 2021-01-20:
Thanks for the explanation. Will burn the new bootloader later.

The AVR128DB32 already has an internal pullup on reset however. See 4.3 in the datasheet. I copied the circuit from that page. 
https://ww1.microchip.com/downloads/en/DeviceDoc/AVR128DB28-32-48-64-DataSheet-DS40002247A.pdf

I'll remove the filtering cap and see what happens.

Miraculix200, 2021-01-20:
I can confirm the new version of the bootloader works on the AVR128DB32 (not sure if you have that one). Didn't try the new feature yet however.

SpenceKonde, 2021-01-20:
Correct - the schematic in the datasheet does not support autoreset. Autoreset is an Arduino thing

The pullup ought to be 10k for 0.1uF cap to get the timing everyone "expects" (and that also gives you a bit more noise immunity). If you're really concerned about noise getting into reset (maybe you had a problem with that) and the serial adapter isn't built in, one could even jump the DTR line to ground when the adapter isn't in use, and then your autoreset cap becomes a filtering cap :-P

I am very aware that you haven't tried the new feature, as I didn't actually commit flashwrite.h and flashwrite.cpp... I was going to, but then I realized the folder name was wrong, so I tried to rename it, but something had an open file in it and I had to close that app, but I couldn't find which app, so I had to close every one it could be - but I couldn't because isome of them still had unsaved changes, and next thing I knew, it was time to go to bed....

Miraculix200, 2021-01-20:
I've been building ATmega328 circuits on perfboards for several years (some even without 100nF decoupling caps) and never had a problem with noise on RST. Always used 100nF cap DTR to RST and 10k pullup. 

After reading the datasheet I thought this was a new thing and adapted my schematic. Now I'm doing it the old way again.