SpenceKonde/DxCore #57: Flash libraries

MCUdude, 2021-01-23:
First of all, it's awesome that you've created a flash write library! In the code somewhere I saw you requested some feedback. So here I am 😉 

1. You'll have to fix your include guards. They're supposed to look like this; if not, they have no effect:

```c++
#ifndef FLASH_H
#define FLASH_H // (you forgot this line)

// Code goes here

#endif
```

2. Why have an object-oriented and a non-object-oriented library? They do exactly the same thing, and it is twice as much work to keep them up to date. My advice would be to stick with one of them. Oh, BTW the non-object-oriented library doesn't compile.

3. The example
The example is way too large/long in terms of code lines IMO. As a start, why can't all the #ifdefs be moved into the library? The example also has TONS of Serial.print/println. Can't some of these be turned into code comments instead? And I'm sure lots of these lines could be shrunk with printf. The problem with dense examples like this is that it is difficult to skim through it and understand what it really does without really diving into it. And that defeats the purpose.

It would be awesome if you created a _dead simple_ example where you just write and read a few bytes and _that's it_, and perhaps create a more advanced example to show what the library is really capable of.

4. Ideas
To make the library even easier to use, how about creating flash read functions as well, that wraps `pgm_read_`? This would make the example code much more readable, and result in way fewer lines.

SpenceKonde, 2021-01-23:
>    1. Ideas

2. Because I want people to tell me which style they like better! No
intention of keeping both.

3. The hell? I tested both....

4. I... that is what I do! I write a few bytes, and then read it back >.< i
am clearly not very good at writing good demonstrations...

Ill fix the guards, and figure out what went wrong with the damned demo. At
least the release hasnt been "released"...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2021-01-24:
Thanks. 

I'm still baffled as to how that issue got through my testing (I thought I had tested it with that change) but I reworked that and put reading functionality into the library.. Also ripped out most of the example code and put a shorter less verbose one in. I'm crap at writing examples.

SpenceKonde, 2021-01-27:
Closing issue - other issue is still open and hoping for feedback.

closed 2021-01-27 by SpenceKonde