SpenceKonde/DxCore #58: Why would one turn off MVIO? (AVR DB)

SpenceKonde, 2021-01-24:
It looks to me like turning off the MVIO fuse is something that nobody would ever want to do except maybe in production. 

If you're running from a single supply, you have all the power pins (including the VDDIO2 pin) tied to VDD anyway. So the only way you could even tell if it was enabled is by checking the fusebits. 

If you're running from dual supplies, on the other hand, disabling MVIO would be an HCF type situation, where the part would suffer immediate hardware failure as it tried to equalize the voltages on the two power supplies. Does anyone want to make the case for my not removing that bit from the configuration options and DxCore always running any chip that supports it with MVIO enabled?

epccs, 2021-01-24:
Is it true that MVIO is enabled from the factory? I have not had a chance to verify that.

SpenceKonde, 2021-01-24:
I'll double check on a virgin DB... but 99% sure it is. It would have to be, otherwise for dual supply systems. you couldn't program the chips in system - without tying the two rails together long enough to turn off MVIO...

SpenceKonde, 2021-01-27:
1.3.x milestone - will remove MVIO from menu and always enable it in next version unless someone can convince me otherwise.

Miraculix200, 2021-01-29:
MVIO was enabled on my virgin AVR128DB32s.

SpenceKonde, 2021-02-03:
I've removed the menu option to disable MVIO. It does not appear to have a compelling use case. Let me know if anyone has a problem with that.

closed 2021-02-03 by SpenceKonde