SpenceKonde/DxCore #66: Can't write to PORTB ???

microPaul, 2021-02-02:
I'm probably doing something incredibly dumb, but I had a blink sketch that works on Spence's AVR128DA48 breakout board, where the LED is on A7, compiled and programmed via the Arduino IDE

I've upgraded to 1.3.0 via the board manager.

I've changed the LED references in the blink sketch from A7 to B3, since the Curiosity Nano has the LED on B3.  But when I try to compile it on the Arduino IDE the compiler complains about PORTB.  Specifically, "'PORTB' was not declared in this scope".  Do I need to include some sort of a .h file for the AVR128DB48?  I don't think I needed to do that for the AVR128DA48.

Here's my code:

`
//============================================
// Setup
//============================================
void setup() {
  PORTB.DIRSET = PIN3_bm; // PB3 is LED, set DDR bit for output
  PORTB.OUTSET = PIN3_bm;  // ensure LED off to begin
}

//============================================
// Super loop
//============================================
void loop() {
  ledOn();
  delay(1000);

  ledOff();
  delay(200);

}

//============================================
// LED ON/OFF functions, LED is active low
//============================================
void ledOff(void) {
  PORTB.OUTSET = PIN3_bm;  // LED off
}

void ledOn(void) {
  PORTB.OUTCLR = PIN3_bm;  // LED on
}
`

thanks,

Paul

SpenceKonde, 2021-02-02:
Can you please copy/paste the full text of the error - that does not make sense to me. and I don't immediately see why there would be a problem here...

microPaul, 2021-02-02:
Sure, here's what I get:

`p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void setup()':
CURNANO-AVR128DB48-blink-210201-01:6:3: error: 'PORTB' was not declared in this scope
   PORTB.DIRSET = PIN3_bm; // PB3 is LED, set DDR bit for output
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:6:3: note: suggested alternative: 'PORT0'
   PORTB.DIRSET = PIN3_bm; // PB3 is LED, set DDR bit for output
   ^~~~~
   PORT0
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void ledOff()':
CURNANO-AVR128DB48-blink-210201-01:29:3: error: 'PORTB' was not declared in this scope
   PORTB.OUTSET = PIN3_bm;  // LED off
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:29:3: note: suggested alternative: 'PORT0'
   PORTB.OUTSET = PIN3_bm;  // LED off
   ^~~~~
   PORT0
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void ledOn()':
CURNANO-AVR128DB48-blink-210201-01:33:3: error: 'PORTB' was not declared in this scope
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:33:3: note: suggested alternative: 'PORT0'
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
   PORT0
exit status 1
'PORTB' was not declared in this scope
An error occurred while uploading the sketch
WARNING: Error loading hardware folder P:\Users\p\Documents\Arduino\hardware\_
  No valid hardware definitions found in folder _.
WARNING: Error loading hardware folder P:\Users\p\Documents\Arduino\hardware\_
  No valid hardware definitions found in folder _.
`

microPaul, 2021-02-02:
Ugh!  Is it any more readable if I don't put it in the code brackets?  Here goes:

p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void setup()':
CURNANO-AVR128DB48-blink-210201-01:6:3: error: 'PORTB' was not declared in this scope
   PORTB.DIRSET = PIN3_bm; // PB3 is LED, set DDR bit for output
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:6:3: note: suggested alternative: 'PORT0'
   PORTB.DIRSET = PIN3_bm; // PB3 is LED, set DDR bit for output
   ^~~~~
   PORT0
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void ledOff()':
CURNANO-AVR128DB48-blink-210201-01:29:3: error: 'PORTB' was not declared in this scope
   PORTB.OUTSET = PIN3_bm;  // LED off
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:29:3: note: suggested alternative: 'PORT0'
   PORTB.OUTSET = PIN3_bm;  // LED off
   ^~~~~
   PORT0
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void ledOn()':
CURNANO-AVR128DB48-blink-210201-01:33:3: error: 'PORTB' was not declared in this scope
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:33:3: note: suggested alternative: 'PORT0'
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
   PORT0
exit status 1
'PORTB' was not declared in this scope
An error occurred while uploading the sketch
WARNING: Error loading hardware folder P:\Users\p\Documents\Arduino\hardware\_
  No valid hardware definitions found in folder _.
WARNING: Error loading hardware folder P:\Users\p\Documents\Arduino\hardware\_
  No valid hardware definitions found in folder _.

SpenceKonde, 2021-02-02:
File -> Preferences. 

Check the box next to "show verbose output when compiling". 

Attempt the compile and post **that** output. It will be f---ing huge. But it will make clear exactly what is and isn't getting pulled in and maybe tell us why. 

At this point I am inclined to suspect that there is something very wrong with your Arduino installation. 

Also - was this a board manager installation or a manual installation? If it was a manual installation, you did it wrong, according to the error message at the end (wrong folder structure? Something else?)

microPaul, 2021-02-02:
I did a board manager install of 1.3.0, but the prior install may have been manual, if that's a possible problem.  Also, I'd be happy to uninstall and reinstall the Arduino IDE if you think that might help.

Here's the verbose error message:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -hardware P:\Users\p\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries P:\Users\p\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8 -vid-pid=03EB_2175 -ide-version=10813 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_775364 -warnings=none -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_67185 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino3.path=C:\Users\p\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.python3.path=C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\p\AppData\Local\Arduino15\packages -hardware P:\Users\p\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\p\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries P:\Users\p\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,mvio=enabled,startuptime=8 -vid-pid=03EB_2175 -ide-version=10813 -build-path C:\Users\p\AppData\Local\Temp\arduino_build_775364 -warnings=none -build-cache C:\Users\p\AppData\Local\Temp\arduino_cache_67185 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\p\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino3.path=C:\Users\p\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3 -prefs=runtime.tools.avrdude.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.python3.path=C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\p\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\p\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino
Using board 'avrdb' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0
Using core 'dxcore' from platform in folder: C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0
Detecting libraries used...
"C:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino3/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.0\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=0UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.0\\cores\\dxcore/api/deprecated" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.0\\cores\\dxcore" "-IC:\\Users\\p\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.0\\variants\\28pin-standard" "C:\\Users\\p\\AppData\\Local\\Temp\\arduino_build_775364\\sketch\\CURNANO-AVR128DB48-blink-210201-01.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
[... cut ...]
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino: In function 'void ledOn()':
CURNANO-AVR128DB48-blink-210201-01:32:3: error: 'PORTB' was not declared in this scope
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
p:\Users\p\Documents\Arduino\CURNANO-AVR128DB48-blink-210201-01\CURNANO-AVR128DB48-blink-210201-01.ino:32:3: note: suggested alternative: 'PORT0'
   PORTB.OUTCLR = PIN3_bm;  // LED on
   ^~~~~
   PORT0
exit status 1
'PORTB' was not declared in this scope

MCUdude, 2021-02-02:
> -mmcu=avr128db28

You're compiling for the wrong chip. The verbose output reveals tht you're compiling for an AVR128DB28, not an AVR128DA48.

microPaul, 2021-02-02:
For what it's worth, I uninstalled the Arduino IDE, renamed the AppData/Local/Arduino15 to _Ardu_ino15, so the IDE would need to make a new Arduino15 file, then reinstalled the IDE (1.8.13), opened my DB48 blink program, compiled, but still got the error message about PORTB.  A forgot to mention last night, but if I change the four port references from PORTB to PORTA then the program compiles without error.  The functionality of the program is incorrect when I do that, but at least it will compile.

SpenceKonde, 2021-02-02:
Well, based on the compile log, as @MCUdude said, it is compiling for the 28-pin one, not the 48-pin one. The 28-pin one doesn't have a PORTB, so that error makes sense. Are you sure you have the right part selected? AVR128DB28 is the default chip selected when you choose to compile for AVR DB-series part; you need to then select the 48-pin one from the tools -> chip submenu.

microPaul, 2021-02-02:
I thought it was probably something dumb on my part. When I select the correct part (DB48) it compiles fine.  Sorry to have put you through that.  Next time I see this type of error I'll know what I'm doing wrong.

closed 2021-02-02 by microPaul

SpenceKonde, 2021-02-02:
Glad to hear it was something so simple. :-)