SpenceKonde/DxCore #67: Which Programmer for Curiosity Nano DB48?

microPaul, 2021-02-02:
In the instructions for programming, its says "NEW starting in 1.3.0... Starting in 1.3.0, DxCore includes a version of pymcuprog, written by Microchip - this adds support for two exciting new programming tools - Microchip nEDBG (used in, among other things, the Curiosity Nano boards), and Serial adapter + 4.7k resistor (like pyupdi)." 

In the Arduino IDE when I select "AVR DB-Series (no bootloader)" as the board, I don't see pymcuprog, and I don't anything that references the Curiosity Nano DB48.    Of the programmers that the IDE does offer, which is the best choice to use to program the DB48 Curiosity Nano?  Of the listed programmers, only one mentions "curiosity" but that's for the ATMSAMD21.

SpenceKonde, 2021-02-02:
Can you let me know where that quote is from? I need to change it, because between when I wrote it, and release, two big things changed: 
1. Speaking to the glue-code writer (who made the layer that goes between the core and pymcuprog - trying to make the IDE synthesize a command line invocation that would do that, I quickly determined, was utterly hopeless) it became apparent that my original plan - to use pymcuprog for everything except uploads through the bootloader, (minimizing use of avrdude, which I have a very low opinion of) was in fact not as practical as I thought - or at least, it would mean he'd have to do a low more work to get it access to the necessary USB libraries to make that work. 
2. Someone patched avrdude to support new versions of nEDBG (I was told, but never tried to verify, that it was someone from Microchip; would make sense - nobody who wasn't being paid (and paid well) to do so would touch that code). 

I had given up hope of there ever being a compatible version of avrdude - but now that there was, and my previous path towards support of curiosity nano being blocked, the solution was obvious. pymcuprog is only used for the pyupdi-like uploads. 

"pymcuprog" is the engine behind the "serial adapter and resistor" programmer (I don't name it that because nobody knows what that is anyway, they all know pyupdi - but the Microchip guys who we were talking to were very clear that they didn't want us to use pyupdi, and that we should use pymcuprog instead. It is definitely listed for me!

I think it's just the nEDBG/Curiosity one you mention. I'm told you can identify them by examining the debugging/programming chip (ie, if it's got an ATSAMD21 debugger chip, which I think the Curiosity boards do, you would select nEDBG, and so on).

microPaul, 2021-02-03:
In this page:  https://github.com/SpenceKonde/DxCore    Search for "NEW starting in 1.3.0"
I'd say it's about 40% down the page.

And yes, I found by trying the different program options that the one titled "Atmel nEDBG (ATSAM21E18/Curiosity)" does seem to program the AVR128DB48 Curiosity Nano.  My blink sketch is working!  

If you have control over how the Tools>Programmer is titled it might be more clear if it was something like
"Atmel nEDBG (ATSAM21E18, AVR128DB48 Curiosity)"

SpenceKonde, 2021-02-03:
Thanks - will correct that and look at the names for the programmers. The thing is, they sell Curiosity boards with a ton of different chips on them...

(Atmel? They go by "Microchip" now - like many other companies in the industry who have fallen before their endless pool of buyout funds.. 

Shame Microchip spent all their money buying competitors instead of investing in manufacturing capacity though. Last I heard their lead-times were sitting at around 1 year.... this is also why we don't have the tinyAVR 2-series yet (I think - either that, or they still can't get the ADC+PGA working well enough to ship, and why god only knows when we will see the AVR DD-series - which I very badly want)

microPaul, 2021-02-03:
Ok, thank you.  Well, I'm very happy to have the AVR128DB48.  I need the DAC for a lot of my projects, to make audio sine waves wih leading and trailing envelope control.  I can do that of course with the tinyAVR1s, but I also often need tones that are accurate to 1% or better, so that needs a clock with crystal accuracy, and the AVR-DB makes that easy to do (except that the HF xtal clobbers the primary serial port on the DB48, but the alt pins will provide a workaround for that).  I'm also thankful that I can now use the Arduino IDE with the DB48 Curiosity and don't have to use MPLAB-X-IDE or Studio 7 if I don't want to, and I usually don't want to.  There just seems to be so many gotchas with Microchips IDEs, and they are often like planting flowers with a backhoe when a small garden hand shovel will do well.  They are very powerful development tools, but my projects are not that complex.

As for silicon lead times, before I retired I was on the customer end of that, but we often bought so many parts across the corporation that we could bump other customers.  Not a nice thing to do, but it was sometimes done.  Also, leadtimes can sometimes vaporize, so maybe in fact they won't be as long is currently projected.  I hope so.

Thanks again for getting things cooking so that the Arduino IDE can compile and download programs to the DB48 CN.  That's going to make my life so much easier.

closed 2021-02-04 by SpenceKonde