SpenceKonde/DxCore #68: Can't compile Optiboot for AVR128DB64

Kozma04, 2021-02-03:
As the title suggests, I am having trouble compliling the bootloader for the AVR128DB64 MCU. When I run "make" in the optiboot folder in Ubuntu, the compilation ends with this:

"avr-gcc: error: unrecognized command line option ‘-mdevice=avr128db64".

Same goes with avr128da64 too.
Also, I had to completely remove the GCC path from Makefile in order to make avr-gcc start at all.

Anyway, here are my logs:
`Using Compiler at: /home/kozma/Desktop/DxCore-master/megaavr/bootloaders/optiboot_dx/avr-gcc
make -f  Makefile optiboot_dx128_ser0.hex UARTTX=A0 TIMEOUT=1 LED=A7 START_APP_ON_POR=1
make[1]: Entering directory '/home/kozma/Desktop/DxCore-master/megaavr/bootloaders/optiboot_dx'
Using Compiler at: /home/kozma/Desktop/DxCore-master/megaavr/bootloaders/optiboot_dx/avr-gcc
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax  -DWDTTIME=1  -DLED_START_FLASHES=3  -DLED=A7   -DUARTTX=A0 -DBAUD_RATE=115200   -DSTART_APP_ON_POR     -Wl,-section-start=.text=0 -Wl,--section-start=.application=0x200 -Wl,--section-start=.spmtarg=0x1fa -Wl,--section-start=.version=0x1fe -Wl,--relax -nostartfiles -nostdlib  -mmcu=avr128da64 -o optiboot_dx128_ser0.elf optiboot_x.c
avr-gcc: error: unrecognized command line option ‘-mdevice=avr128da64’
make[1]: *** [Makefile:143: optiboot_dx128_ser0.elf] Error 1
make[1]: Leaving directory '/home/kozma/Desktop/DxCore-master/megaavr/bootloaders/optiboot_dx'
make: *** [Makefile:159: dx128_ser0] Error 2`

SpenceKonde, 2021-02-03:
why is it passing -mdevice instead of -mmcu? And why is it trying to build for the DB part? 

```
C:\arduino-1.0.6-7.3.0compiler\hardware\arduino\bootloaders\optiboot_dx>..\..\..\tools\avr\utils\bin\make OS=windows ENV=arduino
Using Compiler at: C:/arduino-1.0.6-7.3.0compiler/hardware/tools/avr/bin/avr-gcc
C:/arduino-1.0.6-7.3.0compiler/hardware/tools/avr/utils/bin/make -f  makefile optiboot_dx128_ser0.hex UARTTX=A0 TIMEOUT=1 LED=A7 START_APP_ON_POR=1
Using Compiler at: C:/arduino-1.0.6-7.3.0compiler/hardware/tools/avr/bin/avr-gcc
make[1]: Entering directory `C:/arduino-1.0.6-7.3.0compiler/hardware/arduino/bootloaders/optiboot_dx'
../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax  -DWDTTIME=1  -DLED_START_FLASHES=3  -DLED=A7   -DUARTTX=A0 -DBAUD_RATE=115200   -DSTART_APP_ON_POR     -Wl,-section-start=.text=0 -Wl,--section-start=.application=0x200 -Wl,--section-start=.spmtarg=0x1fa -Wl,--section-start=.version=0x1fe -Wl,--relax -nostartfiles -nostdlib  -mmcu=avr128da64 -o optiboot_dx128_ser0.elf optiboot_x.c
../../../tools/avr/bin/avr-size optiboot_dx128_ser0.elf
   text    data     bss     dec     hex filename
    534       9       0     543     21f optiboot_dx128_ser0.elf
"../../../tools/avr/bin/avr-objdump" -S optiboot_dx128_ser0.elf > optiboot_dx128_ser0.lst
../../../tools/avr/bin/avr-objcopy -j .text -j .data -j .version -j .spmtarg --set-section-flags .version=alloc,load --set-section-flags .spmtarg=alloc,load -O ihex optiboot_dx128_ser0.elf optiboot_dx128_ser0.hex
make[1]: Leaving directory `C:/arduino-1.0.6-7.3.0compiler/hardware/arduino/bootloaders/optiboot_dx'

```
I build on Windows and don't know much about toolchain configuration - they're all designed to build if you copy the optiboot_dx folder into the hardware/arduino/bootloaders folder of an Arduino 1.0.6 windows installation with the 7.3 avr-gcc toolchain copied on top of the ancient one that 1.0.6 comes with >.>

SpenceKonde, 2021-02-03:
Anyway, I think the -mdevice is why it's erroring, it should be -mmcu

Kozma04, 2021-02-04:
Okay, I have done everything as you said, bur now I get an "unknown MCU 'avr128da64' specified" and "'avr128da64' supported for assembler only" when I try to compile Optiboot.

SpenceKonde, 2021-02-04:
Aaaha! That's progress! 

It sounds like you just need to install the latest AVR Dx-series ATpack from Microchip - the parts were just released a few months back The Arduino IDE doesn't come with working compiler for anything released after late spring (that would be the AVR1287DA, which was the latest part supported by the Arduino7 toolchain package..

But that's why when you install through board manager, it pulls in that bootloadertoolchain package? >.>

By the way, why are you trying to build Optiboot anyway? I include pre-built Optiboot binaries, so unless you need some strange special option or combination of options. the optiboot hex files included with megaTinyCore and DxCire (as appropriate) should have you covered...

Kozma04, 2021-02-04:
Thank you! I am trying to modify Optiboot to support flashing the program memory from an SD card by the user, through a UI displayed on the OLED connected to the MCU.

Kozma04, 2021-02-04:
Okay, I have a possibly trivial question. How do I install the AVR Dx-series ATpack in the Arduino compiler? All I know is that the ATpack can be opened as a zip file.

Kozma04, 2021-02-04:
Update: I tried defining and setting the PACKS variable to the pack folder in the Makefile, and now avr-gcc exists with error 0xc0000142.

SpenceKonde, 2021-02-04:
> Okay, I have a possibly trivial question. How do I install the AVR Dx-series ATpack in the Arduino compiler? All I know is that the ATpack can be opened as a zip file.

You don't. They don't have all the pieces for that. I assumed you had like a full non-arduino toolchain set up or something

In Arduino-land, You download a new arduino-friendly package that already has that support when you install a board package that requires one.; that's what board manager did when you installed DxCore. 

I hacked together the azduino3 version from the executables (from the perspective of your computer), and the AVR source, compiled object files, and so on freshly built. It is not a pleasant or graceful task. I don't *actually* know how to make a proper toolchain build, I only know how to create updated .o files and similar and replace those files in the arduino7 version of the toolchain. And I've figured out which files not to copy over because they don't work, too....

But if you just want to compile optiboot, why not just use the azduino3 toolchain package that board manager installs, which is what you use to compile sketches? That's what *I* build the bootloader with! (on windows, so I need the build tools from a 1.0.6 install, but that's a detail)

```

        {
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-azduino3",
          "systems": [
            {
              "size": "39937165",
              "checksum": "SHA-256:7c0ee3e6a0e2a69c58f8de17ba3e3c11ea2de691f30c0e9e305bf67dcdc71be6",
              "host": "i686-mingw32",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-w64-mingw32.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-w64-mingw32.tar.bz2"
            },
            {
              "size": "38550688",
              "checksum": "SHA-256:a6914d8a6e8fae1e56b1fc31dc0598c7fd715973af18702669872885121cdc75",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "35236732",
              "checksum": "SHA-256:b97b594ae54f12271dec5e7c3ebd9b5fb4b907fce77855cd55d792f8d7982315",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-arm-linux-gnueabihf.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-arm-linux-gnueabihf.tar.bz2"
            },
            {
              "size": "37829137",
              "checksum": "SHA-256:0ee13906e857fec854db3e500d849e5c7c1c5e1926ee27a85e3ae0b42c0e1b34",
              "host": "i686-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "37274152",
              "checksum": "SHA-256:a3a7756c0092d78ab19db0cd3fc91be77833ffc6d520396687987b39b7627fde",
              "host": "x86_64-apple-darwin14",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-apple-darwin14.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-apple-darwin14.tar.bz2"
            },
            {
              "size": "38518388",
              "checksum": "SHA-256:8dde3eba569cf4c0fbc9a772f6afb986d37f0f08796dccad08ab9815d4d074c0",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-pc-linux-gnu.tar.bz2"
            }
          ]
        }
```

And, you also already have that on your computer if you have DxCore installed and working - it's buried in the Arduino15 folder, but I think if you do compile with verbose output enabled, it will print the file paths. 

Whether you download fresh copy of it or use the one from my

Kozma04, 2021-02-04:
IT WORKED! Thank you.

closed 2021-02-04 by Kozma04