SpenceKonde/DxCore #69: Manual Installation Problem

btribley, 2021-02-04:
I am trying to get the latest commit to Master for bug fixes. I downloaded the Zip file and put it in the hardware folder:
```
bill@bill-i7x58 MINGW64 ~/Documents/Arduino/Hardware
$ ls
ChangeLog.md  Installation.md  LICENSE.md  README.md  Wiring.md  megaavr/
```
Arduino IDE complained that no hardware was found in megaavr, boards.txt not found. ???

```
bill@bill-i7x58 MINGW64 ~/Documents/Arduino/Hardware
$ ls megaavr
LICENSE.md    bootloaders/                     extras/       programmers.txt
avrdude.conf  cores/                           libraries/    tools/
boards.txt    create_disassembler_listing.sh*  platform.txt  variants/
```
Running the program, ADC0.CTRLA is still zero so I'm picking up the old installation. So I renamed 
C:\Users\bill\AppData\Local\Arduino15\packages\DxCoreNot
 and get this:
```
compilation terminated.
exit status 1
DxCore.h: No such file or directory
```
Oops. I can't select DxCore anymore either. Put it back...
Then I tried copying the newly downloaded tree here:
C:\Users\bill\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0
It's the same as what was there but now it can't find any of the header 

What am I doing wrong? Any help appreciated.

SpenceKonde, 2021-02-04:
The directory structure you want is `(sketchbook)/hardware/DxCore/...` not  `(sketchbook)/hardware/...`

btribley, 2021-02-04:
It's still picking up the installed version. Can I get rid of the installed version? Note this error shows old installed version (which I corrupted by copying the zip files in, I can reverse that):
```
C:\Users\bill\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0\cores\dxcore\wiring_analog.c:375:3: note: each undeclared identifier is reported only once for each function it appears in
exit status 1
```
I did "Show sketchbook folder" and got 
`C:\Users\bill\Documents\Arduino\TestI2c. `
I now have:
`C:\Users\bill\Documents\Arduino\TestI2c\hardware\DxCore\megaavr`
Compiler is still using files from the installed package

SpenceKonde, 2021-02-04:
You did "show sketch folder" which pointed you to the folder containing your sketch, which is inside your sketchbook folder

You want
C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr

btribley, 2021-02-04:
OK, I finally got Boards Manager to show DxCore(sketchbook). I then used Boards Manager to delete the installed package. Restarted everything,  I can still use the DxCore because it's in the sketchbook. Now the compiler errors out in the libraries, which are being pulled from the sketchbook:
```
Arduino: 1.8.13 (Windows 10), Board: "AVR DA-series (no bootloader), AVR128DA28, 24 MHz internal, No, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2, 8ms"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\bill\AppData\Local\Arduino15\packages -hardware C:\Users\bill\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\bill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bill\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrda:chip=128DA28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,startuptime=8 -ide-version=10813 -build-path C:\Users\bill\AppData\Local\Temp\arduino_build_234905 -warnings=none -build-cache C:\Users\bill\AppData\Local\Temp\arduino_cache_4702 -prefs=build.warn_data_percentage=75 -verbose C:\Users\bill\Documents\Arduino\TestI2c\TestI2c.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\bill\AppData\Local\Arduino15\packages -hardware C:\Users\bill\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\bill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bill\Documents\Arduino\libraries -fqbn=DxCore:megaavr:avrda:chip=128DA28,clock=24internal,serialevent=no,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,startuptime=8 -ide-version=10813 -build-path C:\Users\bill\AppData\Local\Temp\arduino_build_234905 -warnings=none -build-cache C:\Users\bill\AppData\Local\Temp\arduino_cache_4702 -prefs=build.warn_data_percentage=75 -verbose C:\Users\bill\Documents\Arduino\TestI2c\TestI2c.ino

Using board 'avrda' from platform in folder: C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr

Using core 'dxcore' from platform in folder: C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.1-dev\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=1UL -DDXCORE_RELEASED=0 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "C:\\Users\\bill\\AppData\\Local\\Temp\\arduino_build_234905\\sketch\\TestI2c.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for Wire.h: [Wire@1.1.0]

ResolveLibrary(Wire.h)

  -> candidates: [Wire@1.1.0]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.1-dev\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=1UL -DDXCORE_RELEASED=0 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\Wire\\src" "C:\\Users\\bill\\AppData\\Local\\Temp\\arduino_build_234905\\sketch\\TestI2c.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.1-dev\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=1UL -DDXCORE_RELEASED=0 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore/api/deprecated" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\cores\\dxcore" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\variants\\28pin-standard" "-IC:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\Wire\\src" "C:\\Users\\bill\\Documents\\Arduino\\hardware\\DxCore\\megaavr\\libraries\\Wire\\src\\Wire.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

[... cut ...]
C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr\cores\dxcore\wiring_analog.c:375:3: error: 'TCA1' undeclared (first use in this function); did you mean 'TCA0'?

   TCA1.SPLIT.CTRLA = 0;          // Stop TCA0

   ^~~~

   TCA0

C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr\cores\dxcore\wiring_analog.c:375:3: note: each undeclared identifier is reported only once for each function it appears in

Using library Wire at version 1.1.0 in folder: C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr\libraries\Wire 

exit status 1

Error compiling for board AVR DA-series (no bootloader).

```

SpenceKonde, 2021-02-04:
Thank you,

Not actually related to the libraries, those are working fine, but I did
all my testing on 64-pin part, need some #ifdefs around stuff that
references TCA1.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: spencekonde@gmail.com

btribley, 2021-02-04:
I just commented the stuff out. Compiles fine. Now this:
```
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

An error occurred while uploading the sketch
```
First time I have gotten this with Arduino IDE. I have the programmer set to Atmel-ICE(UPDI) .

btribley, 2021-02-04:
I will grab your latest commit and try it. Thanks!

btribley, 2021-02-04:
Latest commit compiles. Upload error remains. Going to restart everything.

SpenceKonde, 2021-02-04:
Haven't changed anything there, does the version installed by board manager upload successfully?

btribley, 2021-02-04:
Everything has been uploading successfully until this last shot. Can I have both the board manager and the hardware one installed? Before when I did that the system was ignoring the sketchbook version.

btribley, 2021-02-04:
Detailed error message, I don't see anything wrong with it, signatures match the data sheet. Comparing avrdude and platform files shows no difference. Have to hang it up for the night, day job.... Thanks for taking time to help tonight.

```
Arduino: 1.8.13 (Windows 10), Board: "AVR DA-series (no bootloader), AVR128DA28, 24 MHz internal, No, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2, 8ms"

Sketch uses 4484 bytes (3%) of program storage space. Maximum is 131072 bytes.

Global variables use 656 bytes (4%) of dynamic memory, leaving 15728 bytes for local variables. Maximum is 16384 bytes.

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr/avrdude.conf -v -pavr128da28 -catmelice_updi -Pusb -Ufuse5:w:0b11001001:m -Ufuse8:w:0x00:m -Uflash:w:C:\Users\bill\AppData\Local\Temp\arduino_build_569201/TestI2c.ino.hex:i 

avrdude: Version 6.3-20190619

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\bill\Documents\Arduino\hardware\DxCore\megaavr/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : atmelice_updi

avrdude: Found CMSIS-DAP compliant device, using EDBG protocol

         AVR Part                      : AVR128DA28

         Chip Erase delay              : 0 us

         PAGEL                         : P00

         BS2                           : P00

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 0

         StabDelay                     : 0

         CmdexeDelay                   : 0

         SyncLoops                     : 0

         ByteDelay                     : 0

         PollIndex                     : 0

         PollValue                     : 0x00

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00

           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00

           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00

           lock           0     0     0    0 no          4    0      0     0     0 0x00 0x00

           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00

           flash          0     0     0    0 no     131072  512      0     0     0 0x00 0x00

           eeprom         0     0     0    0 no        512   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI

         Description     : Atmel-ICE (ARM/AVR) in UPDI mode

         ICE hardware version: 0

         ICE firmware version: 1.41 (rel. 137)

         Serial number   : J42700027192

         Vtarget         : 3.29 V

         JTAG clock megaAVR/program: 1000 kHz

         JTAG clock megaAVR/debug:   1000 kHz

         JTAG clock Xmega: 1000 kHz

An error occurred while uploading the sketch

         PDI clock Xmega : 500 kHz

avrdude: bad response to AVR sign-on command: 0xa0

avrdude: retrying with external reset applied

avrdude: bad response to AVR sign-on command: 0xa0

avrdude: retrying with external reset applied

avrdude: JTAGEN fuse disabled?

avrdude: initialization failed, rc=-1

         Double check connections and try again, or use -F to override

         this check.

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

btribley, 2021-02-04:
Reinstalled from Board Manager, renamed hardware directory in the sketchbook directory to use the board manager. The uplolader works:
```
Sketch uses 4366 bytes (3%) of program storage space. Maximum is 131072 bytes.
Global variables use 656 bytes (4%) of dynamic memory, leaving 15728 bytes for local variables. Maximum is 16384 bytes.
C:\Users\bill\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\bill\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0/avrdude.conf -v -pavr128da28 -catmelice_updi -Pusb -Ufuse5:w:0b11001001:m -Ufuse8:w:0x00:m -Uflash:w:C:\Users\bill\AppData\Local\Temp\arduino_build_923519/TestI2c.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\bill\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.0/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : atmelice_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AVR128DA28
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          4    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           flash          0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        512   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel-ICE (ARM/AVR) in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.41 (rel. 137)
         Serial number   : J42700027192
         Vtarget         : 3.29 V
         JTAG clock megaAVR/program: 1000 kHz
         JTAG clock megaAVR/debug:   1000 kHz
         JTAG clock Xmega: 1000 kHz
         PDI clock Xmega : 500 kHz

avrdude: Partial Family_ID returned: "    "
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e970a (probably avr128da28)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
[... 24 lines cut ...]
avrdude: reading on-chip fuse8 data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of fuse8 verified
avrdude: reading input file "C:\Users\bill\AppData\Local\Temp\arduino_build_923519/TestI2c.ino.hex"
avrdude: writing flash (4366 bytes):

Writing | ################################################## | 100% 0.30s

avrdude: 4366 bytes of flash written
avrdude: verifying flash memory against C:\Users\bill\AppData\Local\Temp\arduino_build_923519/TestI2c.ino.hex:
avrdude: load data flash data from input file C:\Users\bill\AppData\Local\Temp\arduino_build_923519/TestI2c.ino.hex:
avrdude: input file C:\Users\bill\AppData\Local\Temp\arduino_build_923519/TestI2c.ino.hex contains 4366 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.14s

avrdude: verifying ...
avrdude: 4366 bytes of flash verified

avrdude done.  Thank you.

```

btribley, 2021-02-04:
Short answer: Works with the released DxCore but doesn't with the one in Sketchbook. I compared the commands to avrdude. The Sketchbook one uses this:
`C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude `
The released package installed with Board Manager uses this:
`C:\Users\bill\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude` 
Any easy way to coerce the sketchbook version to use the same one as the board manager version? I will look into this tonight, any suggestions appreciated.  I may just copy from the working to the non-working after moving the one in the non-working off to the side.

SpenceKonde, 2021-02-04:
That is expected (you'll see the same thing with the compiler toolchain, I suspect - try compiling for DB part with the sketchbook version). 
Without the board manager json file's tools-dependencies section to tell it to use a specific version of the compiler and a specific version of avrdude, it will use the (incompatible) version the IDE came with.

Just copy&replace. That's what I do for the toolchain (avrdude using jtag2updi doesn't have the dependency on avrdude version, and I don't own any Microchip programmers)

SpenceKonde, 2021-02-04:
Closing as this does not, at this point, appear to indicate a defect in the core.

closed 2021-02-04 by SpenceKonde

btribley, 2021-02-04:
Thanks for your help. 

What you are saying is that if a Core developer puts the Core in Sketchbook to work on it they are using a different set of dependencies than will be present when the released core is loaded by board manager. This is a recipe for latent bugs and it prevents effective testing. 

Wondering out loud if it would be possible to have a json file for Sketchbook hosted hardware?

SpenceKonde, 2021-02-04:
Yes, that is what I am saying - though I think it's only this weird for AVR. 

I think for everything else, you install something that uses the same toolchain through board manager, and then the sketchbook cores will use that because that's the only game in town.

Someone maintaining a third party hardware package, though, is a sophisticated user, (if I do say so myself) - having to copy a directory isn't really a barrier compared to the sort of challenges that are routine in the course of developing and maintaining a core. It's easier than implementing the arduino timekeeping API on new hardware, certainly easier than porting self-programming to a new nvm version, and it's easier than reading the Timer/Counter Type D chapter of the datasheet without feeling light-headed and ill.

Personally, I never trust the board manager.

My local environment is:
* latest (or latest good, when those are different) IDE with toolchain updated to the latest one I'm using (eg, azduino3 version of toolchain package, which was made for DxCore 1.3.0_). Board manager has never been used to install anything.
* same version (as above), only portable install, used for board manager. 
* same version, portable install, but sketchbook set to my sketchbook folder.
* 1.0.6, toolchain replaced with last 5.4.x gcc for building classic AVR bootloaders. 
* 1.0.6, toolchain replaced with latest for building dx and tinyAVR 0/1/2-series bootloaders.
I forget which of the toolchains I was using for Micronucleus bootloaders, heh...

btribley, 2021-02-04:
Spence, thank you very much for writing this. It helps me understand how to work with the DxCore and how to ditch board manager completely. It may make it much easier to use Atmel Studio 7 with Arduino.

SpenceKonde, 2021-02-04:
I think most people who are serious about it use a better IDE than the Arduino one - but I have never really had the refined tastes in IDEs that most people seem to :-P

btribley, 2021-02-04:
All I want is the hardware debugger. Atmel Studio 7 gives me that and it seems to work even if you first use Arduino IDE to set things up! I want to investigate how all the board settings are actually stored in the code.  VisualMicro is flat-out broken, stuff that worked in Atmel Studio 7 broke when I installed VisualMicro. Some help.....

btribley, 2021-02-04:
But tonight I am expecting to see analog to digital conversions, that's the big prize right now!

SpenceKonde, 2021-02-04:
I think the hardware debuggers are all locked to the official tooling (maybe it was that they didn't release the protocol specs? The D part of UPDI isn't described in the datasheets at least...)