SpenceKonde/DxCore #70: Serial.print(byte,HEX) Issue

btribley, 2021-02-05:
Code:
```
  Serial2.print("Temperature: ");
  Wire.requestFrom(tempSenseWriteAddr, 2);    
  while (Wire.available()) { // slave may send less than requested
    char c = Wire.read(); // receive a byte as character
    Serial2.print(c,HEX);         // print the character
    Serial2.print('.');
  }
  Serial2.println(" deg");

```
Output at first:
```
Temperature: 15.68. deg                                                     
Temperature: 15.64. deg                                                     
Temperature: 15.64. deg                                                     
Temperature: 15.6C. deg                                                     
Temperature: 15.68. deg                                                    
Temperature: 15.6C. deg                                                    
Temperature: 15.6C. deg                                                    
Temperature: 15.74. deg                                                   
Temperature: 15.70. deg                                                  
Temperature: 15.78. deg
Temperature: 15.78. deg                                                   
Temperature: 15.78. deg                                                   
Temperature: 15.FFFFFF80. deg                                             
Temperature: 15.7C. deg                                                
Temperature: 15.78. deg                                                  
Temperature: 15.FFFFFF84. deg                                           
Temperature: 15.FFFFFF88. deg                                           
Temperature: 15.FFFFFF88. deg                                           
Temperature: 15.FFFFFF84. deg

```
After a period of time:
```
Temperature: 14.FFFFFFD0. deg                               
Temperature: 14.FFFFFFCC. deg                                                
Temperature: 14.FFFFFFDC. deg                                                
Temperature: 14.FFFFFFD8. deg                 
Temperature: 14.FFFFFFD0. deg
```
Still longer:
Temperature: 15.3C. deg                                                  
Temperature: 15.38. deg                                                  
Temperature: 15.3C. deg                                                
Temperature: 15.40. deg                                                      

The production release doesn't do this. I never saw the 4 byte output.

SpenceKonde, 2021-02-05:
It is unclear to me what expected behavior and the wrong behavior is here, and also why this is a bug in the core and/or an included library. 

> The production release doesn't do this. I never saw the 4 byte output.

Please explain what you mean by this.

This is really a rather concerning issue - I see two potential outcomes:
1. This is a bug in user code or device interaction not meeting developer expectations. 
2. There is a deep and really nasty bug, possibly relating to Wire and Serial operating simultaneously. 

Fortunately, at this point, my response to both of these is the same: additional debugging effort is required. Right now, it's not even clear to me whether Serial or Wire is fucking up! Serial.print returns the number of characters printed. On thing I'd start doing is setting a flag if any of those serial prints returns an unexpected value. We may not be able to rely on Serial during the relevant times. Maybe we have to have a global byte array and, eg:
```
(in global scope)
byte log_array[2000]; //some value so high that you *know* the code will never fill - we have oodles of ram here.

int i=0;
while Wire.available() 
  char c =Wire.read();
  log_array[i++]=c;
  log_array[i++]=Serial2.print(c,HEX);
  log_array[i++]=Serial2.print(".");
}
delay(100); // if this is actually some thing deep and terrible, you may want to wait until serial buffer has emptied before you try to get data out. 
for (int j=0; j < i ; j++) {
Serial.print(j);
Serial.print(':");
Serial.println(log_array[j]);
}
```

I hope it's clear what my goal there is; you may not need to go to such lengths, but, if this isn't something on your end, this is going to be a very serious and very challenging bug (I can't remember the last time something frightening in this way has turned up - I feel like the guy in a movie when they see something that doesn't make sense - because they've been walking into a trap for the first 30 minutes, and they're like "no that can't be" and the camera pans out and forboding notes play in the soundtrack - because the landscape they've been walking over is actually an alien lifeform, and they just saw a rock move or something), 

And if it is a defect in the core, it also impacts megaTinyCore, and it is likely a bug in either the bowels of the TWI code that Wire wraps, or the rewritten UART.c code (which I had found several potential crash conditions in - in addition to the code doing truly perverse things and being a monstrous flash hog on megaTinyCore). I have a bad feeling that there is indeed a bug here, you are not the first person who had strange problems with wire. But yeah - I need to go give ATTinyCore some love after I test a few last things and slam out 1.3.1 - they haven't had a release since august and desperately need some attention (and their userbase is larger than DxCore and megaTinyCore combined, even if you double-count the people who use both DxCore and megaTinyCore)

SpenceKonde, 2021-02-05:
But if it does end up being a Wire + Serial thing... we're going to have a problem in terms of my not having a known way to reproduce this because of a dependence on external hardware.

btribley, 2021-02-05:
Wire is working. It is getting a temperature. The LSB of the 4 byte output is correct. But Wire is isolated from Serial by the storage.

In this case, I am defining a simple char then doing a HEX println. The hex output of a char should never be more than two ASCII characters.

It's value dependent. And another big bad bug is 0 - 15 is only a single digit. Someone will argue that this is expected but I disagree. The hex equivalent of an 8 bit byte is always two hex digits.
Test code:
```
  char i=0; 
  while (i <0xff) {
    for(char j=0;j<16;j++)
      Serial2.print(i++,HEX);
    Serial2.println();
  }

```
Result:
```
0123456789ABCDEF
                                                                                                                                                
101112131415161718191A1B1C1D1E1F
                                                                                                                                
202122232425262728292A2B2C2D2E2F
                                                                                                                                
303132333435363738393A3B3C3D3E3F
                                                                                                                                
404142434445464748494A4B4C4D4E4F
                                                                                                                                
505152535455565758595A5B5C5D5E5F
                                                                                                                                
606162636465666768696A6B6C6D6E6F
                                                                                                                                
707172737475767778797A7B7C7D7E7F
                                                                                                                                
FFFFFF80FFFFFF81FFFFFF82FFFFFF83FFFFFF84FFFFFF85FFFFFF86FFFFFF87FFFFFF88FFFFFF89FFFFFF8AFFFFFF8BFFFFFF8CFFFFFF8DFFFFFF8EFFFFFF8F
                                
FFFFFF90FFFFFF91FFFFFF92FFFFFF93FFFFFF94FFFFFF95FFFFFF96FFFFFF97FFFFFF98FFFFFF99FFFFFF9AFFFFFF9BFFFFFF9CFFFFFF9DFFFFFF9EFFFFFF9F
                                
FFFFFFA0FFFFFFA1FFFFFFA2FFFFFFA3FFFFFFA4FFFFFFA5FFFFFFA6FFFFFFA7FFFFFFA8FFFFFFA9FFFFFFAAFFFFFFABFFFFFFACFFFFFFADFFFFFFAEFFFFFFAF
                                
FFFFFFB0FFFFFFB1FFFFFFB2FFFFFFB3FFFFFFB4FFFFFFB5FFFFFFB6FFFFFFB7FFFFFFB8FFFFFFB9FFFFFFBAFFFFFFBBFFFFFFBCFFFFFFBDFFFFFFBEFFFFFFBF
                                
FFFFFFC0FFFFFFC1FFFFFFC2FFFFFFC3FFFFFFC4FFFFFFC5FFFFFFC6FFFFFFC7FFFFFFC8FFFFFFC9FFFFFFCAFFFFFFCBFFFFFFCCFFFFFFCDFFFFFFCEFFFFFFCF
                                
FFFFFFD0FFFFFFD1FFFFFFD2FFFFFFD3FFFFFFD4FFFFFFD5FFFFFFD6FFFFFFD7FFFFFFD8FFFFFFD9FFFFFFDAFFFFFFDBFFFFFFDCFFFFFFDDFFFFFFDEFFFFFFDF
                                
FFFFFFE0FFFFFFE1FFFFFFE2FFFFFFE3FFFFFFE4FFFFFFE5FFFFFFE6FFFFFFE7FFFFFFE8FFFFFFE9FFFFFFEAFFFFFFEBFFFFFFECFFFFFFEDFFFFFFEEFFFFFFEF
                                
FFFFFFF0FFFFFFF1FFFFFFF2FFFFFFF3FFFFFFF4FFFFFFF5FFFFFFF6FFFFFFF7FFFFFFF8FFFFFFF9FFFFFFFAFFFFFFFBFFFFFFFCFFFFFFFDFFFFFFFEFFFFFFFF
                  

```

btribley, 2021-02-05:
Fortunately in my big application I have my own hex converter. The lame PIC compiler didn't even have printf so I had to write a poor man's version....

SpenceKonde, 2021-02-05:
Oh.... well that's anticlimatic, lol... 

Serial.print() not printing leading zeros of hex bytes is correct and expected behavior. Does it suck? Yes! Is it obnoxious and almost always undesirable? For sure. But it is widely known, documented, and disparaged behavior of the standard Arduino code. I am not going to change that.

What I did about the incredibly irritating lack of leading zeros was to create the printHex() methods. I can't guarantee the calling conventions won't change (there are things about them that I'm deeply, profoundly unhappy about, and also embarrassed about) - though I think only the pointer versions are guaranteed to get this added...); the main thing that is frustrating about my printHex() methods is that they lack an equivalent to println() so you end up with Serial.printHex(a); Serial.println(); Serial,printHex(b); Serial.println(); type constructions.

btribley, 2021-02-05:
I didn't know about printHex. I will test that one. The built-in one is unusable with the 4 byte output.

btribley, 2021-02-05:
printHex works, outputs a full int every time which anyone can live with:
0000 0001 0002 0003 0004 0005 0006 0007 0008 0009 000A 000B 000C 000D 000E 000F 
                                                                                
0010 0011 0012 0013 0014 0015 0016 0017 0018 0019 001A 001B 001C 001D 001E 001F 
                                                                                
0020 0021 0022 0023 0024 0025 0026 0027 0028 0029 002A 002B 002C 002D 002E 002F 
                                                                                
0030 0031 0032 0033 0034 0035 0036 0037 0038 0039 003A 003B 003C 003D 003E 003F 
                                                                                
0040 0041 0042 0043 0044 0045 0046 0047 0048 0049 004A 004B 004C 004D 004E 004F 
                                                                                
0050 0051 0052 0053 0054 0055 0056 0057 0058 0059 005A 005B 005C 005D 005E 005F 
                                                                                
0060 0061 0062 0063 0064 0065 0066 0067 0068 0069 006A 006B 006C 006D 006E 006F 
                                                                                
0070 0071 0072 0073 0074 0075 0076 0077 0078 0079 007A 007B 007C 007D 007E 007F 
                                                                                
FF80 FF81 FF82 FF83 FF84 FF85 FF86 FF87 FF88 FF89 FF8A FF8B FF8C FF8D FF8E FF8F 
                                                                                
FF90 FF91 FF92 FF93 FF94 FF95 FF96 FF97 FF98 FF99 FF9A FF9B FF9C FF9D FF9E FF9F 
                                                                                
FFA0 FFA1 FFA2 FFA3 FFA4 FFA5 FFA6 FFA7 FFA8 FFA9 FFAA FFAB FFAC FFAD FFAE FFAF 
                                                                                
FFB0 FFB1 FFB2 FFB3 FFB4 FFB5 FFB6 FFB7 FFB8 FFB9 FFBA FFBB FFBC FFBD FFBE FFBF 
                                                                                
FFC0 FFC1 FFC2 FFC3 FFC4 FFC5 FFC6 FFC7 FFC8 FFC9 FFCA FFCB FFCC FFCD FFCE FFCF 
                                                                                
FFD0 FFD1 FFD2 FFD3 FFD4 FFD5 FFD6 FFD7 FFD8 FFD9 FFDA FFDB FFDC FFDD FFDE FFDF 
                                                                                
FFE0 FFE1 FFE2 FFE3 FFE4 FFE5 FFE6 FFE7 FFE8 FFE9 FFEA FFEB FFEC FFED FFEE FFEF 
                                                                                
FFF0 FFF1 FFF2 FFF3 FFF4 FFF5 FFF6 FFF7 FFF8 FFF9 FFFA FFFB FFFC FFFD FFFE FFFF

btribley, 2021-02-05:
A char is 8 bits. Output should be 00 .. FF but it's 4 bytes. Here is the test code:
```
  char i=0; 
  while (i <0xff) {
    for(char j=0;j<16;j++)
    {
      Serial2.printHex(i++);
      Serial2.print(' ');
    }
    Serial2.println();
  }

```

btribley, 2021-02-05:
I will try uint8_t. What you said about using 26 registers for a port is whacko!

btribley, 2021-02-05:
uint8_t i instead of char fixed it for printHex, so much for their types:
```
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
                                                                                                                
10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 
                                                                                                                
20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 
                                                                                                                
30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 
                                                                                                                
40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 
                                                                                                                
50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 
                                                                                                                
60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 
                                                                                                                
70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 
                                                                                                                
80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 
                                                                                                                
90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F 
                                                                                                                
A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF 
                                                                                                                
B0 B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF 
                                                                                                                
C0 C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF 
                                                                                                                
D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF 
                                                                                                                
E0 E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF 
                                                                                                                
F0 F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FB FC FD FE FF 
                

```

btribley, 2021-02-05:
uint8_t also works for print(i,HEX). Stupid char is not an unsigned int. Case closed, sorry to bug you.

SpenceKonde, 2021-02-05:
Just look in UART.h in the core. 

Also, there is definitely still a bug here with normal print.

SpenceKonde, 2021-02-07:
Okay, that sounds like it's actually just a quirk of C here o_O

closed 2021-02-07 by SpenceKonde