SpenceKonde/DxCore #71: Comparator.cpp:5:54: error: 'PORTE' was not declared in this scope

Miraculix200, 2021-02-05:
I'm not sure what I'm doing here, just started to play around with the comparator of my AVR128DB32. But I'm quite sure this isn't supposed to happen, as the AVR128DB32 doesn't have a PORTE:

`DxCore\hardware\megaavr\1.2.0-a\libraries\Comparator\src\Comparator.cpp:5:54: error: 'PORTE' was not declared in this scope
 AnalogComparator Comparator0(0, AC0, PORTD.PIN2CTRL, PORTE.PIN0CTRL, PORTE.PIN2CTRL, PORTD.PIN6CTRL, PORTD.PIN3CTRL, PORTD.PIN0CTRL, PORTD.PIN7CTRL);
                                                      ^~~~~
`

when trying to compile this: https://gist.github.com/Miraculix200/88d819a550be509b245e4fb95f6932db

SpenceKonde, 2021-02-05:
1.2.0-a? 

Current version of DxCore is 1.3.0 - though that bug is still there!

I don't hand-verify everything (even in terms of compile testing) on low-pin-count parts - and in this case the references to PORTE weren't properly #ifdef'ed out. Though in 1.2.0-a, even on the 64-pin parts, comparator didn't compile, there was some other wrong bitfield name or something. 

Should be simple to fix; will be in the already-planned 1.3.1 release coming tonight. 

This is why we need GODDAMNED CI! But I have 50 million things to do and don't really know yaml whatever the fuck that is, and don't know how to adapt the CI code from megaTinyCore to make it 1) useful on megaTinyCore, and 2)make it work here
This is one of the few bugs that would be caught very efficiently by CI.

Miraculix200, 2021-02-05:
Comparator did compile on 1.3.0 after I replaced the PORTEs with random ports. Now it doesn't work as expected, but that's another problem :)

SpenceKonde, 2021-02-05:
Huh! Interesting - well, I wasn't planning to fix anything that would cause it to hang, nor can I see anything in Comparator.init() that could make it hang! So without more information, it will still hang after 1.3.1

Are you using an interrupt? Does getting rid of that make it not hang? Could there be a bug in your interrupt?

Miraculix200, 2021-02-05:
I thought it hangs because was making a mess in Comparator.cpp. Maybe it's another cause? But it also hangs without using the interrupt.  Anyway, I'll wait for 1.3.1 and then create another bug report if the problem still occurs.

Miraculix200, 2021-02-05:
Btw, for the future (after 1.3.1) you may also want to add VDDIO2 as a reference.

SpenceKonde, 2021-02-07:
Don't thank me *shrug* thank this dude... this - MCUdude

@MCUdude - Okay, I'm looking at this and I'm trying to figure out a graceful way to deal with the pins that don't exist on the parts without a PORTE.... 
Do you have any thoughts on how I should do this without making a heinous mess out of your code?  I would need to do this over on megaTinyCore when I finally have time to port this to the tinyAVR parts. 

The problem here though is just that I still need to pass it *something* that's a register8_t, Like, should I cook up a NOT_A_REGISTER #define that's a pointer to... Maybe one of those 4 reserved bytes in the sigrow that are always 0,? Then even if someone wound up "writing' to them, it's not writable by the application and wouldn't do any damage. I feel like you will probably have a much better suggestion here. Biut time runs short - 1.3.1 has to be released this evening so I can start work on ATTinyCore - people there are getting restless

Miraculix200, 2021-02-07:
No hurry This is a new MCU, so this has to be expected. For now I'm playing around with the 4808 comparator instead.

SpenceKonde, 2021-02-08:
aaa6a001d51f14c889be263af7bfbe8214fdce8e

closed 2021-02-08 by SpenceKonde