SpenceKonde/DxCore #75: Upload to Curiosity Nano

sstaub, 2021-02-08:
The upload to the Curiosity Nano board work fine with Arduino IDE 1.8.13 on macOS Big Sur
but fails with the Arduino Pro IDE 0.13 nightly20210205
`Upload error: Error: 2 UNKNOWN: uploading error: uploading error: exit status 1`
 It seems that the Pro IDE doesn't know the programmer.

SpenceKonde, 2021-02-08:
Sounds like a bug with the Pro IDE, why don't you report it to that development team?

closed 2021-02-08 by SpenceKonde

sstaub, 2021-02-08:
Ok, I will make an issue.

sstaub, 2021-02-08:
Which version of avrdude did you use?
ProIDE says it is missing the programmer ID:
```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is
"/Users/sstaub/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.0/avrdude.conf"
         User configuration file is "/Users/sstaub/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

avrdude: Can't find programmer id "{upload.protocol}"
```

sstaub, 2021-02-08:
I got a message from the Arduino PRO IDE team, you must use explicit "Upload Using Programmer" in the Sketch Menue. Sorry, I found this problem also using the Curiosity Nano 4809 board a while ago.

per1234, 2021-02-08:
Yes, this is a bug in Arduino Pro IDE (https://github.com/arduino/arduino-pro-ide/issues/309), and not caused by anything wrong with DxCore.