SpenceKonde/DxCore #82: Question about usage of symbols in cores

technoblogy, 2021-03-04:
My program for the AVR128DA48 or AVR128DB48 compiled and ran fine on DxCore 1.3.0 and 1.3.1, but failed with compile errors on 1.3.2.

The errors were:
````text
In file included from /Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/cores/dxcore/api/ArduinoAPI.h:52:0,
                 from /Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/cores/dxcore/Arduino.h:23,
                 from /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_880885/sketch/Lisp170.ino.cpp:1:
/Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/cores/dxcore/api/Common.h:10:26: error: expected identifier before numeric constant
 #define HIGH             1
                          ^
/Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/cores/dxcore/api/Common.h:11:23: note: in expansion of macro 'HIGH'
 #define FLOAT         HIGH
                       ^~~~
````
The problem seems to be that I used the symbol FLOAT in my program. Renaming that fixed the problem.

But this raises a more general question: is there any convention about what symbols cores can define, and what we should avoid in our programs, to avoid this happening again? Should there be a naming convention for new symbols, such as \_FLOAT\_, in cores?

SpenceKonde, 2021-03-12:
Yeah, I just couldn't think of a way to do the openDrain in a way that didn't mislead, and had consistency with the other digitial I/O functions - other than defining a new constant. I wasn't happy about doing that, but I really really wanted to put in the openDrain() functions - it's a gaping hole in the stock digital I/O

Didn't have time to get pinConfig ready - that's the other digital I/O function we realllly need. (this is a single function call that takes a uint16_t and can set, unset, toggle or leave unchanged direction, output value, pullup state, interrupt sense configuration, inversion, and for the DB-series, input logic levels. But the #defines i'll need for that are less likely to step on other people's toes - they'll all be prefixed with something obvious.

technoblogy, 2021-03-12:
Great - thanks!

closed 2021-03-12 by technoblogy