SpenceKonde/DxCore #83: DACReference() doesn't seem to work

technoblogy, 2021-03-07:
Assuming I'm calling it correctly, the DAC reference voltage always seems to be VDD whatever I do.

Test program:
````text
void setup() {
  analogWrite(18, 255); // DAC output goes to 5.0V
  delay(1000);
  analogWrite(18, 0);   // DAC output goes to 0V
  delay(1000);
  DACReference(INTERNAL1V024);
  analogWrite(18, 255); // DAC output goes to 5.0V
  delay(1000);
  analogWrite(18, 0);   // DAC output goes to 0V
  delay(1000);
  DACReference(INTERNAL2V048);
  analogWrite(18, 255); // DAC output goes to 5.0V
  delay(1000);
  analogWrite(18, 0);   // DAC output goes to 0V
  delay(1000); 
}
````
I've tried it on an AVR128DA28 and an AVR128DB28.

By the way, the documentation says:

> The DA-series parts have a 10-bit DAC ...

This should say "These parts ...".

SpenceKonde, 2021-03-07:
Does it work if you pass 254? I dont think I ever made an exception to the
255 or 0 to analog write just calls digitalwrite rule for the dac..

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

technoblogy, 2021-03-07:
Yes, that works!

SpenceKonde, 2021-03-12:
Nice okay, well, hmmm.... yeah  I agree; when analogWrite is used to write a 255 to the DAC pin, it should not call digitalWrite and turn off the analog output like we do when you write 255 on a PWM channel.

SpenceKonde, 2021-03-15:
Per my previous comment, I consider the behavior you observed to be a bug; analogWrite(pin,255) ought to actually set the DAC value as you expected rather than call digitalWrite() to turn off the DAC output. analogWrite(pin,255) calling digitalWrite(pin,HIGH) should only be done if that would be  equivalent to the result of linear extrapolation from analogWrite(pin,n) as n approaches 255 (same on the low end - Actually, PWM under some circumstances requires replacing modification of compare value registers with a digitalWrite() for IIRC the low end; CMPx = TOP and CMPx = BOTTOM sometimes are not consistent with what you might expect or hope for - IIRC I ran into that with TCD0 - though in that specific case, in current version, due to the NO_GLITCH TCD0 option, analogWrite(pin_on_timer_d,0_or_255) never turns off PWM and always sets the compare value to something that will never be reached, hence will never result in pin being toggled,

technoblogy, 2021-03-15:
Sorry, I'm a bit confused. Are you saying: if analogWrite() is writing to the DAC output, you should just output the pure DAC voltage with no interference from digitalWrite()?

SpenceKonde, 2021-03-16:
I think what *should* happen is that we should write 255 to DAC0.DATAH and leave the DAC enabled, leaving a value of 1020 on the 10-bit DAC's DATA register, and it should output nearly the DAC reference voltage.
Note that this of course brings to mind the matter of DAC0.DATAL - one might argue that maybe it would be better to put 1 in the two LSB, so analogWrite(PIN_PD6,255) would result in 0x3FF being in the DATA register instead of 0x3FC, which would make writing 0 to the pin more useful too. Actually, I think I'll just do that. 

As always, digitalWrite() on the pin should write it to

technoblogy, 2021-03-16:
> analogWrite(PIN_PD6,255) would result in 0x3FF being in the DATA register

I don't like that idea. With a DAC reference of 1V024 the normal step between values is 0.004V, but this would make the step between 254 and 255 jump to 0.007V.

But then I'm not a fan of the Arduino policy of messing with the PWM values in the first place. With PWM we have a choice between:

a) 256 equal steps of Vcc/256, so the range is 0 to 255*(Vcc/256).

or:

b) 256 equal steps of Vcc/256, but we're going to fudge the last value to make it Vcc so there's a double-sized jump at the end of the range.

Arduino have chosen (b) and I don't understand the logic.

SpenceKonde, 2021-03-16:
lol, I'm not that much of a lunatic - I meant setting the low bits ALL the time, so stepsize would be the same and you'd expect ~3 mv with 1.024 reference from analogWrite(PIN_PD6,0); 

There's another option with the pwm - which is what megaTinyCore and DxCore do:
255 equal steps of Vcc/255, and lower TOP by 1. It also ends up, apparently by coincidence(?) making certain math with timekeeping work out evenly when it otherwise wouldn't come out exact at 16 MHz or 20 MHz.

technoblogy, 2021-03-16:
> lol, I'm not that much of a lunatic - I meant setting the low bits ALL the time, so stepsize would be the same and you'd expect ~3 mv with 1.024 reference from analogWrite(PIN_PD6,0);

But then zero would be 0.003V. One way or the other there's going to be a glitch.

technoblogy, 2021-03-16:
> 255 equal steps of Vcc/255, and lower TOP by 1.

Hmm. I'll have to think about that...

technoblogy, 2021-03-16:
If you really want to mess around with the value the best thing would be for analogWrite(PIN_PD6, n) to set the 10-bit value to:

(n * 1023) / 255

which would be relatively glitch-free, and would give 0V and DACREF at the ends of the scale.

technoblogy, 2021-03-18:
A simpler solution, which is probably equivalent, would be to set the bottom 2 bits of the 10-bit DAC value to the top 2 bits of the 8-bit analogWrite value. This would make analogWrite(PIN_PD6,0) -> 0, and analogWrite(PIN_PD6,255) -> 1023, with everything continuous in between.

SpenceKonde, 2021-03-20:
First off, it's not a matter of "should we mess with those bits". The question is, what should those bits be set to? As is currently implemented, I "decided" to set them to 0, without ever really thinking about it.. Was that the right decision? I still doin't really know.  But now I know how I would mess with it if I wanted to :-P

I was playing with this - at DAC0.DATA=0 you still get a couple of mv out - the analog circuitry doesn't pull quiiiite to ground. Not surprising (it is insufficiently advanced technology in that it is readily distinguishable from magic). What was surprising to me was the distinct pattern in the error. After measuring the voltage at DAC=1023 (4.130 and 1.021 for the 4.096 and 1.024v references  (within 1% on both - dead the fuck on), in my calculations I took that as the maximum, then calculated the expected voltages (intent being to take the accuracy of the reference voltage out of the equation). So DAC = 0 was always going to be high. Increasing the DAC setting to 1 does not change the output voltage, and even bumping it to 2 barely changes the measured voltage (and do note that if I digitalWrite(PIN_PD6,LOW) it would drive the pin all the way down to 0.3mV) so except at DAC=0. it is reading low on the lower end of the scale. Between DAC=1 and DAC=512~640ish, it was consistently low - by 1-3 LSB, with error peaking at around DAC=128.  On the 1.024v, when DAC > 600 or so, the measured voltage was within the readability limit of my DVM (above 600mv, I lose the 0.1mV digit - all measurements were within 1mV of the expected value there). With the 4.096 V reference, the error in the positive direction peaked with DAC = 830ish and the output voltage about 1 LSB high, and peaked in the negative direction at around 1.5-1.8 lsb low, and the same DAC=128..

Anyway - If one was to "mess with" the LSBs of the DAC value.... when VREF is 4.096, set DATAL to 01 when DATAH  < 128-ish. But when VREF =1.024? while DATAH is below 112 or so, set it to 3 unless DATAH is 1 or 2 (then set it to that), above DATAH = 112, back off DATAL by like... 1 per 16 points on DATAH until 0.  And do all your math pretending you didn't do that, and the results will come out closer to your expectations :-P 
Would be interesting to know how much this behavior varied between individual chips.... 

I imagine that 2.048 and 2.50 are somewhere between 1.024 and 4.096 in terms of DAC behavior. And of course, this was under no-load conditions.(but that's all you should be using a DAC for anyway :-P )

But yeah, I think I will leave it at no DATAL = 0... not sure I feel like crawling deeper into this rabbithole (and I do think it's more of a rabbithole than a rathole, I'm pretty confident that one could get slightly better accuracy out of the DAC doing what I described). I want to, but I think I want to get other shit done more. And need to get other shit done!

closed 2021-03-20 by SpenceKonde

technoblogy, 2021-03-20:
First of all, sorry about using the phrase "mess with" - I didn't mean it in a derogatory way.

It's good that you tested it in practice, but there may be differences between samples of one chip, and differences between different parts in the families - as you say, a rabbit hole. In the absence of an obvious better solution the best thing is to keep it simple, as you've decided.

SpenceKonde, 2021-03-20:
Huh? I didn't take any offence to the term - I thought it was an appropriate description of the territory I would be wading into. It's the sort of thing where I hope someone goes and investigates it on a few dozen chips, makes spreadsheets and makes a lovely post about it on the forums that I can read, or a library I can use that improves correctss the values passed to a DAC used in 8-bit mode... Though I'd honestly prefer the article to the practical code, since I could count on one hand the number of times I've needed to programmatically generate analog voltage.... I think there is only one project where I've actually needed or wanted a DAC that wasn't just a "lookit what is possible, isn't it cool" kind of thing. And I haven't actually gotten to where I generate the analog voltages there. 

Anyway, you've got to be a lot ruder than that to offend me! Accusing me of "messing with" a timer, when that is exactly what I am discussing, doesn't cut it.