SpenceKonde/DxCore #85: Create Azduino4 avr-gcc package, add to DxCore, megaTinyCore

SpenceKonde, 2021-03-12:
New ATPACKs are out: AVR64DD and tinyAVR 322x parts  now have headers!

These need to be brought into the cores, and the necessary adjustments made (DD-series should go into DxCore real easy.

The tinyAVR 2-series of course is all working with megaTinyCore except for the new ADC, which is like 95% of it....

SpenceKonde, 2021-04-06:
This is done and ready.

closed 2021-04-06 by SpenceKonde

killerkalamari, 2021-09-14:
@SpenceKonde How can I obtain the source code for the DxCore patched Azduino4 avr-gcc?

SpenceKonde, 2021-09-14:
There is no source code. The binaries are from the Arduino7 official version, I don't know how to compile it, let alone crosscompile it for 6 archtectures. The Azduino toolchain are made just by copying files from the latest ATpacks over the existing Arduino7 toolchain and recompressing it.

killerkalamari, 2021-09-14:
Thank you, that makes sense. By any chance is there a script or instructions on how I could do that myself?

SpenceKonde, 2021-09-14:
Per a .md file.....
## Deploy AWS EC2 instance

Associate IP with it. Use C5-xlarge toe expedite build.

Install needed programs:

sudo apt-get install git build-essential gperf bison subversion texinfo zip automake flex libtinfo-dev pkg-config python2.7-dev libexpat1-dev guile-2.0-dev

## Build our special toolchain

mkdir Arduino

cd Arduino

(git clone official arduino toolchain repo, but replace build.conf with corrected one. 

cd toolchain-avr

./package-avr-gcc.bash

Bring home the .tar.bz and **shut down instance**

## Process archive
Download and extract

Within extracted files, we want to retain only the following files, delete everything else.
* /avr/avr/include/avr/io*.h (only need it for parts that were updated)
* /avr/avr/lib/* and subfolders except ld-scripts
* /avr/avr/lib/ld-scripts needs to be deleted. 
* /avr/lib/gcc/avr/7.3.0/ and subfolders

Test by copying over toolchain on normal use Arduino instance.

Extract latest Arduino official toolchain for windows (arduino7). 

Copy what's left of the newly built toolchain (900ish.a, .o, and .h files) tree over it, replacing files. Recompress and rename.  Windows toolchain is set.

That is windows. 

## For linux:
```
cd toolchain
mkdir versionnum
cd versionnum

```
copy the unmodified, as buuilt archive to this location from windows. 
```
mkdir azduinon
cd azduinon
mv ../*azduinon*.bz2 .
tar -xf *.tar
cd ..
```
so you are back in versionnum, with a azduinon directory inside it.
Reach in from windows and delete everything in that extracted except the files we want per above.

### Enhance the Arduino7 toolchain packages with new files

```
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-arm-linux-gnueabihf.tar.bz2
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-aarch64-pc-linux-gnu.tar.bz2
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-apple-darwin14.tar.bz2
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-pc-linux-gnu.tar.bz2
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-pc-linux-gnu.tar.bz2

tar -xf avr-gcc-7.3.0-atmel3.6.1-arduino7-aarch64-pc-linux-gnu.tar.bz2
cp -r azduino4/avr/* ./avr
tar -cvf avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar avr
bzip2 -z avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar
rm -rf avr

tar -xf avr-gcc-7.3.0-atmel3.6.1-arduino7-arm-linux-gnueabihf.tar.bz2
cp -r azduino4/avr/* ./avr
tar -cvf avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar avr
bzip2 -z avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar
rm -rf avr

tar -xf avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-apple-darwin14.tar.bz2
cp -r azduino4/avr/* ./avr
tar -cvf avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar avr
bzip2 -z avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar
rm -rf avr

tar -xf avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-pc-linux-gnu.tar.bz2
cp -r azduino4/avr/* ./avr
tar -cvf avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar avr
bzip2 -z avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar
rm -rf avr

tar -xf avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-pc-linux-gnu.tar.bz2
cp -r azduino4/avr/* ./avr
tar -cvf avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar avr
bzip2 -z avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar
rm -rf avr

```

## Creating info for board manager

```
ls -l *azdu*aarch64*.bz2
sha256sum *azdu*aarch64*.bz2
ls -l *azdu*arm*.bz2
sha256sum *azdu*arm*.bz2
ls -l *azdu*i686*.bz2
sha256sum *azdu*i686*.bz2
ls -l *azdu*x86_64-a*.bz2
sha256sum *azdu*x86_64-a*.bz2
ls -l *azdu*x86_64-p*.bz2
sha256sum *azdu*x86_64-p*.bz2
```

find:
```
^drazzy.*\n.*drazzy (\d{8}).*(avr.*)\n.*\n([0-9a-f]+).*ino\d+-([^\.]*).*
```
replace:
```
{
  "size": "\1",
  "checksum": "SHA-256:\3",
  "host": "\4",
  "archiveFileName": "\2",
  "url":"https://spencekondetoolchains.s3.amazonaws.com\2"
},
```

BAM! Was that easy or what?

Check a linux archive, extract to tar in windows and compare sizes of directoeries in the tar with 7zip explorer. 

Upload the 6 new toolchain archives to AWS S3 bucket. 
Upload the x64 linux one to github too, as a release asset. Change the URL in json appropriately. This is to keep the bandwidth charges for the CI runs from eating me alive.

Include new version in json and referemce new toolchain from core.

## DONE

```

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-aarch64-pc-linux-gnu.tar.bz2

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-arm-linux-gnueabihf.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-arm-linux-gnueabihf.tar.bz2

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-pc-linux-gnu.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-i686-pc-linux-gnu.tar.bz2

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-apple-darwin14.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-apple-darwin14.tar.bz2

ls -l avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-pc-linux-gnu.tar.bz2
sha256sum avr-gcc-7.3.0-atmel3.6.1-azduino3-x86_64-pc-linux-gnu.tar.bz2

```

```
ls -l (.*)\n.*user (\d+).*\n.*o3-(.*).tar.bz2\n([a-f0-9]+)
```

```
            {
              "size": "\2",
              "checksum": "SHA-256:\3",
              "host": "\4",
              "archiveFileName": "\1",
              "url":"https://spencekondetoolchains.s3.amazonaws.com/\1"
            },
```

SpenceKonde, 2021-09-14:
build.conf
AVR_VERSION=3.6.1
BUILD_NUMBER=azzy9

AVR_SOURCES="http://downloads.arduino.cc/tools/opensource/Atmel-AVR-GNU-Toolchain/${AVR_VERSION}"
ATMEL_PACKS_SOURCES="http://packs.download.atmel.com/"
GNU_SOURCES="https://ftp.gnu.org/gnu/"
MPC_SOURCES="http://repository.timesys.com/buildsources/m/mpc/"

# The following version numbers are by default parsed out of the SOURCES.README
# in the Atmel distribution, you can override here if you want (or if it breaks)

# GCC_VERSION=4.9.2
# AUTOCONF_VERSION=2.64
# AUTOMAKE_VERSION=1.11.1
# GMP_VERSION=5.0.2
# MPFR_VERSION=3.0.0
# MPC_VERSION=0.9

GCC_VERSION=7.3.0
MPFR_VERSION=3.1.0
MAKE_JOBS=8

ATMEL_ATMEGA_PACK_VERSION=1.6.364
ATMEL_ATMEGA_PACK_FILENAME=Atmel.ATmega_DFP.${ATMEL_ATMEGA_PACK_VERSION}
ATMEL_ATMEGA_PACK_URL=${ATMEL_PACKS_SOURCES}${ATMEL_ATMEGA_PACK_FILENAME}.atpack

ATMEL_ATTINY_PACK_VERSION=1.8.332
ATMEL_ATTINY_PACK_FILENAME=Atmel.ATtiny_DFP.${ATMEL_ATTINY_PACK_VERSION}
ATMEL_ATTINY_PACK_URL=${ATMEL_PACKS_SOURCES}${ATMEL_ATTINY_PACK_FILENAME}.atpack

ATMEL_DX_PACK_VERSION=1.6.76
ATMEL_DX_PACK_FILENAME=Atmel.AVR-Dx_DFP.${ATMEL_DX_PACK_VERSION}
ATMEL_DX_PACK_URL=${ATMEL_PACKS_SOURCES}${ATMEL_DX_PACK_FILENAME}.atpack

# With any luck, you don't need to edit the below
################################################################################

if [[ "$OSTYPE" == "darwin"* ]]; then
SED=gsed
else
SED=sed
fi

GREP=grep

#fallback for grep without pearl regexp
SOURCES_README="$(wget -O - ${AVR_SOURCES}/SOURCES.README 2>/dev/null)"
[ -z "$GCC_VERSION"       ] && GCC_VERSION="$(echo "$SOURCES_README" | $GREP "GCC"  | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"
[ -z "$AUTOMAKE_VERSION"  ] && AUTOMAKE_VERSION="$(echo "$SOURCES_README" | $GREP "automake"  | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"
[ -z "$AUTOCONF_VERSION"  ] && AUTOCONF_VERSION="$(echo "$SOURCES_README" | $GREP "autoconf"  | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"
[ -z "$GMP_VERSION"  ] && GMP_VERSION="$(echo "$SOURCES_README" | $GREP "gmp"  | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"
[ -z "$MPFR_VERSION" ] && MPFR_VERSION="$(echo "$SOURCES_README" | $GREP "mpfr" | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"
[ -z "$MPC_VERSION"  ] && MPC_VERSION="$(echo "$SOURCES_README" | $GREP "mpc" | egrep -o "([0-9]{1,}\.)+[0-9]{1,}")"

# Build the URLs from which to grab the archives for those components
AUTOCONF_SOURCE="${GNU_SOURCES}/autoconf/autoconf-${AUTOCONF_VERSION}.tar.bz2"
AUTOMAKE_SOURCE="${GNU_SOURCES}/automake/automake-${AUTOMAKE_VERSION}.tar.bz2"
GMP_SOURCE="${GNU_SOURCES}/gmp/gmp-${GMP_VERSION}.tar.bz2"
MPFR_SOURCE="${GNU_SOURCES}/mpfr/mpfr-${MPFR_VERSION}.tar.bz2"
MPC_SOURCE="${MPC_SOURCES}/mpc-${MPC_VERSION}/mpc-${MPC_VERSION}.tar.gz"

# For debugging, spit it all out
cat <<EOF
VERSIONS
-------------------------------------------------------------------------------
GCC:         ${GCC_VERSION}
AVR:         ${AVR_VERSION}
BUILD:       ${BUILD_NUMBER}
AUTOCONF:    ${AUTOCONF_VERSION}
AUTOMAKE:    ${AUTOMAKE_VERSION}
GMP:         ${GMP_VERSION}
MPFR:        ${MPFR_VERSION}
MPC:         ${MPC_VERSION}

SOURCES
-------------------------------------------------------------------------------
AUTOCONF:    ${AUTOCONF_SOURCE}
AUTOMAKE:    ${AUTOMAKE_SOURCE}
GMP:         ${GMP_SOURCE}
MPFR:        ${MPFR_SOURCE}
MPC:         ${MPC_SOURCE}
AVR:         ${AVR_SOURCES}/*

killerkalamari, 2021-09-15:
Wow, that was extremely helpful, thanks a lot! Using your instructions I created a Bourne shell script to create the toolchain on/for Debian-compatible Linux systems: [build-toolchain.sh.gz](https://github.com/SpenceKonde/DxCore/files/7165762/build-toolchain.sh.gz)

My goal was to be able to set up a Makefile build for my project, so that someone else could compile my project without needing to install the Arduino IDE or relying on internal Azduino URLs (that saves you a little bandwidth, too!). I didn't attempt to create something Arduino IDE-compatible. As a result, some things are moved around or possibly missing in the script-created toolchain-avr-dx directory compared to what you get when you install DxCore in the Arduino IDE. For example, on my system:

- .arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin -> toolchain-avr-dx/avr/bin
- .arduino15/packages/DxCore/hardware/megaavr/1.3.6 -> toolchain-avr-dx/megaavr

The toolchain created by this script works to compile and link programs from the command line, but since it's not usable from the Arduino IDE, you need to already to know which commands to run. An easy way to determine that is to perform a preliminary dummy build using the Arduino IDE. Install DxCore in the Arduino IDE and then code up a quick blink sketch or such. Compile that in the Arduino IDE, then save the generated command output. That output can be used to determine the generated avr-gcc/avr-g++/avr-objcopy commands and options. The sources for 3rd party libraries can be downloaded and built into the project (I've done that anyway as I almost always want to tweak the interface and remove functionality I don't need).

p.s. It took just under 8 minutes to clone and compile toolchain-avr on my discontinued [i7-8700K](https://ark.intel.com/content/www/us/en/ark/products/126684/intel-core-i7-8700k-processor-12m-cache-up-to-4-70-ghz.html) (used 12 "threads"). Not bad!