SpenceKonde/DxCore #86: Flash Writer minor comments

technoblogy, 2021-03-19:
A few minor comments after testing out the flash:

**writeWord** and **writeByte** get syntax highlighted, but **readWord** and **readByte** don't.

In the documentation at:

https://github.com/SpenceKonde/DxCore/blob/master/megaavr/libraries/Flash/README.md#api

**checkFlashWrite()** should be **Flash.checkWritable()**.

**flashErasePage** should be **Flash.erasePage** (two occurrences).

SpenceKonde, 2021-03-20:
Thanks, fixed. How generally usable is it? Is

technoblogy, 2021-03-20:
I'm not doing anything very fancy - just copying a block of data from RAM to flash and back again, but it seems to work well. Still exploring it - I'll let you know if anything crops up.

SpenceKonde, 2021-03-21:
I think the big question in my mind is whether it needs a more formal method to "update" a page, ie, something where you tell it "read this page into a 512b buffer in ram (what would this be best implemented as? I feel like the natural form that the data would want to be stored in ia union of a 512-member uint8_t array and a 256-member uint16_t array... probably a member variable of an instance of a FlashPage class? Flash could have a getPage(page) method, FlashPage would have a public data propertythat's that array I described, and a read-only uint8_t storing the page number... so then you'd do whatever you wanted to the data in the FlashPage, and FlashPage.Write() would just call Flash.erasePage( (uint32_t) page_num) <<  9, 1); Flash.writeWords( (uint32_t) page_num) <<  9, data.words,256);  

I don't know C++ well enough to implement it without more C++ learning even though I'm pretty sure the stuff I don't know isn't hard. Have many much more pressing priorities atm.....

technoblogy, 2021-03-21:
My personal view is that something like that isn't needed; I for one wouldn't use it. What you've got at the moment is brilliant, and more than I need...

Also, I would vote to keep the Arduino cores as free as possible of C++ stuff. It's difficult to understand, and (I'm told by an expert) really slow to compile.

technoblogy, 2021-03-21:
Actually I had one query:

* Is there any advantage in me only erasing the number of pages I actually need to save my data? At present I’m erasing the full 16K I’ve reserved, as it seemed a bit of work to make sure the size is a power of 2.

technoblogy, 2021-03-21:
Another query:

* Isn't it invalid to specify addresses less than 0x10000? So shouldn't:

      Flash.erasePage(0xFE00, 1);

  return an error?

SpenceKonde, 2021-03-21:
No, why would it be?
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

technoblogy, 2021-03-21:
Looking at the AVR128DA48 memory map (Section 7.2 in the datasheet) the In-System Reprogrammable Flash goes from 0x18000 to 0x1FFFF, so I assumed this is all you could access with Flash.erasePage().

SpenceKonde, 2021-03-21:
Aah - that is trying to explain how there are two address spaces, code space (flash) and data space (ram, registers, and the memory mapped flash. The code space (Flash) has addresses from 0x00000 to 0x1FFFF - that's your 128k of in system reprogrammable flash [17 bit address space spm and elpm use RAMPZ as the high bit and the Z pointer register as the lower 16 for byte addressed access to the full program memory; the program counter and things that change it (like jmp, call, and so on) although you pass them numbers of bytes in assembly, the machine code that comes out of them has rightshifted all the numbers 1 bit - they are all word addressed, so they don't need to worry about the extra bit)) - the left half of the diagram. The right half of the diagram depicts the data space (what is accessed with load/store instructions.
The part that you describe is trying (apparently not very successfully in this case) to show how the area of the code space from 0x18000 to 0x1FFFF is mapped to the data space from 0x8000 to 0xFFFF. 

That's where variables you declare with MAPPED_PROGMEM get stored. I think I have some semblance of an example in the demo where I write something to the mapped flash, then cast the address to a pointer, and demonstrate that I can read back what I just wrote there. And I think Flash contains a function to convert between memory-mapped addresses and the equivalent address within the code space. Note that you can change which section of flash is mapped (the last one is just the default, as it should be) with the FLMAP bits in NVMCTRL.CTRLB. If you do that, and you declared things MAPPED_PROGMEM or otherwise make use of the memory mapped flash, if you change the FLMAP bits, those pointers will now access the wrong section of memory. I don't think most people have much reason to change the FLMAP bits, though.

technoblogy, 2021-03-21:
So does Flash.erasePage() automatically assume you're talking about the data space if you give it an address between 0x8000 and 0xFFFF, and the code space if you give it an address between 0x18000 to 0x1FFFF?

SpenceKonde, 2021-03-21:
All addresses passed to all functions of flash are addresses in the code space. The only way that anything in the data space is involved s the Flash.flashAdderess and Flash.mappedPointer, - the former takes a pointer, and if it points to the portion of the code space contaning memoy mapped flash, it returns the adderess of what t s pontng at n the code space. considering the current FLMAP setting. The othe does the reverse. 

(any address between 0x00000 and 0x1FFFF is valid address within the code space. Assuming you're using a mode that allows writing anywhere in the flash (either optboot board def, or Flash writing: anywhere), any address between 0x00200 and 0x1FFFF is a valid address to erase/write (You can'd blow away the first page - that's in the bootloader section (if a bootloader section is not defined, all self programming is disabled on these parts; so even if you're not using a bootloader, fi flash writing is enabled, the first page is set to be bootloader and  i ensure that there is an SPM entry point there, but set t to se interrupt vectors at the start of the flash and no bootloader s actally involved anywhere.

technoblogy, 2021-03-22:
OK, got it now thanks!

technoblogy, 2021-03-23:
On an AVR128DA48 with 26MHz clock:

I've been doing some timings, and it takes 1019ms to write 0x4000 bytes to flash (excluding the time to erase it), and 11ms to read them back. That's an average writing rate of about 16kHz. Does that sound right?

Would you expect that to be fairly uniform, or would it be necessary to put data in a circular buffer first to ensure the data rate is consistent?

SpenceKonde, 2021-03-23:
As I understand it there is a fixed time for each word written and each
page erase. I think it is independent of clock speed (probably times with
internal 32khz osc). Uniform over what? And why would you need a consistent
data rate?

Cpu is halted during the writes, btw. There is no read while write on these
parts, which I am content with, dealing with rww was just plain annoying
iirc - what are you going to do during a flash write, other than busywait
for it to finish and dont forget to enable read on the rww section (you
could only read the nrww section, which included bootloader section, while
writing to the rww section. Nrww section halted CPU during writes)

Is this related to that issue with serial overflow? You making a custom
bootloader type thing? I think the usual solution is for the upload tool to
know when the target is going to do a batch of writes and stop sending more
data to it until it sends something to confirm that it has written it, and
go ahead and send more. That is effectively how stk500 works (it stk500 -
and maybe any viable programming protocol - is also, effectively, half
duplex - at every point in time, there is only one device whose turn it is
to send the next byte, and If the other one does, they will get out of
sync, it will be detected, and connection will be terminated

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

technoblogy, 2021-03-23:
No, not to do with serial overflow. I was thinking whether it would be possible to record audio straight to flash under interrupt.

technoblogy, 2021-03-24:
By the way, I'm using Flash Writer to implement saving programs on my Lisp interpreter:

[uLisp news!](http://www.ulisp.com/show?1DQ5)

SpenceKonde, 2021-03-28:
> No, not to do with serial overflow. I was thinking whether it would be possible to record audio straight to flash under interrupt.

Aaah - I am not very optimistic what with the whole CPU halted white the flash is being written or erased (though they are bringing RWW back it looks like) thing....

technoblogy, 2021-03-28:
Thanks! Perhaps it would be possible a page at a time with a one page RAM buffer.

technoblogy, 2021-03-31:
> I am not very optimistic what with the whole CPU halted while the flash is being written or erased

You're right - this is not going to work :(

SpenceKonde, 2021-03-31:
IMHO for working ith audio, in any way shape or fom, you are askng a hell of a lot fom an 8-bt medoce spec mcocontolle.

SpenceKonde, 2021-04-01:
Closing this - I don't see any indication of suggestions or issues outstanding. Please reopen if I am mistaken here.

closed 2021-04-01 by SpenceKonde

SpenceKonde, 2022-02-04:
Randomly wound up here - one thing I think maybe I wasn't fully cognizant of at the time - Writing flash is actually like, really fucking fast, and is byte/word oriented.  Assuming you've got a block of pre-erased pages you're filling, the cpu gets halted for 70us after each word is written - peripherals still operate! So you should be able to say, have the ADC in free running mode performing conversions during that time. If you were to configure the ADC to free run with a period just longer than a write cycle, you could write a word, then poll the RESRDY until you had new data, immediately write it to flash, repeat, it might be less terrible that it seemed at first glance. max theoretical sampling rate.... It would fill flash goddamned fast though...

technoblogy, 2022-02-04:
Thanks for revisiting this! You've inspired me to revive my idea of designing an audio recorder that records straight to flash. I'll let you know if I get something working.