SpenceKonde/DxCore #87: Serial drops characters unless you increase the buffer size

technoblogy, 2021-03-20:
I've done tests on an AVR128DA48-based Curiosity Nano, pasting characters into the Serial Monitor input field and displaying them in the window, and you get dropped characters unless you increase the buffer sizes to 256 in UART.h with:

    #define SERIAL_TX_BUFFER_SIZE 256
    #define SERIAL_RX_BUFFER_SIZE 256

I can post the evidence if it's useful. I propose these are made the defaults, at least on parts with 16K RAM.

technoblogy, 2021-03-20:
I've closed this issue while I investigate a bit more. The change makes a much bigger effect on memory than I expected.

technoblogy, 2021-03-20:
OK, I was worried that my suggested change would allocate an additional 6 * 192 * 2 bytes because there are 6 serial ports, but it only allocates for what your program uses so that's OK.

SpenceKonde, 2021-03-20:
I am not convinced this isn't an inefficient application - possibly made worse by some quirk of the curiosity nano's serial adapter (microcontrollers used as serial adapters always have weird quirks). Does it reproduce with other serial adapters and/or on other USARTs? Does 256 prevent the issue, or just let you do larger blocks of text before stuff is lost. I have a curiosity nano, so if you post code and instuctions to reproduce, I can try to do so and compare to non-curiosity-nano.

technoblogy, 2021-03-20:
No, many Arduino platforms have the same problem, but some are worse than others. Some actually hang up if you send more than a certain number of characters! Some time when I've got time I want to test all the boards I've got and see if I can see a pattern.

> Does 256 prevent the issue, or just let you do larger blocks of text before stuff is lost.

I think the latter actually. Here's my test for Curiosity Nano AVR128DA48:
````text
#define Serial Serial1

void setup() {
  Serial.begin(9600);
  while (!Serial);
  delay(5000);
  Serial.println(SERIAL_TX_BUFFER_SIZE);
  Serial.println(SERIAL_RX_BUFFER_SIZE);
  
}

int count = 0;

void loop() {
  if (count % 64 == 0) {
    Serial.println(); Serial.print(count); Serial.write(' ');
  }
  while (!Serial.available());
  Serial.write((int)Serial.read());
  count++;
}
````
Here's my test text to cut and paste into the Serial Monitor input box:
````text
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
````
and this is what I get. First dropped characters at about byte 1470:

[image]

SpenceKonde, 2021-03-20:
So you send a ginormous block of text. It must continue to ingest this text at the speed it comes down the wire, while simultaneously echoing back the text PLUS the ongoing character count? While the computer does not have to send any extra characters? If I have to send 70 characters for every 64 that you send, and you're sending without pause at the speed of the line, the serial is the same speed in both directions, it's inevitable that I won't be able to keep up. I think that test may directly measure the combined size of the two serialbuffers....

technoblogy, 2021-03-20:
OK fair enough, I suppose Serial is a flawed protocol without handshaking. Thanks for considering it.

closed 2021-03-20 by technoblogy

technoblogy, 2021-03-24:
I've found an alternative solution to this, which is to provide the option of turning off the echoing of characters when pasting a large block of text into the Serial Monitor. It then works without dropped characters, even on the original buffer size of 64 bytes.

I don't know why I didn't think of this before...

SpenceKonde, 2021-03-24:
Oh - yeah, now that I see what you're doing with it - yeah, you 100% have to turn off echo when sending a big block of text through a serial port to something. Espruino (which is conceptually similar, but runs JS on STM32, Espressif and those Nordic parts with the integratyed bluetooth) has a command that turns off echo which gets sent at the start of any upload of a block of js code. that is definitely the correct approach. 

IMO saying serial is flawed is  little harsh when you're the one asking for it to do something that the laws of mathematics say simply cannot be done.  ;-)

technoblogy, 2021-03-24:
> IMO saying serial is flawed is little harsh

OK, fair enough! Is suppose what bugs me a bit is that there's an established solution, handshaking, which could have been implemented.

SpenceKonde, 2021-03-24:
Well, there are a great many ways to do handshaking. Do you want the people who wrote the Arduino IDE to be the ones choosing which one to use?! I have full confidence in their ability to make choices here that made everything worse :-P. The Arduino IDE's serial monitor isn't really fit for any purpose more advanced than basic Serial.print() debug anyway...

technoblogy, 2021-03-24:
No, please let's leave it the way it is and work around it!