SpenceKonde/DxCore #90: Compile error using Wire1 on AVR128DA48

technoblogy, 2021-03-26:
I'm trying to use the AVR128DA48 as an I2C master using pins SDA=PF2, SCL=PF3, so I've done:

    Wire1.swap(1);
    Wire1.begin();

However I'm getting this compile error:

> In file included from /Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/libraries/Wire1/src/Wire1.cpp:27:0:
/Users/david/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.2/libraries/Wire1/src/utility/twi1.h:88:1: error: 'TWI1_BUSSTATE_t' does not name a type; did you mean 'TWI_BUSSTATE_t'?
 TWI1_BUSSTATE_t TWI1_MasterState(void);
 ^~~~~~~~~~~~~~~
 TWI_BUSSTATE_t

Any suggestions?

Also, a documentation comment: you say:

> Wire.swap(1) or Wire.swap(0) will set the the mapping to the alternate (1) or default (0) pins.

I assume this should include **Wire.swap(2)**, since in PORTMUX there is DEFAULT, ALT1, and ALT2?

technoblogy, 2021-03-26:
Doing the change the compiler suggested then produces a whole lot of other errors, so it's not that simple...

SpenceKonde, 2021-03-26:
Don't use Wire1 - it's still in the core?! I thought I pulled that from 1.2.0 after it became  clear that I had botched the implementation so severely that it could never be used for any practical application.

I will ensure that the non-functional Wire1 library is fully removed. 

I do not have the C/C++ skill to create a working Wire1 library.

closed 2021-03-26 by SpenceKonde

technoblogy, 2021-03-26:
OK, fair enough! Thanks, David

technoblogy, 2021-03-26:
This comment is still valid for Wire though. You say:

> Wire.swap(1) or Wire.swap(0) will set the the mapping to the alternate (1) or default (0) pins.

I assume this should include **Wire.swap(2)**, since in PORTMUX for TWI0 there is DEFAULT, ALT1, and ALT2.

SpenceKonde, 2021-03-26:
The way this works on Wire, since dual mode is not supported (and without dual mode default and alt1 are identical)  is that swap(1) gets you alt2, swap(0) gets you default, and alt1 is not used because it is redundant. Obviously if someone got dual mode working , this decision would have to be revisited (it predates my pulling in the swap (i suspect you've read my code enough that, if you read the source, it would be readily apparent that it was non-spence-origin code - I tend to think that, for better or for worse, my coding style is at least somewhat distinctive.