SpenceKonde/DxCore #95: Curiosity Nano DB - programmer failure on Linux

anarchodin, 2021-03-30:
I just got an AVR128DB48 Curiosity Nano board, installed DxCore ... and promptly errored out.

I _can_ upload to the board, by using the mass storage device and asking the Arduino IDE to export binaries. But trying to use the Arduino IDE to do the upload consistently gives this warning:

    avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
    avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
    avrdude: jtag3_edbg_prepare(): failed to send command to serial port

This appears to be blocking me from setting any of the fuses, which is a little annoying. (Using the USB drive to flash the device is honestly _better_ for my workflow.) Any idea what might be going on?

SpenceKonde, 2021-03-30:
I have no experience using these parts on linux. I have not a clue. Could there be some "modem manager" that opens any serial port connected to the computer? do you have appropriate permissions to use serial ports?  does the mass storage need to be unmounted to work? I'm just grasping at straws here, I've barely used the curiosity board, and don't have physical access to any linux systems to test on.

SpenceKonde, 2021-03-30:
This is not really specific to this core, since that error is coming from avrdude, and would happen independently of my core... Moving to discussions section in hopes of someone else seeing it over there, and because it does not represent a defect in DxCore.

closed 2021-03-30 by SpenceKonde