SpenceKonde/DxCore #97: AVR128DB48 Curiosity Nano serial

dpetican, 2021-04-04:
I've had good success so far using the core with an AVR128DB28 chip.  Not so much using the nano.  I can upload a program using the built-in programmer, but can't get any output over the virtual com port using Serial(0) or any other Serial(x).
I tried using Serial.swap(1) and attaching a serial adapter to PA4, PA5.  However, the serial output while present is garbled.  What am I doing wrong?  It feels like a clock/baud issue because if I lower the baud rate from 115200 to 9600 I get a different type of garble.

SpenceKonde, 2021-04-05:
Well you haven't told me what you are doing, and I have not used my curiosity nano yet, so I couldn't say for sure....

Curiosity Nano AVR128DB48 uses USART3 for communication with the CDC adapter per schematic, so you'd use Serial3 to talk to it.

What clock source did you have selected? I would expect all of the internal options to work, along with external 16 MHz crystal.

dpetican, 2021-04-05:
Yes Serial3 works, but...  If I upload the sketch then open the virtual com port from within Arduino it takes 3-5sec for the output to appear.  If the output window is already open and I upload the sketch it takes 12-16sec for the output to appear.  Also, the output keeps repeating itself which since the the println is in setup must mean the nano is continuously rebooting.  Same behavior using another serial monitor program.

`void setup() {
  Serial3.begin(115200);
  Serial3.println("Hello, World!");
  Serial3.println("Success!");
}
`
If someone could try to replicate this it would be appreciated.  Maybe I have bad board, the embedded debugger is doing something strange, the fuse values are wonky, or... ?  Haven't had any problems of this sort with the DB28 chip on a breadboard or my custom PCB.  I designed the PCB just by breadboarding and did not use the curiosity board at all.  I used the MCP23017 port expanded in dip form this allowed me to turn around the first rev of the PCB very quickly.  Next rev will be all surface mount and use the DB48 so I need the nano to verify my pin selections.

SpenceKonde, 2021-04-06:
Huh. I cant get the IDE to upload to them at all! Not sure what this error mean, but all three of the curiosity nano boards I got are doing it:

avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: bad response to AVR sign-on command: 0xa0
avrdude: retrying with external reset applied
avrdude: JTAGEN fuse disabled?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

dpetican, 2021-04-06:
Just in case here is the correct programmer selection:

[image]

Not sure but I think the drivers were installed as part of Atmel Studio.  They just showed up when I plugged in the Curiosity board.  I use the Atmel-ICE(UPDI) for the bare chip.  It seems this curiosity board is very curious indeed ;)

SpenceKonde, 2021-04-06:
aaaah I think I know what the issue I was having is, avrdude never got updated because I don't use board manager on my main development instance.

SpenceKonde, 2021-05-17:
sorry, I had forgotten about this ... did you get it sorted? Test sketches like that work for me. It sounds like either driver problem, bad firmware on the device, or defective unit.

SpenceKonde, 2021-05-19:
Please reopen with additional information if you are still having issues - though this might be a Microchip suopport tier of issuel. I did an impressively poor job of managing to get the right pins on the right serial port referenced. I'm not much a fan of the way those boards are laid out, they really wantt you to put the pins pointing down, into the solderless breadboard that I wouldn't be caught dead using. Well,

closed 2021-05-19 by SpenceKonde