SpenceKonde/DxCore #100: Unable to build for DB chips?

anarchodin, 2021-04-07:
I have previously used DxCore to build a program that's running on a Curiosity Nano AVR128DB48 board. I upgraded DxCore, and suddenly found myself confronted with this error:

    avr-g++: error: device-specs/specs-avr128db48: No such file or directory

It went away when I downgraded DxCore back to version 1.3.2. I assume there's a mixup with the compiler support file location, as searching through my .arduino15 library revealed one version of the device-specs directory with said file, and one version without. I haven't the first clue how to fix it, though...

SpenceKonde, 2021-04-07:
Oh ffs... will investigate;

SpenceKonde, 2021-04-07:
What OS? 

And are you saying that the azduino4 version is missing that file?

SpenceKonde, 2021-04-07:
Okay, yeah, device specs for like... everything... are missing from all the non-windows packages (I can't test rthem as I don't have a proper system for this and there's no CI set up yet to test new builds) when I am done with work I will rebuild the archives and put up new version.

anarchodin, 2021-04-07:
I did the upgrade again, and now the DB files are missing from both versions - I could have sworn I did that listing before downgrading, though. In any case, that would explain the issue, yes.

SpenceKonde, 2021-04-08:
ugh, finished the process then discovered they were all bad because in one place i'd made a typo in a script command that corrupted the location I copied from.

SpenceKonde, 2021-04-12:
this has been fixed since the release of 1.3.4 at the end of last week.

closed 2021-04-12 by SpenceKonde