SpenceKonde/DxCore #103: 1.3.4 Appears Broken in Arduino IDE 1.8.13

btribley, 2021-04-13:
After upgrading I got this:
```
Detecting libraries used...
"C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4a/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.4\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=4UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.4\\cores\\dxcore/api/deprecated" "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.4\\cores\\dxcore" "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.4\\variants\\28pin-standard" "C:\\Users\\bill\\Documents\\Arduino\\builds\\sketch\\dbcc.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4a/bin/avr-g++": file does not exist
Error compiling for board AVR DA-series (no bootloader).

```
Downgraded to 1.3.3 and restarted the ide:

```
Detecting libraries used...
"C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.3\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=3UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.3\\cores\\dxcore/api/deprecated" "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.3\\cores\\dxcore" "-IC:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.3\\variants\\28pin-standard" "C:\\Users\\bill\\Documents\\Arduino\\builds\\sketch\\dbcc.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist
Error compiling for board AVR DA-series (no bootloader).

```
Downgraded to 1.3.2 fixed it. When attempting to patch Visual Studio Code include paths for the upgrade I noticed the addition of an avr in several tool paths.
For 1.3.4:
"c:\\users\\bill\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4a\\avr\\lib\\gcc\\avr\\7.3.0\\include"
For 1.3.2:
C:\Users\bill\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino3\lib\gcc\avr\7.3.0\include

Sorry to be the bearer of bad news.

SpenceKonde, 2021-04-13:
Yes I am fully aware of this and am in the process of repacking the toolchain. will be fixed in 1.3.5 tonight. I *really* wish someone had reported this in the  6 days that busted versions were available - BEFORE I had released a second core with the broken toolchain.

btribley, 2021-04-13:
I didn't notice the upgrade until tonight. Checked preferences, there is no setting for check-for-updates. It takes a little while for the window to pop, since I am using VS Code as my editor I probably never saw them. Will wait longer from now to see if updates appear. 

On another note, I am subscribed to Notifications on this site. I just checked spam, they are going there. But, I didn't see any notification for 1.3.3 and 1.3.4 release. Is there another setting to make sure I get these? Whitelisting notifications now.

Thanks for your work on this.

SpenceKonde, 2021-04-13:
1.3.5 is now available, it corrects the toolchain packaging problem on windows.

closed 2021-04-13 by SpenceKonde

btribley, 2021-04-14:
@SpenceKonde VS Code Intellisense updated and gave me this. I didn't type any of the 1.3.4 stuff in. Where can I find this in the Arduino IDE so I can check to see if it wasn't updated?

```
                "C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\libraries\\Wire\\src",
                "C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\libraries\\EEPROM\\src",
                "c:\\users\\bill\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\users\\bill\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\users\\bill\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Users\\bill\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\cores\\dxcore\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=24000000L",
                "ARDUINO=10813",
                "ARDUINO_avrda",
                "ARDUINO_ARCH_MEGAAVR",
                "DXCORE=\"1.3.4\"",
                "DXCORE_MAJOR=1UL",
                "DXCORE_MINOR=3UL",
                "DXCORE_PATCH=4UL",

```etc etc.