SpenceKonde/DxCore #104: Setting external crystal on AVR128DB28

classiphil, 2021-05-04:
Hey there,

when I try to set an external crystal I am presented the following error message:
error: 'CLKCTRL_SELHF_CRYSTAL_gc' undeclared (first use in this function); did you mean 'CLKCTRL_SELHF_XTAL_gc'?

The datasheet acually wants the crystal to be set by "CLKCTRL_SELHF_CRYSTAL_gc". By changing it to "XTAL" the code compiles, but the external clock is not used.

SpenceKonde, 2021-05-07:
Sorry, thought I had replied here. The datasheet _doesnt_ tell you that. The datasheet, actually, doesn;t tell you what any of the group code defines are named! 

The patterns in naming suggest it, but they never said that the names in the io headers would always match the datasheet. there are always odd ones like that. The headers almost always use xtal not crystal. In any event, the names that are used in the headers are not my department. If you don't like Microchip's inconsistent naming, complain to them, not me ;-) 

I had no issues making the core work with a crystal,  but why are you configuring that manually? Your initialization code may be getting it's changes undone by the core's initialization. You don't need to put anything in your code to use an external crystal - just select the appropriate option from the Tools -> Clock Speed menu! (note: that menu may get renamed to either Clock or Clock Source in a future release in order to ke[p all my cores consistent. 

In fact...... now that I think about it,  **nothing but woe and misery** awaits those who try to initialize the main clock source themselves instead of using the core functionality provided for that purpose.... For any time-dependant functionality, the core needs to be told what the clock speed; this can only be done by the tools -> Clock Speed submenu because it needs to be known *before* the compiler has gotten to any of the text you typed in the .ino file, in order for things to worrk

If you need a custom F_CPU for your crystal that isn't in the tools -> clock speed menu , you need to modify the boards.txt in order to ensure that the F_CPU define is set by a command line -DF_CPU= parameter passed to the compiler (this is true of other arduino cores for AVR parts too - none of them will completely work  if F_CPU isn't set that way, and the IDE neither provides a way for thecore author to give the user a way to enter a number, as opposed to a "pick one" type deal)

But as I said, whatever frequency you want is probably already supported :-)

Note that, unlike classic AVR devices, these have the fractional baud rate generator so all those wackiy UART crystal frequencies are unneeded.+

SpenceKonde, 2021-05-07:
Personally I've never been able to program the modern AVRs withoiut three windows  - code in one sub;oime text window, io header for the part in the other sublime window, and PDF vierwer for the datasheet.. I almost never needed to have the headers open on classic AVRs.... and yet, I would always picok a modern AVR these days.,

classiphil, 2021-05-09:
Dear SpenceKonde,

thank you!! I will see, where my mistake was then. I am at my 4808 project now, So the AVR128 has to wait a while. I am sure it is a user problem.

I really am grateful. You and the other Core-Developers enable us passionate Hobbyists to use these new chips. It is awesome!

SpenceKonde, 2021-05-13:
I am very curious though, as to why you were implemeneting use of a crystal as clock source yourself, when all that is done for you by DxCore (and whatever problems you had, are likely because of a conflict between the two). In Arduino-land, clock source is supposwed to be handled for the user.. Am happy to suggest an approach on how to address it if I knew more

SpenceKonde, 2021-05-17:
Good for a laugh maybe? 
From recent release notes fir the last ATPACKs:

"Added support for AVR64DD. Added TRUTH bitfields to CCL.TRUTHn registers. Changed nomenclature for serial communication. Renamed external clock source selection in AVR DB from CRYSTAL to XTAL. Added value NONE (0x00) to TCA0.SINGLE.EVCTRL EVACTB. Renamed CLKCTRL.XOSCHFCTRLA.SELHF value CRYSTAL to XTAL for AVR DB devices. We should cut them a bit of slack here, they hadn't released an AVR with an HF crystal since they bought the company :-P still figuriing out new conventions.

SpenceKonde, 2021-05-19:
This is corrected in 1.3.6, released today.

closed 2021-05-19 by SpenceKonde

classiphil, 2021-05-19:
Dear SpenceKonde,

sorry for not answering so long.

There was no specific reason for a different clock. I am just tinkering. ;)
Funny indeed. :)

Looking forward to your new release.

Best

Hans  from Germany