SpenceKonde/DxCore #105: Servo library does not compile due to extra tokens after an #endif

DaleMitchell, 2021-05-11:
```
Compiling libraries...
Compiling library "Servo_DxCore"
"C:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=avr32da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.4\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=4UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\cores\\dxcore/api/deprecated" "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\cores\\dxcore" "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\variants\\28pin-standard" "-IC:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\libraries\\Servo_DxCore\\src" "C:\\Users\\dalem\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.5\\libraries\\Servo_DxCore\\src\\megaavr\\Servo_DxCore.cpp" -o "C:\\Users\\dalem\\AppData\\Local\\Temp\\arduino_build_891688\\libraries\\Servo_DxCore\\megaavr\\Servo_DxCore.cpp.o"
C:\Users\dalem\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.5\libraries\Servo_DxCore\src\megaavr\Servo_DxCore.cpp:114:8: warning: extra tokens at end of #endif directive [-Wendif-labels]
 #endif {
        ^
C:\Users\dalem\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.5\libraries\Servo_DxCore\src\megaavr\Servo_DxCore.cpp:115:3: error: expected initializer before 'ServoHandler'
   ServoHandler(0);
   ^~~~~~~~~~~~
C:\Users\dalem\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.5\libraries\Servo_DxCore\src\megaavr\Servo_DxCore.cpp:116:1: error: expected declaration before '}' token
 }
 ^
Using library Servo_DxCore at version 1.2.1 in folder: C:\Users\dalem\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.5\libraries\Servo_DxCore 
exit status 1
Error compiling for board AVR DA-series (no bootloader).
```

SpenceKonde, 2021-05-13:
Thanks for PR, I went a bit further cleaning up that area of the file, though. And also fixed issues with the examples not referencing the right library.

SpenceKonde, 2021-05-13:
Fixed with 
64a1b00aaf24621f1e648db5ebc8161f015cee0b

closed 2021-05-13 by SpenceKonde