SpenceKonde/DxCore #107: Can PF6 really be used as output for CCL LUT 3?

DaleMitchell, 2021-05-12:
The PF6 pin is noted to be input only in the datasheets from Microchip; however, the pinout diagrams in this GitHub repository show PF6 to be an alternate pin for CCL LUT 3 output.

SpenceKonde, 2021-05-12:
Nope.. Maybe on the DD series when those finally arrive., but not the DA/DB series

SpenceKonde, 2021-05-13:
Fixed. Normally I summon @MCUdude  (he does the pinout diagrams for me) - but I was able to do this myself. I'll have to make sure that this change is communicated to him when I get changes made to them inthe future - I do need to have him adapt them to show that the core supports PWM all of thje TCA mappings, and on PA4/5/6/7 for TCD . for the other mappings of the Type A's you just have the same symbols, but lighten the color so it looks faded or l;ike a disabled option, which it is. But how to express in the diagram that from TCD you get PA4 and PA5, PA5 and PA6, PA6 and PA7 or PA 4 and PA7, but if you try to have PA4 and PA6 or PA5 and PA7 on at the same time, setting either of them changes duty cycle of the other too.

closed 2021-05-13 by SpenceKonde