SpenceKonde/DxCore #111: Found two bugs in TWI library

MX682X, 2021-05-20:
When I was trying to interface a I2C Sensor with a AVR128DA48 it didn't work. After some debugging I found the two problems in the twi.c file:
1. When changing the clock, the TWI Master is disabled and reenabled. But when you disable the module, the BUSSTATUS becomes undefined too, thus I suggest adding 
`TWI0.MSTATUS = TWI_BUSSTATE_IDLE_gc;` 
at the end of the 
`void TWI_MasterSetBaud(uint32_t frequency)` 
function. 

2. At the end of a transaction the MCU was issuing a NACK. I think it should be an ACK so I removed the `TWI_ACKACT_bm` from line 573 & 575.

SpenceKonde, 2021-05-20:
Yikes! Thanks.
#1 is new in this version, as previous versions didn't disable the TWI module before setting the TWI SCL clock to a frequency that was not what the user requested. Since it's been like that for ages and I didn't receive any complaints until a couple of weeks back about the fact that the clocks speeds it set it to were totally bogus, I'm going to say that this is not critical. Do you agree, or am I missing something (like that being called more frequently than I realize leaving Wire unusable)?
 
Can you describe in more detail what was going on with #2? That wasn't changed recently. Under what circumstances is that happening? That sounds like a serious defect, and yet, as I said, that hasn't been touched since the creation of the core, nor in megaTinyCore before it, and you're the first one to notice. Is this something that only causes problems in edge cases? 

I ask these questions \ because I'm trying to figure out whether the release (which was only 2 days ago) introduced a critical regression and needs an emergency fix, or if it can wait until it gets picked up by normal releases.

Thanks again for not only finding these but doing the detective work to find the cause of the problems

MX682X, 2021-05-20:
Well, for 1. , my sensor (MLX90614) was sending NACKs because of a too low clockrate (maybe a bit less then 100kHz?). It might not be really critical, since other devices may be more tolerant, but it would have saved me some time on solving the issue.

For 2., I was a bit too hasty on the judgement. After double checking the I2C User Manual (Paragraph 3.1.6) 

> There are five conditions that lead to the generation of a NACK: [..] 5. A master-receiver must signal the end of the transfer to the slave transmitter.

the code works correctly in that regard. 

I actually forgot to mention a third issue: `Wire.snap()` doesn't work as expected. The 48-pin version has 2 alternative multiplex positions, but the code doesn't allow to use the second. I wanted to have my I2C Master on PC2 & PC3 which would require to set `PORTMUX.TWIROUTEA = 2;` but `Wire.snap(2)` failed the checks. `Wire.snap(1)` would not have changed the Master pins at all, only the Slave pins in Dual mode.

SpenceKonde, 2021-05-20:
The sensor is incompatible with speeds that close to spec??? That's weird - the I2C spec has 100 kHz as standard speed. So slightly under 100 kHz should be fine (though it sounds like switching clock speed when enabled is broken?). We should be defaulting to ~100 kHz, which is what the standard calls for, unless setClock() is called. 

The TWI hardware is byzantine... I hadn't realized that the the status register has to be written to after enabling the TWI module. How bad is the current behavior? It sounds like if the user does not call setClock() with TWI enabled, it just won't work; is that accurate?  
I've checked in a correction for this. We shouldn't be doing this blindly in general. Setting it before we begin was called, it would enable the master whether we were ready or not...)

Instead of blindly disabling it as it does now, I  switched to do: 
```
  uint8_t restore = TWI0.MCTRLA;
  if (restore & TWI_ENABLE_bm) {
    TWI0.MCTRLA = 0; // this is way more flash-efficient than &= could ever be; saves 6 or 8 bytes
}

```
Then at the end:
```
  if (restore & TWI_ENABLE_bm) {
    TWI0.MCTRLA  = restore;
    TWI0.MSTATUS = TWI_BUSSTATE_IDLE_gc;
  }
```
For me it's compiling examples to only 6 bytes more than the version that went out with 1.3.6. The reason I'm so aware of flash usage is that this has to get ported to megaTinyCore too, where people have been on my ass about flash usage... The people using parts with 2 and 4k of flash are a surprisingly vocal minority (and less of a minority thanyou might think, too, thanks to Microchip not releasing an ATtiny812. Folks really like their 8-pin packages.

Okay cool re:2. 

Re: 3 
Wire.swap() is working as intended. 

Wire.swap() has two options, zero (0), corresponding to the default pin position 
-and-
One (1), corresponding to the "ALT_2" pin position. 

Since Wire.h does not support Dual Mode, and `ALT_1` is identical to default except when Dual Mode is enabled, The Wire library does not support dual mode for the simple reason that nobody has written the code to do it. It wasnt clear to me how to adapt the state machine model that the library uses to accounjt for two nearly independent functions, or how independent slave and master mode actually end up being. Our wire library was taken from the official arduino megaavr boards package. Now you may notice that all the hard stuff is in twi.c and twi.h. That's not a coincidence; Those two files, unlike the rest of the library were written by.... _By a Microchip employee, as part of his official duties as such. It's the reference implementation of TWI_ for the ATmega4809. 

I do not know when I might get to look at this. I have a HUGE list of action items that I do know how to do; obviously where I can make progress on something and know that I'm not wasting time barking up the wrong tree are going to take priority. I don;'t think I've used I2C on the Dx-series parts. Period. I was going to, but then someone else did and reported that it worked for them and I went off to deal with the next bug report.

MX682X, 2021-05-20:
I've got two of this sensors and both behave oddly with the working frequency.... maybe I got a bad batch, so just forget about that

Re:1 - Indeed, I have seen no activity on the lines as long as the BUSSTATUS was 0x00, even if MADDR and MDATA were written.
Your code improvement seems to work, but I have no other I2C devices to double-check it.
You should probably add to the documentation that the code has to call `Wire.setClock()` after `Wire.begin()`, otherwise it will be overridden by the Default Frequency of 100kHz

Re:3 - `Wire.swap(1)` is not working though: ` PORTMUX.TWIROUTEA = (PORTMUX.TWIROUTEA & 0xFC) | 0x01;` the `0x01` sets the TWI0 interface to the `ALT_1` mode. (Wire.cpp, line 224). It worked correctly when I changed it to `0x02`

MX682X, 2021-05-20:
I was thinking and realized, calling `Wire.setClock()` before `Wire.begin()` is useless since the latter is overwriting the clock with the default frequency anyway. This means that `Wire.setClock()` should only be called when the Master functionality is already enabled.
I suggest to change the TWI_MasterSetBaud function this way: 
Add this at the start:
```
if (twi_mode != TWI_MODE_MASTER) {
		return;   //Abort if called before Master mode is initialized
}
uint8_t restore = TWI0.MCTRLA; // The TWI master should be disabled while changing the baud rate
TWI0.MCTRLA = 0; // this is way more flash-efficient than &= could ever be; saves 6 or 8 bytes 
			      
```

and this at the end:
```
TWI0.MCTRLA  = restore;
TWI0.MSTATUS = TWI_BUSSTATE_IDLE_gc;
```

It would make the code smaller.

SpenceKonde, 2021-05-22:
I honestly had no idea that the frequency set before begin wasn't retained. It needs to either set an internal count of what the frequency is, or do nothing until begin is called. I get a bad feeling from making it fail silently -  but that's what it was doing before if you tried to set baud rate before callintg begin, right because you'd set it and then it would get automatically unset?

SpenceKonde, 2021-05-22:
The swap situation is going to need some work. What I wrote in the docs about available options is wrong. I think setting it to swap(2) should have worked, except that I messed up the test, and it instead checked whether there was a third swap option before including the code to handle a 2.

MX682X, 2021-05-22:
> I honestly had no idea that the frequency set before begin wasn't retained. It needs to either set an internal count of what the frequency is, or do nothing until begin is called. I get a bad [...]

I just checked the original Arduino Wire library and there `begin()` also overwrites the TWI frequency with the deafult 100kHz . Based on some Forum posts it seems common knowledge to call `setClock()` after` begin()` though. I checked a OLED I2C library too, and there it was also called after the `begin()`.
I considered using an internal counter for the frequency, but that would have to be an uint32_t, so 4 bytes of RAM? Not a good idea since theese 4 bytes would be allocated uselessly for 99% of the time. Another way would be to check in the `begin()`, if the TWI baudrate register is NULL: if it is, go to default frequency, otherwise do nothing. It compiled to an even bigger code Flash-wise though.
Best option would be to make `TWI_MasterSetBaud(uint32_t frequency)` return NULL in case of an error and the local variable baud at the end of the functon. The compiler will optimize away the returns if it is not checked in the main code keeping the size small.

SpenceKonde, 2021-05-24:
Okay well, them I will follow what the official library does says and document the need to call begin first.

SpenceKonde, 2021-05-27:
I think I finally found the bug you were running into with the pinswaps too!

closed 2021-05-27 by SpenceKonde