SpenceKonde/DxCore #113: Serial enhancements required

SpenceKonde, 2021-05-26:
1. We need the capability to support a larger number of pin mapping options. Currently it's only 2, we need to support up to 7 including none, and we need to do it in a vaguely efficient manner, not like it's done now. 
2. We need the ability for a serial port to start by default in a given swap state (for example, Serial0 whbeb

3. And if we could shrink the size of the binaries it generates that'd be awesome too. print and the ISRs are fat and slow, The megaTinyCore people on 2k and 4k parts are complaining about the size, and it's only a miracle people aren't complaining about the speed too.

MX682X, 2021-10-29:
Well, there we go, removed the trailing whitespaces that caused some errors in the checks. Also finished the pinmux none option. Right now, it can be only enabled with the swap function, not with pins. Also, if the none option is available, the swap function accepts any negative value to enable the pinmux none, because that's smaller.

SpenceKonde, 2021-11-21:
This is checked in now. Huzzah!  Plus loopback/opendrain/RS485 mode and the cleanup of the variants that this permitted. Event and RX only, TX only modes untested. The RX/TX only i'm confident in. I have no idea what I'll find with the event stuff. The documentation for some features borders on useless.

closed 2021-11-21 by SpenceKonde