SpenceKonde/DxCore #115: Problems burning Bootloader onto DB-Series MCU

ProxyPlayerHD, 2021-06-06:
I recently got my PCBs for my AVR128DB48 Arduino-like Board.

but after assembling one i was unable to burn optiboot onto it.
i used the same exact programmer (jtag2updi using an Arduino Nano) as for my previous board for the AVR128DA48.
here the Error log after trying jtag2updi: [PASTE](https://pastebin.com/wht34qVZ)

both boards are almost identical, only noticable differences are the MCU itself (DB/DA) and the DB Board having the same reset circuit as regular Arduino Boards while the DA Board's Reset is only controlled by a push-button.
here an image of the Schematic in case the mnistake is hardware related: 
[image]

i also tried the other method, Serial UPDI, only using a resistor because i have no compatible diodes at home. the UART/USB Module i used is called "SH-U09" which has the CP2102 Chip on it.
either way it didn't work either, here it's Error Log: [PASTE](https://pastebin.com/VZs7PWH3)

i'm really not sure why this won't work...

SpenceKonde, 2021-06-06:
I don't suppose you have the push button the wrong way around do ya, with the 2 pins that are connected to eachother going to the two sides of the circuit, instead of pins that are only connected when pressed. That happens all the time with those buttons, and I notice your part library doesn't even make explicit which pins are which. If reset is held low, you can't program them. Check that reset isn't being held low.

I also will note that DB parts were always slightly pricklier to program with jtag2updi, and I could never figure out why (the sort of thing where it would eventually work, after powercycling and resetting programmer and target a few times). That was one of the reasons I gave up on jtag2updi (do note that it is no longer the recommended programming method - it's unreliable garbage and always was, but now we have something else that isn't garbage and that WE CAN FIX, unlike jtag2updi, where you need to be able to crosscompile avrdude for what, 6 platforms? to do a release with - and then you'd be known as a person who knows how to modify avrdude, and which point you're a marked man - you'd have to wear a disguise to go to the pharmacy lest someone see you and ask when you're going to fix one of the other several thousand open bugs that they care about...

ProxyPlayerHD, 2021-06-06:
the Button is correctly wired, using my Logic probe i can see that the Reset pin is high by default and pressing the button pulls it low.

also when programming with Serial UPDI i just had the resistor on a breadboard, could that cause issues? i also selected the lowest baudrate to maybe minimize the amount of noise you get from breadboards

SpenceKonde, 2021-06-06:
I've got no solutions, other than to note that I did see that failure mode when reset was being mishandled. I think in my case it was a solder bridge between reset and updi, rather than it being shorted high or low. 

It's always been a wiring problem when I had issues where one persistently refused to program (meaning assembly problem because schematic is correct as drawn.)

ProxyPlayerHD, 2021-06-06:
i also checked the Resistance between the UPDI Pin and various other things on the board, it only has direct continuity with PD7 and PB3 (and of course the Programming Header)

and no it doesn't seem to be a mistake on the PCB that PD7, PB3, and UDPI are connected as i checked those same pins on my second, unsoldered/untouched AVR128DB48 i still have lying around, and aparently that is just a thing this MCU does for some reason...

I'm thinking about ordering some 1N4148 Diodes and seing if i can get the ideal Serial-UDPI circuit set up.
if that still won't work then i'll just design a custom PCB for the Serial-UPDI Programmer (shouldn't be hard, it's just an FT232, a Diode and a Resistor, just all in one neat package)
and if that still won#t work, then i'll just stick with the Raspberry Pico and my STM32 board until i come up with a solution.

SpenceKonde, 2021-06-06:
Don't use 1N4148. I was mistaken when I had recommended it, and it's not
actually a schottky. You need a small signal _schottky_ diode.

If hand assembling with SMD parts, you probably want the SOT-23 ones - they
only go on one way (no near invisible polarization line - one of the people
I have hired to help assemble small runs refuses to place SOD-123 diodes
because they can't see the line). and they don't wobble side to side like
SOD-123 does.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SpenceKonde, 2021-06-06:
And, euhm... I measure 800kOhm to 1.5 meg between UPDI and PD7 and PB3 on my parts... both DB48 and DB64. And between PD7 and UPDI. 

Sure it's not PC3? Because then you'd be measuring between the three supply pins.... and the only mystery would be why the VDDIO2 was shorted to supply... and if it was mounted on a PCB with single supply, that would be expected.  Hopefully (for your sake) that this was a fumbled measurementm and you didn't either get a bunch of boards fabbed with the pinout off-by-one, or end up with something you can't assemble with acceptable yield due to the part ending up with each pin between a two pads instead of properly centered (only way I've ever reworked that successfully is to remove the chip and throw it out, and then drag solder on a new one)

ProxyPlayerHD, 2021-06-07:
nope, the pinout is correct, i made the PCB Footprint myself using the offical datasheet (i still rechecked it to make sure).
and rechecking the "shorts" it's between pin 7 (PB3), 27 (PD7), and 41 (UPDI).
and again this cannot be related to the PCB at all as i'm getting the same results from my fresh, untouched AVR128DB48 as well.

can it be that i got some dead ICs that have some internal shorts? it seems very unlikely but i cannot really think of any other reason why you can't measure the same shorts

SpenceKonde, 2021-06-07:
I would say it is inconceivable that they would ship product that badly broken -even though this is the company who shipped the AVR32DA to customers without having even tested any of them with the equivalent of blink (and hushed up the whole thing) - but that's a systematic testing failure, a hole in their testing coverage (albeit one large enough to drive a truck through), and it's no secret that they have a lot of those (as documented in the errata). They're going to do tests for shorted pins when they do the factory oscillator and temp calibration. In fact, I'm pretty sure they do that using UPDI with a special factory-calibration key. 

Hmmm. I looked at the pinout chart again. It wouldn't make sense if those were shorted together on  the 48-pin package - it doesn't line up with any power/ground pins...

Unless....

you're mistaking the orientation markings, and have the chip rotated 90 degrees from the right way around? In that case, what you thought was pin 7 (PB3) would be pin 43 (ground), what you thought was pin 41 (updi) would be pin 29 (ground), and what you thought was pin 27 (PD7) would be pin 15 (ground). That is just so much more likely than a batch of bad chips...

ProxyPlayerHD, 2021-06-08:
bruh.

i just saw the fucking dot on the Chip.

for some reason i didn't look at it while soldering and thought that the text on the chip had to be right side up for pin 1 to match, but that was wrong. Pin 1 of the IC is on Pin 13.

i'm dumb

this is exactly why i always buy multiple chips in case i fuck one up.

i'll solder a new board and maybe attempt to save the one i already have. but desoldering is pretty hard if only have a regular soldering iron

SpenceKonde, 2021-06-08:
I've assembled probably half a dozen boards with the chip backwards? It's up there on the list of "ways I've fucked up that have happened more than once", and certainly my top "unforced" error. Like, if I do a lousy job with the solder paste, and make a mess of bridges, that's poor technique, and my fault, yes, but it's in a different category than putting chips on backwards. I don't think I've done that with a QFP actually, always SOIC-14's because the only orientation guidance is direction of the text that is supposedly on the top of the chip (I literally cannot read it under normal conditions, and I am young and have good eyesight) and an almost invisible rounding on one edge. I really wish they used a dot instead of that rounded edge. 

Our of a fraction of the sample size though, I have managed to put on about as many 64-pin Dx-series parts cockeyed or one-row-of-pins off. . Only way I've ever fixed one is to heat-gun the chip off, throw it in the trash (it's always mangled by then) and drag solder on a new one. Some time I'm gonna try running one through a reflow cycle, opening the oven at the end, and then nudging the chip into position.... 

But yeah, considering my experience reworking cockeyed DA64's, faced with a wrong-angle DA48 (same fine pitch) chip,  I would write off the chip itself, and only attempt to salvage the rest of the loaded boards if the effort it would save soldering the rest of the parts in was worth it. IMO the iron is going to be useless. IMO, you'll spend so much time on it that you could ha ve earned enough to buy a dozen of them working sy minimum wage, and I'd predict you'd end up lifting pads on the PCB by the time the chip was off. the methods that could plausibly work. If all I'd use a butane torch before I'd try to take that off with an iron (and as you might guess, the torch is WAY too aggressive and trashes things very quiickly - at least then you waste at most a minute on it.

closed 2021-06-08 by SpenceKonde

ProxyPlayerHD, 2021-06-08:
it's kinda adorable that you closed the issue like this is over. seriously though i'm still having problems...

alright i made a new board, and damn that is the most perfect SMD soldering job i've ever made:
[image]

anyways, Serial-UPDI still didn't work, i used a 2.2k resistor instead of a diode.
here the error log: [PASTE](https://pastebin.com/NxCmeere)
jtag2updi seems to have worked but i cannot upload sketches via USB, here jtag2updi's log: [PASTE](https://pastebin.com/pGqJ8UgA)
and the error i get when trying to upload Blink via arduino's "upload" button: [PASTE](https://pastebin.com/KDxjNXym)

SpenceKonde, 2021-06-10:
So serialupdi with unspecified hardware and an incorrect-value resistor instead of the recommended **schottky** diode didn't work. Is this surprising?  I can't help further on that without knowing what serial adapter (or chip model + resistance between tx pin of chip and tx pin coming out of the board (a picture is probably sufficient here, most of them are pretty recognizable, and I've tried to buy at least one each of the common models that are making the rounds these days), as well as confirmation of where the resistor was connected (unless the  adapter board has an internal 2.2k resistor too, which it at the high end of the range  I would expect a high risk of it not working with an external 2.2 even with otherwise correct topology. My experience has been that when using the resistor, the  range of values of resistors in various places  in the that it will work with which operating voltages is quite narrow.  I always had a hard time making the resistor method work, it wasn't until two people suggested the diode within days of each other, I did the math, realized a decent schottky diode would bring us well into friendly territory, and wired that up, and it performed flawlessly. Of course some other people are swearing that it's the other way around, so more study is clearly needed. If I saw a scope trace capture (signal mode, trigger on UPDI rising or falling)  I suspect the problem would be immediately apparent to me. 

Anyway, the jtag2updi that succeeds show's a burn bootloader operation with the (no bootloader) option selected. That just sets fuses, and erases anything in the flash. including any bootloader, as it should because you['re preparing the chip for use without a bootloader) in the flash. But the second one shows you trying to upload through the serial port with a different board definition selected; You would need to burn bootloader with the optibootboard def selected if you want the bootloader to be installed. 

Finally, even once those things are corrected, it would appear that you have TX of the serial adapter connected top PA0. which is TX of USART0 in default pins. Serial gets wired TX (transmit) to (receive) in each direction. 

And a third point, you're using an avr DB aren't you:? Why are you using the default serial0 pins at all? Are you not planning to use an external high frequency crystal, and just rely on the internal oscillator?

SpenceKonde, 2021-06-10:
I've had a wonderful time with my Siglent SDS1202 - Seems I live in a Siglent family. Didn't stop using my Siglent SDS1202 until this christmas, when my father got us both shiny new SDS2204X'w with the touchscreen and every feature (had to buy something nice during the pandemic, but we were all afraid to go out to christmas shopping) 

Yeah - you're the second person to have this misunderstanding... but here's the weird part - The otyher person was just a couple of months ago. And I've maintained cores for like 5 years. What changed in how people view those options, I wonder? 

I wish there were some way to put a notice where people would see it to tell them to set the bootloader options for what you want, and then "burn bootloader" to make the chip match what you've selected. "dress for the settings you want, not the settingsyou have" :-P 

,my bad on the schematic, I somehow read them backwards.. 

Can't argue with that -. the internal oscillator on the Dx-series certainly is a nice piece of hardware. Can crank it to 32 no problem at room temp! And the PLL will work at 4x muiltiplier on that  too. These things arte spec'ed with an incredible amount of headroom (of course they asre, some are speced hot enough to be used at temperatures in excess of boiling point of water.... I wouldn't mind the internal oscillator if the cal setting had finer granularity, but like it's around half a percent per increment or something like that, which kinda take the point out of autotune, you know? I don't know what all happened with the modern AVRs, but we losrt the two MSB, of OSCCAL:, and only the 2 series have gotten one of those back (they have an AWESOME internal oscillator, some are stable at room temp all the way up at 32! (and even 1-series did 32 off ecxternal clock, if you were a nice healrthy 5v, but died not far below that voltage)

I had a Dx doing 40 off ext clock no problem but it wouldn't do 48 :-/ And you can't nudge the voltage up to the very upper edge of the range to try to make it work, because the voltage gets regulated down anyway. Not that I have any particular need for a 48 MHz AVR, but because I can, I got some of the high temp spec ones to try, just in case those go a little higher . Alsi planning to try crystal @ 40 MHz for the hell of it. :-P 

I do like having a real crystal for the timebase, personally - not that I always or even usually need it, but fo a serious business board, where I have plenty of pins for it, I always want to puit one in :-P 

IMO if you're doing anything demanding on an AVR, you'd be nuts to not use a Dx-series - what else would you use, a mega2560 classic running at half the speed, with store, push cbi and spi each running twice as long, and periherals that were stale a decade ago? If they hadn't come out with something soon, AVR was not going to rekmain a viable architecture. Especially not with the prices they charged for cl;assic AVRs.