SpenceKonde/DxCore #117: analogWrite is broken

Kozma04, 2021-06-12:
When I attempt to call analogWrite on any PWM pin on the AVR128DB64, I get the following compilation error:
`C:\Users\Kozma\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.6\cores\dxcore\wiring_analog.c:40:7: error: call to 'badArg' declared with attribute error: 
       badArg("digital I/O function called  is constant, but not a valid pin");
       ^
`

SpenceKonde, 2021-06-13:
How could this be???? This version was released weeks ago!

SpenceKonde, 2021-06-18:
What the fuck why has nobody else reported this; I am not in a good position to work ion it right now because I have the entire core taken apart in the process of sorting out dsomerthing else

MCUdude, 2021-06-18:
@SpenceKonde can't you just git stash the stuff you're working on, fix the issue and git stash pop them back?

SpenceKonde, 2021-07-28:
This is the absolute top priority among the things that are niot poart of the even higher prorities the sweet grey kitty who loves in my toom now fed, keepin myselfg  gong anywhrete to an y sort of thing..... tbose all feel hopelessly overwhelming, I'At the start of the pandimict I was struggling with my goal of leraving the house every day, Inow I'm lucky if I go further than the peooperty line monthly.

SpenceKonde, 2021-10-01:
No issues with doing that. You can remove the calls to them or replace the body of the function with a 'return' etc. 

Anything to defang the check functions is safe. The purpose of those is to generate compile errors when users do something that we *know* will always be ignored (because they're trying to do something impossible) at runtime. The idea is that when we know that a line is always going to not do what was asked of it, we should point that out, so that it's an "easy to fix" compile error. not a difficult to find behavioral bug visible only at runtime, where hours or days later you discover that you fatfingered the pin number and are trying to write to pin 189 instead of 19 in one instance or something. Better to error on compile when someone tries to write to pin 189 right? Because no correct program would ever do that, This goes under the heading of trying to do as much as possibler to help the user with debugging on a platform with no exception handling and no way to report errors at ru. No part of that or any other check function makes it into the binary. This is the first time I've had one of these tests.... act like this,... 

I have reproduced the issue though, and I am really confused as to how, in my test program, it could possibly conclude that 18 >= 27 there.

SpenceKonde, 2021-10-01:
Well, your jumbled code from my core gave me the key insight to fix this, and the problem is truly moronic. 

See where there's that `{}`

One of them should go down to the end of the function.. the call to badCall was supposed to be within that conditional, but I accidentally (I blame my editor's autobracket feature), instead had it conditionally doing nothing and then calling the badCall whenever the arguments were constant. Okay, well that was dumb.

dpetican, 2021-10-01:
Yes.  The technical term for that is derp ;)  I should have seen that as well.

`line 1line 2`

Speaking of derps what am I doing wrong with code blocks?  I just types that out by hand and no line breaks.

SpenceKonde, 2021-10-01:
You are trying to do  multiple lines with an inline comment. 
backtick-code-backtick is `code`

newline, three of the backtick, newline, code, newline 3xbacktick, newline 
```
a multiline
code 
block
```

closed 2021-10-01 by SpenceKonde